OSSystems/meta-browser #631: /usr/lib64/chromium/chromium-bin: No such file or directory

StGale, 2022-04-23:
Hello,
I have created an image YOCTO for intel core E3900 following https://github.com/intel/iotg-yocto-bsp-public/tree/e3900/master
then using meta-browser I also compiled chromium-x11 on my local.conf.
The HDDIMG on board start but when I try to run chromium the problem is this message
root@intel-corei7-64:~# chromium
root@intel-corei7-64:~# /usr/bin/chromium: line 51: /usr/lib64/chromium/chromium-bin: No such file or directory
but the file is present
root@intel-corei7-64:~# ls -las /usr/lib64/chromium
total 183480
     4 drwxr-xr-x  4 root root      4096 Apr 23 01:19 .
    64 drwxr-xr-x 73 root root     61440 Apr 23 11:41 ..
    20 -rwsr-xr-x  1 root root     18672 Apr 23 01:14 chrome-sandbox
  1096 -rw-r--r--  1 root root   1120967 Apr 23 01:14 chrome_100_percent.pak
  1508 -rw-r--r--  1 root root   1541344 Apr 23 01:14 chrome_200_percent.pak
157864 -rwxr-xr-x  1 root root 161700472 Apr 23 01:14 chromium-bin
     4 -rwxr-xr-x  1 root root      1962 Apr 23 01:14 chromium-wrapper
 10260 -rw-r--r--  1 root root  10505952 Apr 23 01:14 icudtl.dat
     4 drwxr-xr-x  2 root root      4096 Apr 23 01:19 locales
 11868 -rw-r--r--  1 root root  12156738 Apr 23 01:14 resources.pak
   236 -rw-r--r--  1 root root    240440 Apr 23 01:14 snapshot_blob.bin
     4 drwxr-xr-x  2 root root      4096 Apr 23 01:19 swiftshader
   548 -rw-r--r--  1 root root    558968 Apr 23 01:14 v8_context_snapshot.bin

can someone help me?
Thanks
Stefano

StGale, 2022-04-23:
just to add info I cloned the warrior branch of meta-browser

StGale, 2022-04-23:
other info:
Ubuntu 18.04 used to compile the yocto image.

during the chromium compile phase I have created two symbolic link on 
    cd ../build/tmp/work/corei7-64-poky-linux/chromium-x11/80.0.3987.132-r0/recipe-sysroot
    mv lib lib32
    ln -s lib64 lib
    cd usr/
    ln -s lib64 lib
just to prevent an error on search file 64bit on lib directory.

to complete the history, I put my local.conf:
stefano@stefano:~/ibexYoctoIntel/yocto_build/build$ cat conf/local.conf 
#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

HOSTTOOLS += "git-lfs"

#
# Machine Selection
#
# This sets the default machine to be intel-corei7-64 if no other machine is selected:
MACHINE = "intel-corei7-64"

# This set the default kernel to linux-intel version 4.14 LTS
PREFERRED_PROVIDER_virtual/kernel = "linux-intel"
PREFERRED_VERSION_linux-intel = "4.19%"

#
# Leaf Hill BSP specific settings
#
# This sets BSP_SUBTYPE to leafhill.
# This setting is meant for Bitbake to pull the correct set of
# kernel config fragments in yocto-kernel-cache.
#
# It is not recommend to remove if your target platform is
# based on Intel Atom E3900 SoC Family.
BSP_SUBTYPE = "leafhill"

#Exclude piglit, ltp and mesa-demos packages
PACKAGE_EXCLUDE = "packagegroup-core-apl-extra"

#
# Apollo Lake Yocto BSP supports different Audio features,
# i.e. legacy HD-Audio, CAVS HD-Audio, and CAVS SSP Audio codec support.
#
# By default, this configuration will build CAVS HD-Audio feature.
# To build CAVS SSP Audio feature, uncomment the line below.
#AUDIO_FEATURES = "ssp"

#
# Machine Configuration Override for Apollo Lake platform
#
# This is a stopgap measure for configuring machine settings for Apollo Lake platform
# primarily for Pre-Production SW release until the changes are upstream to Yocto Project.
# These settings are applicable for Intel Apollo Lake project only.
# Please comment out when you are compiling for other projects.

# Uncomment when you are integrating closed source media stack
#MACHINE_HWCODECS = ""
[... 287 lines cut ...]
#SSTATE_MIRRORS ?= "\
#file://.* http://someserver.tld/share/sstate/PATH;downloadfilename=PATH \n \
#file://.* file:///some/local/dir/sstate/PATH"

#
# Qemu configuration
#
# By default qemu will build with a builtin VNC server where graphical output can be
# seen. The two lines below enable the SDL backend too. This assumes there is a
# libsdl library available on your build system.
#PACKAGECONFIG_append_pn-qemu-native = " sdl"
#PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
#ASSUME_PROVIDED += "libsdl-native"

# CONF_VERSION is increased each time build/conf/ changes incompatibly and is used to
# track the version of this file when it was generated. This can safely be ignored if
# this doesn't mean anything to you.
CONF_VERSION = "1"

# Enable the security flags as per SDL 4.0 requirement
require conf/distro/include/security_flags.inc
SECURITY_CFLAGS = "-fstack-protector-strong -pie -fpie -fPIC -D_FORTIFY_SOURCE=2 -O2 -Wformat -Wformat-security"
SECURITY_NO_PIE_CFLAGS = "-fstack-protector-strong -D_FORTIFY_SOURCE=2 -O2 -Wformat -Wformat-security"
SECURITY_LDFLAGS = "-Wl,-z,relro,-z,now,-z,noexecstack"
AUDIO_FEATURES = "ssp"

rakuco, 2022-04-25:
The "no such file or directory" error message sometimes means you're trying to run an executable that's not supported (a 32-bit binary in a 64-bit installation that does not support it, for example).

* Are you doing a regular 64-bit build targeting a 64-bit platform?
* chromium-x11 80.0.3987.132 is _very_ old. Would it be possible to update the recipe to a more recent version? Similarly, the warrior branch has not been supported upstream for years, is there any chance of moving to a supported branch?
* What happens if you manually run the `chromium-bin`? Does it still show up the same "no such file or directory" message?

StGale, 2022-04-28:
hi found a solution disabling the multi libraries

# Multi-libraries support is now "ON"
# Please uncomment the 4 lines below to enable multilib support.
require conf/multilib.conf
DEFAULTTUNE = "corei7-64"
MULTILIBS = "multilib:lib32"
DEFAULTTUNE_virtclass-multilib-lib32 = "corei7-32"

comment out the 4 lines on default local.conf.sample after checkout MR5-B-04

https://github.com/intel/iotg-yocto-bsp-public/tree/e3900/master

rakuco, 2022-04-28:
OK, thanks. I'm closing this issue since it doesn't seem to be a meta-chromium bug.

closed 2022-04-28 by rakuco