OSSystems/meta-browser #639: Steps needed to build

MarkGhanz, 2022-05-16:
This is more of a question than an issue. Are there detailed instructions about how to build a chromium browser out of this project?

rakuco, 2022-05-18:
It depends a bit on what exactly you want to build:
* https://chromium.googlesource.com/chromium/src/+/refs/heads/main/docs/linux/build_instructions.md contains instructions for building a git checkout using Chromium's llvm binaries and bundled third-party dependencies.
* If you want to build a tarball and/or build a release with your own toolchain or build arguments, you're pretty much on your own: the best approach is to see how others (including meta-browser) do it and do something similar.

nros, 2023-01-25:
See my example here: https://github.com/firecoder/yocto-meta-browser-test