OSSystems/meta-browser #640: How to add chromium or firefox to yocto dunfell

eduardocorralmunoz, 2022-05-25:
Hello. I need to add a web browser (chromium or firefox) to my dunfell project. I have tried repeatedly without result, I have searched for solutions on googe but also without result. There are always a lot of chained errors due to the meta-clang layer:

ERROR: ParseError at /home/eduardo/var-fslc-yocto/sources/meta-clang/conf/nonclangable.conf:2: unparsed line: 'TOOLCHAIN:pn-babeltrace2 = "gcc"'

If you comment line 2 in "nonclangable.conf" the error jumps on line 3 and so on

Could you please help me?

Thanks in advance.

kraj, 2022-05-25:
you need latest `dunfell-clang12` branch of meta-clang

eduardocorralmunoz, 2022-06-06:
I tried dunfell-clang12, but still have Errors.
ERROR: ParseError at /home/eduardo/var-fslc-yocto/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_101.0.4951.54.bb:33: unparsed line: 'CHROMIUM_EXTRA_ARGS:append = " --ozone-platform=wayland"'

TejasAbhang4042, 2022-06-24:
Hello,
Did you succeed in resolving the above error and running the chromium browser?
I am facing same issue.

rakuco, 2022-06-24:
These look like errors that happen when you're using a Bitbake version that's too old and does not support the new override syntax. See https://github.com/OSSystems/meta-browser/issues/543#issuecomment-912605725

TejasAbhang4042, 2022-06-24:
Instead of applying the patch for bitbake, can I simply replace ":" with "_" in the chromium's .bb file?
Because all other layers that I am using has "_" in their .bb files.

rakuco, 2022-06-24:
You can try, but your mileage may vary. To be clear, If you update Bitbake, the old syntax won't stop working on dunfell AFAICS.

TejasAbhang4042, 2022-06-28:
> These look like errors that happen when you're using a Bitbake version that's too old and does not support the new override syntax. See [#543 (comment)](https://github.com/OSSystems/meta-browser/issues/543#issuecomment-912605725)

Thanks for the response @rakuco .

I tried replacing ":" with " _ ", but had no success compiling it.

So I applied patches from :
https://git.openembedded.org/bitbake/commit/?h=1.46&id=c5418eae56cc50dbae7951c055434a0c456c53a4
AND
https://git.openembedded.org/bitbake/commit/?h=1.46&id=a6d5fb7554e3cf071e453db56a1e7469ac44277c

It seems to be working, but I am facing some issues compiling the meta-clang for now.

TejasAbhang4042, 2022-07-01:
I've cloned the master branch of meta-browser and I am using it with Dunfell.
I got following errors while compiling:

ld.lld: error: found local symbol 'NSS_3.9.3' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.8' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.4' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.6' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.2' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.9' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.15' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.13' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.7' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.3' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.2.1' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.18' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.10' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
ld.lld: error: found local symbol 'NSS_3.12.10' in global part of symbol table in file /home/wp4k/var-fslc-yocto-dunfell/build_chromium_xwayland/tmp/work/aarch64-mx8mn-fslc-linux/chromium-ozone-wayland/101.0.4951.54-r0/recipe-sysroot-native/usr/lib/libsmime3.so
clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

Do I need to use another branch of meta-browser for dunfell?

TejasAbhang4042, 2022-07-01:
As per issue #352 
I have also tried commenting following line in **"dynamic-layers/browser-layer/recipes-browser/chromium-x11_%.bbappend"**

**GN_ARGS += "use_lld=true use_gold=false"**

Still got same results.

robin-bob-duprat, 2022-08-18:
Hello [TejasAbhang4042](https://github.com/TejasAbhang4042),

I was facing the same issue as [yours](https://github.com/OSSystems/meta-browser/issues/640#issuecomment-1171943553).

And with the proposal of the per issue [#352](https://github.com/OSSystems/meta-browser/issues/352), I tried this :

GN_ARGS += "use_lld=**false** use_gold=**true**"

Getting then:
```
ERROR at //build/config/compiler/BUILD.gn:647:5: Assertion failed.
|     assert(use_lld, "LTO is only supported with lld")
|     ^-----
| LTO is only supported with lld
| See //build/config/BUILDCONFIG.gn:334:3: which caused the file to be included.
|   "//build/config/compiler:compiler_arm_fpu",
|   ^-----------------------------------------
| WARNING: exit code 1 from a shell command.
```

So I disable lto doing :

GN_ARGS += "use_lld=false use_gold=true **use_thin_lto=false"**

And then I was able to pass the do_compile() :+1: 

Regards,