OSSystems/meta-browser #645: chromium-ozone-wayland does not detect touchscreen pointer device

apbr, 2022-07-19:
I use chromium-ozone-wayland version: 96.0.4664.110
I start chromium-ozone-wayland with
`/usr/bin/chromium --kiosk --noerrdialogs --no-first-run --enable-wayland-ime --autoplay-policy=no-user-gesture-required http://localhost:80`

To check which pointer type is recognized by chrome following JavaScript code can be used:
```js
// True if at least one coarse pointer device exists.
matchMedia('(pointer:coarse)').matches;
// True if at least one fine pointer device exists.
matchMedia('(pointer:fine)').matches;

```

As I understand it, websites adapt their style according to the detected pointer device of the browser.
A mouse or a touchpad is a fine pointer device. Touchscreens are coarse pointer devices.

My device has a touchscreen.
**Observed behaviour:**
Chromium detects neither a coarse nor a fine pointer device. If I connect a mouse it only detects a fine pointer device.
**Expected behaviour:**
Chromium should always detect a coarse pointer device as the touchscreen is always present.

Is there any flag I have to pass to chromium or anything I have to change in my system to fix this?

rakuco, 2022-08-05:
Chromium 96 is pretty old by Chromium's standards, so I'd start by trying a newer release. Additionally: is it possible to test an official Google Chrome release for your platform and see if that one works as expected?

apbr, 2022-08-09:
I can reproduce it with 103.0.5060.129-r0.
Unfortunately I can't easily test an official Google Chrome release.