OSSystems/meta-browser #646: update to chromium-103.0.5060.129 -- error: unknown argument: '-no-opaque-pointers'

rwmacleod, 2022-07-20:
I've gotten the standard and musl patches to apply for chromium-103.0.5060.129.
In compiling for glibc, qemux86-64, I'm seeing the error the error below. 
I'll see if I can figure out how to deal with it in the morning but I wanted to
let people know that this work is underway and to give someone who may
know more about the state of the clang toolchain to reply. Thanks.

x86_64-poky-linux-clang -target x86_64-poky-linux \
   -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian \
   -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 \
   -Wformat -Wformat-security -Werror=format-security \
   --sysroot=/.../tmp/work/core2-64-poky-linux/chromium-x11/103.0.5060.129-r0/recipe-sysroot \
   -MMD -MF obj/base/third_party/libevent/libevent/evdns.o.d \
   -DHAVE_CONFIG_H -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD \
   -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES \
   -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 \
   -I../../base/third_party/libevent/linux -I../.. -Igen -fno-delete-null-pointer-checks \
   -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables \
   -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants \
   -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 \
   -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables \
   -m64 -msse3 -Xclang -fdebug-compilation-dir \
   -Xclang . -ftrivial-auto-var-init=pattern -fno-omit-frame-pointer -g0 -fvisibility=hidden \
   -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare \
   -O2 \
   -fdata-sections -ffunction-sections -fno-unique-section-names \
   -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-unused-but-set-variable \
   -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter \
   -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional \
   -Wno-unknown-warning-option -Wno-psabi -Wno-ignored-pragma-optimize -Wno-unqualified-std-cast-call \
   -Xclang -no-opaque-pointers \
   -std=c11  -O2 -pipe  \
   -feliminate-unused-debug-types \
   -fmacro-prefix-map=/.../tmp/work/core2-64-poky-linux/chromium-x11/103.0.5060.129-r0=/usr/src/debug/chromium-x11/103.0.5060.129-r0 \
   -fdebug-prefix-map=/.../tmp/work/core2-64-poky-linux/chromium-x11/103.0.5060.129-r0=/usr/src/debug/chromium-x11/103.0.5060.129-r0 \
   -fdebug-prefix-map=/.../tmp/work/core2-64-poky-linux/chromium-x11/103.0.5060.129-r0/recipe-sysroot= \
   -fdebug-prefix-map=/.../tmp/work/core2-64-poky-linux/chromium-x11/103.0.5060.129-r0/recipe-sysroot-native= \
   -g1 -c ../../base/third_party/libevent/evdns.c \
   -o obj/base/third_party/libevent/libevent/evdns.o
error: unknown argument: '-no-opaque-pointers'

@msisov  @kraj

kraj, 2022-07-20:
yeah this option is only in clang15+ and its turned on by default. So its disabling it in chromium. For clang14 this option does not exist and hence disabled so we can delete this option safely for now. Try something like

```patch
--- chromium-102.0.5005.61/build/config/compiler/BUILD.gn.orig	2022-05-19 10:35:52.991415777 +0000
+++ chromium-102.0.5005.61/build/config/compiler/BUILD.gn	2022-05-19 10:36:11.102017131 +0000
@@ -1538,15 +1538,6 @@ config("default_warnings") {
         cflags += [ "-Wno-deprecated-non-prototype" ]
       }
 
-      if (!is_nacl && !(is_chromeos ||
-                        default_toolchain == "//build/toolchain/cros:target")) {
-        # TODO(https://crbug.com/1316298): Re-enable once test failure is figured out
-        cflags += [
-          "-Xclang",
-          "-no-opaque-pointers",
-        ]
-      }
-
       if (is_fuchsia) {
         # TODO(https://bugs.chromium.org/p/fuchsia/issues/detail?id=77383)
         cflags += [ "-Wno-deprecated-copy" ]
```

rwmacleod, 2022-07-21:
@kraj  Thanks, that seems to be what the doctor ordered. My build is 3/4 done so I'll test the musl build as well, then glibc runtime qemux86-64, qemuarm64, clean-up the patches, update the commit log to document all the license changes that I rushed through and send a patch in the coming days.

rwmacleod, 2022-07-21:
chromium-x11 builds and works fine for glibc on qemux86-64/arm64 on poky+<layers> on master branch. 
I'm building chromium-ozone-wayland now.
TCLIBC=musl did work but the mallinfo patch doesn't apply now so I'll fix that next.

@kraj Given that @msisov is on vacation until early August, would the following additional tests make sense:

    Test-built:
    
    * chromium-ozone-wayland:
     - kirkstone, clang, MACHINE="qemuarm64, qemux86-64"
     - dunfell, clang**, MACHINE="qemuarm64"
    
    * chromium-x11
     - kirkstone, clang, MACHINE="qemuarm64, qemux86-64"
     - dunfell, clang**, MACHINE="qemuarm64, qemux86-64"

I can work this out manually but if there is scripting to support it please let me know.
I'm hoping to get this merged early next week to meet an internal deadline. Fun!

kraj, 2022-07-21:
@rwmacleod I think its a major version update so I would not like to regress on the qualification criteria if possible. I can help with master builds for qemux86-64 and qemuarm64, since I have those setups handy.

rwmacleod, 2022-07-21:
@kraj  Agreed. Let me get the musl patch fixed, and write the commit log. PR tomorrow hopefully.

rwmacleod, 2022-07-23:
or on the weekend! :-(

rwmacleod, 2022-07-24:
Resolved in: https://github.com/OSSystems/meta-browser/pull/648

closed 2022-07-24 by rwmacleod