OSSystems/meta-browser #647: chromium-ozone-wayland 101 SVG issue

ambrice, 2022-07-20:
Upgraded from chromium 92 -> 101 on i.MX8QM using wayland, and most SVG files are not displaying (with no error messages in stderr or console).  Did anything change recently in SVG handling?  The same version of x86 chromium run on my laptop (with wayland) displays the SVGs correctly.

ambrice, 2022-07-21:
If I run chromium without --in-process-gpu, the SVG images display correctly (but no HW accel).  Does SVG rendering use the GPU?  Is there a flag to force software rendering of SVG images?

ambrice, 2022-07-21:
More narrowly, if I run --in-process-gpu --disable-gpu-rasterization then SVGs display ok.  So something in GPU rasterization in chromium 101 on i.MX8QM vivante GPU is causing problems for SVG image rendering.

escherstair, 2022-12-07:
As far as I understand, this is a Chromium bug.
Is there a link to a ticket in the Chromium development?

leograba, 2023-05-04:
Is this bug still present in newer versions of Chromium?

leograba, 2023-06-21:
We don't see this issue anymore on the NXP 5.15-based BSP.

Even the older version of Chromium where the bug was present now works when running against the new 5.15, whereas it presented the bug on 5.4.

escherstair, 2023-10-30:
I see the same behavior running [Torizon OS 6.4.0+build.5](https://artifacts.toradex.com/artifactory/torizoncore-oe-prod-frankfurt/kirkstone-6.x.y/release/5/verdin-imx8mp/torizon/torizon-core-docker-evaluation/oedeploy/torizon-core-docker-evaluation-verdin-imx8mp-Tezi_6.4.0+build.5.container.tar)  that is based on NXP 5.15 BSP.
The svg images are shown with big squares of the right color.