OSSystems/meta-browser #649: chromium 103 does not build for imx8qm

ambrice, 2022-08-03:
Building chromium-ozone-wayland on i.MX8QM for wayland only, no x11 libs.  Have the following patch to disable dri:

```
diff --git a/content/gpu/BUILD.gn b/content/gpu/BUILD.gn
index 1a6063b03..c766f5ee3 100644
--- a/content/gpu/BUILD.gn
+++ b/content/gpu/BUILD.gn
@@ -135,7 +135,7 @@ target(link_target_type, "gpu_sources") {
   }
 
   # Use DRI on desktop Linux builds.
-  if (current_cpu != "s390x" && current_cpu != "ppc64" && is_linux &&
+  if (current_cpu != "s390x" && current_cpu != "ppc64" && is_linux && ozone_platform_x11 &&
       !is_castos) {
     configs += [ "//build/config/linux/dri" ]
   }
```

Similar patch worked for 101, but in 103 I'm getting a compilation error:
```
| python3 "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- aarch64-poky-linux-clang++ -target aarch64-poky-linux  -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/var/tmp/yocto/work/aarch64-mx8-poky-linux/chromium-ozone-wayland/103.0.5060.129-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -flto=thin -Wl,--thinlto-jobs=20 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -Wl,-mllvm,-import-instr-limit=5 -fwhole-program-vtables -Wl,--no-call-graph-profile-sort -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-z,defs -Wl,--as-needed -pie -Wl,--disable-new-dtags -Wl,--lto-O2 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/var/tmp/yocto/work/aarch64-mx8-poky-linux/chromium-ozone-wayland/103.0.5060.129-r0=/usr/src/debug/chromium-ozone-wayland/103.0.5060.129-r0                      -fdebug-prefix-map=/var/tmp/yocto/work/aarch64-mx8-poky-linux/chromium-ozone-wayland/103.0.5060.129-r0=/usr/src/debug/chromium-ozone-wayland/103.0.5060.129-r0                      -fdebug-prefix-map=/var/tmp/yocto/work/aarch64-mx8-poky-linux/chromium-ozone-wayland/103.0.5060.129-r0/recipe-sysroot=                      -fdebug-prefix-map=/var/tmp/yocto/work/aarch64-mx8-poky-linux/chromium-ozone-wayland/103.0.5060.129-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now  -stdlib=libc++ -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -latk-1.0 -latk-bridge-2.0 -lgio-2.0 -ldbus-1 -lresolv -ljpeg -lexpat -luuid -ldrm -lxkbcommon -lm -lz -lwayland-client -lgbm -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lasound -lwayland-egl -lpci -latspi -lFLAC -lxslt -lxml2
| aarch64-poky-linux-ld.lld: error: undefined symbol: gl::NativeViewGLSurfaceEGL::NativeViewGLSurfaceEGL(gl::GLDisplayEGL*, void*, std::__1::unique_ptr<gfx::VSyncProvider, std::__1::default_delete<gfx::VSyncProvider> >)
```

maskimthedog, 2022-08-25:
I am seeing the same issue building chromium-ozone-wayland for wayland only for i.MX8MM after applying the same patch.

| aarch64-poky-linux-ld.lld: error: undefined symbol: gl::NativeViewGLSurfaceEGL::NativeViewGLSurfaceEGL(gl::GLDisplayEGL*, void*, std::__1::unique_ptr<gfx::VSyncProvider, std::__1::default_delete<gfx::VSyncProvider> >)
| >>> referenced by gl_surface_wayland.cc:35 (./../../ui/ozone/platform/wayland/gpu/gl_surface_wayland.cc:35)
| >>>               thinlto-cache/Thin-7f2605.tmp.o:(ui::GLSurfaceWayland::GLSurfaceWayland(gl::GLDisplayEGL*, std::__1::unique_ptr<wl_egl_window, ui::EGLWindowDeleter>, ui::WaylandWindow*))

ensc, 2022-09-22:
The immediate problem is that from

```c
#elif defined(__GBM__)
typedef void               *EGLNativeWindowType;
...
#elif defined(USE_OZONE)
typedef intptr_t EGLNativeWindowType;
```

the first branch is used for some sources while the second one for other.  This changes C++ function signatures.

ensc, 2022-09-23:
setting

```
CXXFLAGS += "-D__GBM__"
```

workarounds it for me.

5o5uk3, 2022-10-18:
Hello,

I had exactly the same issue _(but with chromium 105 build)_ and I have chosen the opposite approach.
From my point of view `gbm.pc` file should not add `-D__GBM__` flag as it is already set through `gbm.h` when it is included. That's why I have `patched imx-gpu-viv_6.4.3.p2.0-aarch64` recipe to remove this flag and the chromium compilation is OK.
FYI, the patch applied below:

```patch
Index: imx-gpu-viv-6.4.3.p2.0-aarch64/gpu-core/usr/lib/pkgconfig/gbm.pc
===================================================================
--- imx-gpu-viv-6.4.3.p2.0-aarch64.orig/gpu-core/usr/lib/pkgconfig/gbm.pc
+++ imx-gpu-viv-6.4.3.p2.0-aarch64/gpu-core/usr/lib/pkgconfig/gbm.pc
@@ -8,4 +8,4 @@ Description: vivante gbm library
 Version: 17.2
 Libs: -L${libdir} -lgbm
 Libs.private: -ldl
-Cflags: -I${includedir} -DLINUX -D__GBM__
+Cflags: -I${includedir} -DLINUX
```

Villemoes, 2022-10-28:
Building for imx8mp, having the same issue. Any consensus on whether it needs fixing in chromium or imx-gpu-viv?

5o5uk3, 2022-11-03:
@Villemoes What I can assert is that chromium does not seem to use GPU acceleration with my method. I have to check if it behaves better with open source drivers.

ambrice, 2023-03-27:
We have been stuck on chromium 92 on i.MX8QM for a while but I copied this recipe from NXP's meta-imx layer:

https://github.com/nxp-imx/meta-imx/tree/langdale-6.1.1-1.0.0/meta-sdk/dynamic-layers/chromium-browser-layer/recipes-browser/chromium/

Seems to work, with GPU acceleration, gets us to chromium 101 at least.  I'm using with commit 6079a8b67da524d6cb19c163e689b21e0d7575b1 of meta-browser for chromiums 101 recipe.

maskimthedog, 2023-03-29:
> We have been stuck on chromium 92 on i.MX8QM for a while but I copied this recipe from NXP's meta-imx layer:
> 
> https://github.com/nxp-imx/meta-imx/tree/langdale-6.1.1-1.0.0/meta-sdk/dynamic-layers/chromium-browser-layer/recipes-browser/chromium/
> 
> Seems to work, with GPU acceleration, gets us to chromium 101 at least. I'm using with commit [6079a8b](https://github.com/OSSystems/meta-browser/commit/6079a8b67da524d6cb19c163e689b21e0d7575b1) of meta-browser for chromiums 101 recipe.

What about hardware video decode?

ambrice, 2023-03-29:
I haven't tried hardware video decode but a commit says it was added:

commit fe0d3587d905a8fa32a9830d177f97c90352fdc5
Author: Hou Qi <qi.hou@nxp.com>
Date:   Thu Sep 15 22:28:13 2022 +0800

    chromium-ozone-wayland: Add V4L2VDA support on Linux [YOCIMX-6545]
    
        - support on i.MX 8 series platform.
        - 8MM/8MP/8MQ support h264/hevc/vp8/vp9 hardware decoding.
        - 8QM/8QXP support h264/hevc/vp8 hardware decoding.
        - support seek, playrate change and resolution change.
        - support video acceleration for online media platform such as Youtube, bilibili.

maskimthedog, 2023-03-29:
> I haven't tried hardware video decode but a commit says it was added:
> 
> commit fe0d3587d905a8fa32a9830d177f97c90352fdc5 Author: Hou Qi [qi.hou@nxp.com](mailto:qi.hou@nxp.com) Date: Thu Sep 15 22:28:13 2022 +0800
> 
> ```
> chromium-ozone-wayland: Add V4L2VDA support on Linux [YOCIMX-6545]
> 
>     - support on i.MX 8 series platform.
>     - 8MM/8MP/8MQ support h264/hevc/vp8/vp9 hardware decoding.
>     - 8QM/8QXP support h264/hevc/vp8 hardware decoding.
>     - support seek, playrate change and resolution change.
>     - support video acceleration for online media platform such as Youtube, bilibili.
> ```

Wow, that's very timely.  I will check it out.  Thank you.

otavio, 2023-04-13:
Is the patchset public?

ambrice, 2023-04-13:
Yes, available at https://github.com/nxp-imx/meta-imx/tree/langdale-6.1.1-1.0.0/meta-sdk/dynamic-layers/chromium-browser-layer/recipes-browser/chromium/

russdill, 2023-06-29:
I've found this to be a somewhat smaller hammer

`diff --git a/ui/gl/BUILD.gn b/ui/gl/BUILD.gn
index 761888bf4a540..0d31ab3b34f35 100644
--- a/ui/gl/BUILD.gn
+++ b/ui/gl/BUILD.gn
@@ -260,6 +260,13 @@ component("gl") {
       if (angle_shared_libvulkan && !is_chromeos) {
         data_deps += [ "//third_party/vulkan-deps/vulkan-loader/src:libvulkan" ]
       }
+      defines += [ "WAYLAND_GBM" ]
+
+      deps += [
+        "//third_party/minigbm",
+        "//ui/gfx:memory_buffer",
+        "//ui/gfx/linux:gbm",
+      ]
     }
   }
 `

russdill, 2023-07-26:
One of the changes chromium 114 brought in was 20f14755 "ui: gbm_wrapper: Ensure to create BOs with
 import-compatible modifier". This change doesn't play well with the imx gbm library. One issue is that the file descriptors to things within the gbm_bo are kept open across a call to gbm_bo_destroy, which causes errors on close. Another issue is that the library doesn't seem to react well with this attempt to use different modifiers within a loop. 9ccd092ec "[m114] Close FDs after use in gbm_wrapper::Device:CereateBufferWithModifiers()" sits on top of this commit, so reverting it is also necessary.

asdasdomar, 2023-07-30:
> One of the changes chromium 114 brought in was 20f14755 "ui: gbm_wrapper: Ensure to create BOs with import-compatible modifier". This change doesn't play well with the imx gbm library. One issue is [...]

Yes, reverting those commits worked for me as well.
Thanks

escherstair, 2023-11-10:
Hello to everybody,
this issue is more than one year old (at the moment).
It's for chromium v103, but then (some months after) v105 and v114 are mentioned.
My question is: can the latest release available at the moment (I imagine v118 or v119) be built without issues for imx8qm/imx8mp?
Are patches required for a successful build?

VaginAY, 2024-03-12:
Confirm for 123 version. After make all patches from meta-browser applicable (or remove such backport).

Pathces from meta-imx doesn't help.

georgekelly1097, 2024-04-19:
It seems NXP attempted to patch this twice. 

First in meta-imx commit https://github.com/nxp-imx/meta-imx/commit/45ae537ebe435c25c53ecb982840deeae2a73fca with the comment " Disable VPU temporarily until the patches are rebased based on newer version of chromium". Then they reverted that in commit 5401bc9. 

Next, they renamed the top level folder "meta-sdk" to "meta-sdk-imx", so the commit history is hard to follow. Now NXP's chromium patches are here: https://github.com/nxp-imx/meta-imx/tree/nanbield-6.6.3-1.0.0/meta-imx-sdk/dynamic-layers/chromium-browser-layer/recipes-browser/chromium/chromium-ozone-wayland. It appears that these 3 patches are the relevant ones:
  0004-Fix-chromium-build-failure.patch \
  0005-Revert-ui-gbm_wrapper-Ensure-to-create-BOs-with-impo.patch \
  0006-Fixed-chromium-crash-after-upgrading.patch \

I've got a build chugging now to see if they resolve this issue