OSSystems/meta-browser #651: No kirkstone branch for firefox

dimonic, 2022-08-11:
What it says on the tin.

kraj, 2022-08-11:
master branch is what we make work on multiple yocto releases, thats why you dont see it.

rwmacleod, 2022-08-31:
@dimonic so as Khem stated, you can use meta-browser/master with poky/meta-oe/...   checkout out to dunfell, kirkstone or master. If that works for you, please close this issue.

shr-project, 2023-02-27:
meta-firefox is pretty much unmaintained for quite long time.

Last major update from 60 to 68.0 was in 2019 by @ashie.

Then I did smaller upgrade to 68.8 and 68.9 in 2020 and then couple more fixes to keep it buildable with newer OE releases (mostly newer rust), but I don't use firefox from meta-browser (and never did) - I just unfortunately included it in my OE build time benchmark (https://github.com/shr-project/test-oe-build-time) long time ago and needed to measure its build time with newer OE releases as well :), never did any runtime testing with it as stated in the PRs.

So if you do use firefox in OE builds (and I've just noticed it being mentioned by you in EasyOS https://bkhome.org/news/202212/easyos-advancing-to-kirkstone-series.html) then patches welcome.