OSSystems/meta-browser #653: Update to  chromium 104.0.5112.101

rwmacleod, 2022-08-23:
This version of chromium has been out for a while so I'll start working on it unless someone is already underway.

rakuco, 2022-08-23:
cc @msisov

rwmacleod, 2022-08-24:
Patches updated, builds on master for qemux86-64, qemuarm64, qemuarm. 
Runs for 64 bit machines with minimal testing (1.1.1.1 , html5test.com, /.  ;-)  )
Building kirkstone overnight, then dunfell tomorrow. Oh I also confirmed that musl patches and it's building now for kirkstone.

BTW, I used quilt to refresh the patches, so there's some noise in the diffs aside from that required to shave the fuzz. @shr-project might care about that.

kraj, 2022-08-24:
@rwmacleod thanks and I see you took care of musl pieces too thats cool. Generally using devtool works best for patch refreshes.

rwmacleod, 2022-08-24:
Pushed a draft: https://github.com/OSSystems/meta-browser/pull/654
@kraj  I'll give devtool a go tomorrow.

msisov, 2022-08-24:
thanks for handling that!

rwmacleod, 2022-08-24:
I have the patches in good shape now. I'm doing a few more builds, and I'll check runtime again. 
Then I'll push what I think is a patch that's ready for merge.

Next time I do a chromium update, I plan to be less verbose and likely just send a single PR. _Some_ _people_ were **excited** about this update so I wanted feedback along the way.

rwmacleod, 2022-08-25:
Update pushed, ready for more testing and/or merge.

rwmacleod, 2022-08-25:
@rakuco @msisov  
Any comments? It's been 19 hours! ;-) It would be good to know what to expect here: hours, days, weeks?
This update does fix several CVEs, one of which may have a serious risk:

> "_Google is aware that an exploit for CVE-2022-2856 exists in the wild._"

https://chromereleases.googleblog.com/2022/08/stable-channel-update-for-desktop_16.html

Let me know if I can do anything to help move things along. 
Thanks,
../Randy

rwmacleod, 2022-08-31:
The update is merged.

closed 2022-08-31 by rwmacleod