OSSystems/meta-browser #657: chromium-ozone-wayland: Evaluate if 0001-ozone-add-va-api-support-to-wayland.patch needs to be readded

rakuco, 2022-08-26:
#648 removed `0001-ozone-add-va-api-support-to-wayland.patch` because neither @rwmacleod nor myself knew how to rebase the patch on top of M103.

We need to figure out if the patch needs to be rebased and readded or whether its functionality has been implemented upstream differently.

gustavqmatic, 2022-09-01:
We have tested a few different combinations of yocto release and Chromium releases and we don't see any trace calls to VAAPI with 103 or 104 but we get them with 101 (version before the patch was removed). The "Video Acceleration Information" list on chromium://gpu is also empty. So something seems to be turned off without the patch. Can it be that this part of the patch is still needed?
```
diff --git a/ui/ozone/platform/wayland/ozone_platform_wayland.cc b/ui/ozone/platform/wayland/ozone_platform_wayland.cc
index 8d82c2d..6bb1670 100644
--- a/ui/ozone/platform/wayland/ozone_platform_wayland.cc
+++ b/ui/ozone/platform/wayland/ozone_platform_wayland.cc
@@ -291,6 +291,9 @@ class OzonePlatformWayland : public OzonePlatform,
       properties->supports_global_screen_coordinates =
           features::IsWaylandScreenCoordinatesEnabled();
 
+      // Let the media know this platform supports va-api.
+      properties->supports_vaapi = true;
+
       initialised = true;
 }
```

But this is just a guess, I'm no expert on this code.

msisov, 2022-09-01:
@gustavqmatic, maybe yes. I'll work on that in the following days along with update to 105.0.5195.52

msisov, 2022-09-15:
There were some refactorings and it was decided to enable vaapi for ozone/wayland/x11 in the upstream. A patch is on its way https://chromium-review.googlesource.com/c/chromium/src/+/3646633?usp=dashboard