OSSystems/meta-browser #658: IMX6 Dual not getting Hardware Accelerated also Stack Smashing - Chromium 103.0.5060.129

WillOwnU, 2022-08-30:
Getting some EGL errors and Stack smashing errors when starting Chromium. Also gpu is not Hardware Accelerated.

Hardware:
- imx6 Dual with GC2000, GC355 
Software:
- Clang: Branch dunfell-clang12 also tryed to dunfell Branch
- Chromium 103.0.5060.129
- Yocto dunfell-17.0-28
- Linux Kernel:  5.10.104
- Weston 8.0.0

What did i already tried?
- Clang Branches
- Chromium 95.0.4638.69(just a random chromium version to see if that works)
- Deleted libEGL.so and libGLESv2.so from usr/lib/chromium

First of all Weston log:
[weston.txt](https://github.com/OSSystems/meta-browser/files/9450792/weston.txt)

**chrome://GPU Information **
Graphics Feature Status
Canvas: Software only. Hardware acceleration disabled
Canvas out-of-process rasterization: Disabled
Direct Rendering Display Compositor: Disabled
Compositing: Software only. Hardware acceleration disabled
Multiple Raster Threads: Disabled
OpenGL: Disabled
Rasterization: Software only. Hardware acceleration disabled
Raw Draw: Disabled
Skia Renderer: Enabled
Video Decode: Software only. Hardware acceleration disabled
Video Encode: Software only. Hardware acceleration disabled
Vulkan: Disabled
WebGL: Disabled
WebGL2: Disabled
WebGPU: Disabled
Driver Bug Workarounds
clear_uniforms_before_first_program_use
enable_webgl_timer_query_extensions
exit_on_context_lost
disabled_extension_GL_KHR_blend_equation_advanced
disabled_extension_GL_KHR_blend_equation_advanced_coherent
disabled_extension_GL_MESA_framebuffer_flip_y
Problems Detected
WebGPU has been disabled via blocklist or the command line.
Disabled Features: webgpu
Raster is using a single thread.
Disabled Features: multiple_raster_threads
Accelerated video encode has been disabled, either via blocklist, about:flags or the command line.
Disabled Features: video_encode
Accelerated video decode has been disabled, either via blocklist, about:flags or the command line.
Disabled Features: video_decode
Gpu compositing has been disabled, either via blocklist, about:flags or the command line. The browser will fall back to software compositing and hardware acceleration will be unavailable.
Disabled Features: gpu_compositing
GPU process was unable to boot: GPU process crashed too many times with SwiftShader.
Disabled Features: all
Clear uniforms before first program use on all platforms: 124764, 349137
Applied Workarounds: clear_uniforms_before_first_program_use
Disable KHR_blend_equation_advanced until cc shaders are updated: 661715
Applied Workarounds: disable(GL_KHR_blend_equation_advanced), disable(GL_KHR_blend_equation_advanced_coherent)
Expose WebGL's disjoint_timer_query extensions on platforms with site isolation: 808744, 870491
Applied Workarounds: enable_webgl_timer_query_extensions
Some drivers can't recover after OUT_OF_MEM and context lost: 893177
Applied Workarounds: exit_on_context_lost
Disable GL_MESA_framebuffer_flip_y for desktop GL: 964010
Applied Workarounds: disable(GL_MESA_framebuffer_flip_y)
Version Information
Data exported	2022-08-30T08:36:02.671Z
Chrome version	Chrome/103.0.5060.129
Operating system	Linux 5.10.104-guf
[... 340 lines cut ...]
[2529:2529:0830/082932.305625:ERROR:shared_context_state.cc(543)] : Failed to make current since context is marked as lost
[2529:2529:0830/082932.306741:ERROR:skia_output_surface_impl_on_gpu.cc(1703)] : Failed to make current.
[2529:2529:0830/082932.309114:ERROR:gl_surface_egl.cc(822)] : EGL Driver message (Error) eglCreateContext: dri2_create_context
[2529:2529:0830/082932.321717:ERROR:shared_image_stub.cc(519)] : SharedImageStub: context already lost
[2529:2529:0830/082932.327491:ERROR:shared_image_stub.cc(519)] : SharedImageStub: context already lost
[2529:2529:0830/082932.382542:ERROR:gl_surface_egl.cc(822)] : EGL Driver message (Error) eglCreateContext: dri2_create_context
[2529:2529:0830/082932.712841:ERROR:gbm_wrapper.cc(292)] : Failed to export buffer to dma_buf: No such file or directory (2)
[2529:2529:0830/082932.717252:ERROR:gbm_pixmap_wayland.cc(75)] : Cannot create bo with format= RGBA_8888 and usage=SCANOUT
[2529:2529:0830/082932.719159:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(198)] : Failed to create pixmap 1280x1024, RGBA_8888, usage SCANOUT
[2529:2529:0830/082932.739327:ERROR:gbm_wrapper.cc(292)] : Failed to export buffer to dma_buf: No such file or directory (2)
[2529:2529:0830/082932.745173:ERROR:gbm_pixmap_wayland.cc(75)] : Cannot create bo with format= RGBA_8888 and usage=GPU_READ
[2529:2529:0830/082932.747160:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(198)] : Failed to create pixmap 1280x1024, RGBA_8888, usage GPU_READ
[2529:2529:0830/082932.748899:ERROR:shared_image_backing_factory_gl_image.cc(355)] : CreateSharedImage: Failed to create bindable image
[2529:2529:0830/082932.750562:ERROR:shared_image_factory.cc(770)] : CreateSharedImage: could not create backing.
GpuProcessHost: The GPU process crashed!
[2543:2543:0830/082933.154245:ERROR:angle_platform_impl.cc(43)] : Display.cpp:1006 (initialize): ANGLE Display::initialize error 0: Internal Vulkan error (-3): Initialization of an object could not be completed for implementation-specific reasons, in ../../third_party/angle/src/libANGLE/renderer/vulkan/RendererVk.cpp, initialize:1297.
[2543:2543:0830/082933.155398:ERROR:gl_surface_egl.cc(822)] : EGL Driver message (Critical) eglInitialize: Internal Vulkan error (-3): Initialization of an object could not be completed for implementation-specific reasons, in ../../third_party/angle/src/libANGLE/renderer/vulkan/RendererVk.cpp, initialize:1297.
[2543:2543:0830/082933.156123:ERROR:gl_surface_egl.cc(1352)] : eglInitialize SwANGLE failed with error EGL_NOT_INITIALIZED
[2543:2543:0830/082933.156814:ERROR:gl_ozone_egl.cc(21)] : GLSurfaceEGL::InitializeOneOff failed.
[2543:2543:0830/082933.201038:ERROR:viz_main_impl.cc(186)] : Exiting GPU process due to errors during initialization
GpuProcessHost: The GPU process exited normally. Everything is okay.
[2551:2551:0830/082933.329071:ERROR:gpu_init.cc(486)] : Passthrough is not supported, GL is disabled, ANGLE is
[2551:2551:0830/082933.333229:WARNING:sandbox_linux.cc(376)] : InitializeSandbox() called with multiple threads in process gpu-process.

Maybe somone has any idea what the problem is.

WillOwnU, 2022-09-08:
Not really found the Problem but build Chromium 91.0.4472.77-r0 Version worked. Hardware is now accelerated and WebGL works now. Probably some driver are not longer Supported by newer Chromium versions.

closed 2022-09-08 by WillOwnU

bmx666, 2022-09-16:
Hi @WillOwnU this version works fine on kirkstone branch, but newer version breaks GPU acceleration - https://github.com/OSSystems/meta-browser/issues/663