OSSystems/meta-browser #659: third_party/swiftshader/include/vulkan/vulkan.h:41:10: fatal error: 'wayland-client.h' file not found

rwmacleod, 2022-08-31:
In a non-poky distro using the kirkstone branch, without wayland or vulkan included in DISTRO_FEATURES,
I'm seeing:

>  FAILED: obj/third_party/swiftshader/src/Vulkan/swiftshader_libvulkan/VkDescriptorSet.o
> | x86_64-wrs-linux-clang++ -target x86_64-wrs-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-seco
> | In file included from ../../third_party/swiftshader/src/Vulkan/VkDescriptorSet.cpp:15:
> | In file included from ../../third_party/swiftshader/src/Vulkan/VkDescriptorSet.hpp:18:
> | In file included from ../../third_party/swiftshader/src/Vulkan/VkObject.hpp:22:
> | In file included from ../../third_party/swiftshader/include/vulkan/vk_icd.h:26:
> | ../../third_party/swiftshader/include/vulkan/vulkan.h:41:10: fatal error: 'wayland-client.h' file not found
> | #include <wayland-client.h>
> |          ^~~~~~~~~~~~~~~~~~
> | 1 error generated.

I'm adding wayland to DISTRO_FEATURES to confirm that that is the significant difference from all of the successful builds I did with poky which has:
   POKY_DEFAULT_DISTRO_FEATURES = "largefile opengl ptest multiarch wayland vulkan"
 in all of dunfell, kirkstone, master.

rwmacleod, 2022-08-31:
Adding wayland to DISTRO_FEATRES seems to have resolved the build error (at ~85% now, failed earlier without) but I'd like to see why it's required now and hasn't been required in earlier releases. We could add a REQUIRED_DISTRO_FEATURE to the chromium-x11 recipe but I'm reluctant to do that for/to the old branches.

rakuco, 2022-09-15:
Do you mean this is a chromium-x11 build that's failing because the Wayland headers aren't present during the build?

rakuco, 2022-09-19:
cc @msisov do you know what the situation is upstream? Do we just need to depend on the headers for the build?

msisov, 2022-09-19:
Something defines this https://source.chromium.org/chromium/chromium/src/+/main:third_party/swiftshader/include/vulkan/vulkan.h;l=40?q=VK_USE_PLATFORM_WAYLAND_KHR .

Oquirella, 2022-10-18:
Any update on this?

rakuco, 2022-11-18:
@rwmacleod @Oquirella can you check if #678 that I just merged fixes the issue for you?

rwmacleod, 2022-11-18:
I will check but not until next week at the earliest.