OSSystems/meta-browser #660: Update to chromium 105.0.5195.52

rwmacleod, 2022-08-31:
https://chromereleases.googleblog.com/2022/08/stable-channel-update-for-desktop_16.html

Is anyone able to do this update ? I'm busy with other work and then on vacation until Oct 1st.

rakuco, 2022-08-31:
maybe @msisov?

msisov, 2022-09-01:
I will

msisov, 2022-09-09:
After solving some problems with Wayland, langdale and kirkstone branches passed. However, dunfell fails with the following error -

```
| [581/56619] python3 ../../tools/polymer/html_to_wrapper.py --in_folder gen/chrome/browser/resources/history/preprocessed --out_folder gen/chrome/browser/resources/history/preprocessed --in_files app.html history_item.html history_list.html history_toolbar.html router.html side_bar.html synced_device_card.html synced_device_manager.html shared_icons.html --minify
| FAILED: gen/chrome/browser/resources/history/preprocessed/app.html.ts gen/chrome/browser/resources/history/preprocessed/history_item.html.ts gen/chrome/browser/resources/history/preprocessed/history_list.html.ts gen/chrome/browser/resources/history/preprocessed/history_toolbar.html.ts gen/chrome/browser/resources/history/preprocessed/router.html.ts gen/chrome/browser/resources/history/preprocessed/side_bar.html.ts gen/chrome/browser/resources/history/preprocessed/synced_device_card.html.ts gen/chrome/browser/resources/history/preprocessed/synced_device_manager.html.ts gen/chrome/browser/resources/history/preprocessed/shared_icons.html.ts
| python3 ../../tools/polymer/html_to_wrapper.py --in_folder gen/chrome/browser/resources/history/preprocessed --out_folder gen/chrome/browser/resources/history/preprocessed --in_files app.html history_item.html history_list.html history_toolbar.html router.html side_bar.html synced_device_card.html synced_device_manager.html shared_icons.html --minify
| Traceback (most recent call last):
|   File "../../tools/polymer/html_to_wrapper.py", line 132, in <module>
|     main(sys.argv[1:])
|   File "../../tools/polymer/html_to_wrapper.py", line 99, in main
|     raise err
|   File "../../tools/polymer/html_to_wrapper.py", line 94, in main
|     node.RunNode(
|   File "/home/msisov/code/yocto/dunfell/builds/ci-rpi3-x11/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-x11/105.0.5195.102-r0/chromium-105.0.5195.102/out/Release/../../third_party/node/node.py", line 34, in RunNode
|     raise RuntimeError('Command \'%s\' failed\n%s' % (' '.join(cmd), err))
| RuntimeError: Command '/home/msisov/code/yocto/dunfell/builds/ci-rpi3-x11/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-x11/105.0.5195.102-r0/chromium-105.0.5195.102/out/Release/../../third_party/node/linux/node-linux-x64/bin/node ../../tools/polymer/html_minifier.js /home/msisov/code/yocto/dunfell/builds/ci-rpi3-x11/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-x11/105.0.5195.102-r0/chromium-105.0.5195.102/out/Release/gen/chrome/browser/resources/history/preprocessed /home/msisov/code/yocto/dunfell/builds/ci-rpi3-x11/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-x11/105.0.5195.102-r0/chromium-105.0.5195.102/out/Release/gen/chrome/browser/resources/history/preprocessed/tmp6xmlibk5 app.html history_item.html history_list.html history_toolbar.html router.html side_bar.html synced_device_card.html synced_device_manager.html shared_icons.html' failed
| internal/modules/cjs/loader.js:818
|   throw err;
|   ^
| 
| Error: Cannot find module 'fs/promises'
```

It seems like a newer version of nodejs is required. Can somebody please remind me how to request a specific version of a package for a recipe?

rwmacleod, 2022-09-09:
Maybe : https://docs.yoctoproject.org/dev/ref-manual/variables.html?highlight=preferred_version#term-PREFERRED_VERSION
( that's usually done in as distro or layer preference rather than in a recipe )

Btw, are you or can you switch to 105.0.5195.102  to deal with yet another chromium CVE that is known to be being exploited in the wild?  https://chromereleases.googleblog.com/2022/09/stable-channel-update-for-desktop.html

msisov, 2022-09-09:
that may work. Thanks. 

yes, I'm using that version. 

-- 
Best Regards,
Maksim Sisov
* Usual work time - 08:00-16:00 (EEST). 

Links:
------
[1]
https://github.com/OSSystems/meta-browser/issues/660#issuecomment-1242074645
[2]
https://github.com/

msisov, 2022-09-12:
fs/promises is not part of node v12 that dunfell uses according to https://nodejs.org/api/fs.html#promises-api.

https://stackoverflow.com/questions/70041947/import-filehandle-across-node-js-versions/70042193#70042193 suggests const fileHandle = require("fs").promises;

rwmacleod, 2022-09-12:
@msisov  I don't see an 105 update PR. Did you intend to close this issue?

PR #664 (2022-09-14): chromium: update to 105.0.5195.102
```
Release notes -
https://chromereleases.googleblog.com/2022/09/stable-channel-update-for-desktop.html

License changes:
----------------

Added licenses:
  * third_party/crubit/LICENSE
  - https://chromium-review.googlesource.com/c/chromium/src/+/3756853

  * third_party/swift-toolchain/LICENSE
  - https://chromium-review.googlesource.com/c/chromium/src/+/3722369

Removed licenses: (none)

Updated licenses:
  * third_party/libevent/LICENSE
  * (The license library's folder was moved from //base to //third_party)
  - https://chromium-review.googlesource.com [... cut]
commit 0f11e5ed (2024-06-13): chromium: update to 105.0.5195.102

Release notes -
https://chromereleases.googleblog.com/2022/09/stable-channel-update-for-desktop.html

License changes:
----------------

Added licenses:
  * third_party/crubit/LICENSE
  - https://chromium-review.googlesource.com/c/chromium/src/+/3756853

  * third_party/swift-toolchain/LICENSE
  - https://chromium-review.googlesource.com/c/chromium/src/+/3722369

Removed licenses: (none)

Updated licenses:
  * third_party/libevent/LICENSE
  * (The license library's folder was moved from //base to //third_party)
  - https://chromium-review.googlesource.com/c/ [... cut]
closed 2022-09-14 by msisov