OSSystems/meta-browser #663: chromium GPU acceleration doesn't work on imx6

bmx666, 2022-09-12:
I created issue on [meta-freescale](https://github.com/Freescale/meta-freescale) - https://github.com/Freescale/meta-freescale/issues/1203
But our team tested build and found issue in [new version of Chromium](https://github.com/OSSystems/meta-browser/commit/b8e658b5a4da60a0d297fb7c2cdb7c7c251ec8bf)

bmx666, 2022-09-16:
same issue with new version - [105.0.5195.102](https://github.com/OSSystems/meta-browser/commit/0f11e5ed0da3a622bbd7952400e7470815ded96a)

bmx666, 2022-09-18:
`chrome://gpu` doesn't display`Skia Renderer: Enabled` anymore

bmx666, 2022-09-18:
Since **104.0.5112.101** build configuration file `out/Release/args.gn` include by default `angle_use_wayland = false`. After change it back to `angle_use_wayland = true` build with clang-15 returns an error related to Chromium subrepo https://github.com/abseil/abseil-cpp and not yet included patches in Chromium version **105.0.5195.102** - https://github.com/abseil/abseil-cpp/commit/0c92330442d6b1be934e0407115c8084250ef347 and https://github.com/abseil/abseil-cpp/commit/cfe27e79cfcbefb2b4479e04f80cbb299bc46965 . Also Chromium subrepo [angle](https://chromium.googlesource.com/angle/angle/) has error during build of wayland, patch to fix that can be found here - https://bugs.chromium.org/p/angleproject/issues/detail?id=7582#c3

MastaG, 2022-09-19:
Thanks for the fixes!

bmx666, 2022-09-20:
I successfully built Chromium 108.0.5309.0, only few fixes are required since 105.0.5195.102:
- remove patch `0012-Build-X11-parts-only-for-ozone_platform_x11.patch` because it already included
- update md5 hash of licenses
- patch `third_party/wayland/src` from `upstream/1.19.0` to `upstream/1.20.0` otherwise a lot of errors during compilation. Patch can be generated from [wayland repo](https://chromium.googlesource.com/external/anongit.freedesktop.org/git/wayland/wayland/) and [angle patch](https://bugs.chromium.org/p/angleproject/issues/detail?id=7582#c3) is not required  anymore
- `GN_ARGS += " use_qt=false use_system_libffi=true "`

My major issue - GPU acceleration still doesn't work

bmx666, 2022-09-21:
Tested Chromium 104.0.5112.0 - GPU acceleration doesn't work.
A lot of changes between 103 and 104...

MastaG, 2022-09-23:
Well I did a few tests on my ODroid XU4 which is armv7 (32bits).
It uses the Mesa Panforst driver and runs on Gnome 43 wayland.
For me the problem is that I can only run it with "--in-process-gpu" otherwise it seems to segfault.

I've rebuilt chromium-ozone-wayland 105.0.5195.102 with the angle patch: https://bugs.chromium.org/p/angleproject/issues/detail?id=7582#c3 and ```angle_use_wayland = true```
But that doesn't seem to make a difference, what does Angle do anyways? It seems to make the binary a lot bigger.

These are the parameters for making it run on my GPU:
```--use-gl=egl --ozone-platform=wayland --in-process-gpu --reduce-gpu-sandbox --gpu-sandbox-start-early --ignore-gpu-blocklist --ignore-gpu-blacklist --enable-gpu-rasterization --enable-features=RawDraw,CanvasOopRasterization,WebRTCPipeWireCapturer```

bmx666, 2022-09-23:
Hi @MastaG

I'm launching chromium with flags `--use-gl=egl --ozone-platform=wayland --enable-gpu --enable-gpu-rasterization --enable-gpu-memory-buffer-video-frames --enable-accelerated-2d-canvas --enable-accelerated-vpx-decode --enable-native-gpu-memory-buffers --enable-zero-copy --gpu-no-context-lost --ignore-gpu-blacklist --in-process-gpu`

I can try to add flags `--reduce-gpu-sandbox --gpu-sandbox-start-early --ignore-gpu-blocklist --enable-gpu-rasterization --enable-features=RawDraw,CanvasOopRasterization,WebRTCPipeWireCapturer` and check with version 105

bmx666, 2022-09-23:
`what does Angle do anyways?`

About ANGLE ->
https://chromium.googlesource.com/angle/angle/+/master/README.md
https://chromium.googlesource.com/angle/angle/+/refs/heads/main/doc/BuildingAngleForChromiumDevelopment.md

bmx666, 2022-09-23:
I build version 103, but with DEPS from version 104

```
-  'skia_revision': 'b301ff025004c9cd82816c86c547588e6c24b466',
+  'skia_revision': '5a4dbb2e97302f7e574f0ba962ac4d8b3a62a36e',

-  'angle_revision': '53e1711046b815dfff3321c3027f190c9b0b899b',
+  'angle_revision': '4a65a669e11bd7bfa9d77cbf7001836379ec29b5',

-  'swiftshader_revision': 'f1c2c0b0728152d7ca24472e696e07e957e6d09a',
+  'swiftshader_revision': 'd68d367047f562e0e0716fd72b9a2bc3e2c8f4c0',
```

And GPU acceleration still works fine.

The next step will be to replace other components

```
-  'src/third_party/glfw/src':
-    Var('chromium_git') + '/external/github.com/glfw/glfw.git@' +  '94773111300fee0453844a4c9407af7e880b4df8',
-

-  'src/third_party/vulkan-deps': '{chromium_git}/vulkan-deps@c6a60f3cc711e0758ebb58d8d1db9f97b5973ccc',
+  'src/third_party/vulkan-deps': '{chromium_git}/vulkan-deps@12989fc8d7d273b5b95ef988d37469738eb9c39b',

   'src/third_party/webgl/src':
-    Var('chromium_git') + '/external/khronosgroup/webgl.git' + '@' + 'cf04aebdf9b53bb2853f22a81465688daf879ec6',
+    Var('chromium_git') + '/external/khronosgroup/webgl.git' + '@' + 'bb289ce3cb15bbabd42fdcb01439367846d9069d',

   'src/third_party/webgpu-cts/src':
-    Var('chromium_git') + '/external/github.com/gpuweb/cts.git' + '@' + 'b32ac31a0eab7386807f01c7cb8d77a6e1104c23',
+    Var('chromium_git') + '/external/github.com/gpuweb/cts.git' + '@' + '84183b561b7f44b2eae6a5f268d5085accb0ad7a',

   'src-internal': {
-    'url': 'https://chrome-internal.googlesource.com/chrome/src-internal.git@2abd00903939dc3d89433d476402b4ebb6ee982d',
+    'url': 'https://chrome-internal.googlesource.com/chrome/src-internal.git@6956112deae1e1529cc3fe8eb3112e60c5104b58',
```

bmx666, 2022-09-23:
since Chromium `108.0.5315.0` DEPS wayland was updated to version `upstream/1.20.0`

bmx666, 2022-09-27:
on i.MX6 `Chromium 103.0.5060.129` only these extra flags enables most of all HW acceleration features

`--in-process-gpu --enable-features=RawDraw,CanvasOopRasterization --enable-gpu-rasterization --num-raster-threads=4 --enable-zero-copy`

[image]

MastaG, 2022-10-07:
Try adding:
```--reduce-gpu-sandbox --gpu-sandbox-start-early --ignore-gpu-blocklist --ignore-gpu-blacklist ```

bmx666, 2022-10-07:
@MastaG no luck, I've tried all the argument options since version 104 and later... Chromium doesn't have GPU rasterization on imx6, everything is slow and laggy...

bmx666, 2023-02-07:
I ported customer's firmware on Qt5 kiosk browser, it has no issue with HW acceleration at all like Chromium has on i.MX6

closed 2023-02-07 by bmx666

Rutvij-dev, 2023-07-21:
Hi Folks,

Not sure weather I can reopen it or not, but wants to understand the **status on it** ?

The reporter (@bmx666 ) had to switch the solution (Framework from Chromium to QT based), but the problem in Chromium is still persist I believe !?

If this is the case, then Chromium issue is not resolved, please correct me if I am wrong.

In this case please suggest the optimized Chromium with HW Chromium If anyone have achieved so far ?

--
BR

bmx666, 2023-07-21:
Hi @Rutvij-dev 

Unfortunately, I don't have access to that i.MX6 hardware and Yocto build anymore. It was a short-term customer project, and I was looking for a stable solution. If QT WebKit supports hardware acceleration, then it means something might be broken in Chromium since version 104.

MaxIhlenfeldt, 2023-07-24:
Has anyone tested the current version to see if HW acceleration is still not available? If it still doesn't work, we can re-open this.

bmx666, 2023-07-24:
I think we could wait results on the latest chromium 118 from reporter of issue #735 😉