OSSystems/meta-browser #668: Yocto compilation error (fatal error: 'atomic' file not found)

bitglitcher, 2022-10-18:
Hello, I'm trying to add meta-browser to our distro. I added the meta-browser. It fails while compiling, the issue is that it seems it can't find atomic library? A whole bunch of files have this issue.

I have this line of code defined on my `layer.conf` file. Which solve an issue with `gn-native`. I got no idea of that has anything to do with this issue.
```
# There are two versions of gn-native available and the wrong version is getting picked up.
# ./meta-qt6/recipes-qt/qt6/gn-native_git.bb
# ./meta-browser/meta-chromium/recipes-browser/chromium/gn-native_96.0.4664.110.bb
BBMASK += "/meta-browser/meta-chromium/recipes-browser/chromium/gn-native_.*\.bb"
```
Also I added line to `CORE_IMAGE_EXTRA_INSTALL`
```
	chromium-ozone-wayland \
```
This is how the errors looks like.
```
| In file included from ../../third_party/skia/src/opts/SkOpts_ssse3.cpp:8:
| In file included from ../../third_party/skia/src/core/SkOpts.h:11: 
| In file included from ../../third_party/skia/include/core/SkTypes.h:192:
| In file included from ../../third_party/skia/../../skia/config/SkUserConfig.h:125: 
| ../../skia/ext/skia_histogram.h:10:10: fatal error: 'atomic' file not found  
| #include <atomic>   
|          ^~~~~~~~    
| 1 error generated. 
```
These are the revisions that I have on my Manifest file.
meta-browser = `revision="d25d8ee98a656b2534d0eec6138ef264529fab4f"` 
meta-clang = `revision="85d956d95401479ca666139e31f662f60c156d5f"`

I don't know if it's an issue from meta-browser itself, or I'm missing something. Thanks.

rakuco, 2022-10-18:
Note that that meta-browser revision is fairly old and you're strongly encouraged to use the latest Chromium release to avoid all sorts of security issues.

With that said, can you post the compiler invocation that failed? It is not clear if this is an issue building target code or the bits that are built for and run on the host.

bitglitcher, 2022-10-18:
Hello, here is the compiler invocation. There are similar errors but they al boil down to not finding the CPP libs.
`
| [63/38214] clang++ -MMD -MF yocto_native/obj/skia/skia_core_and_effects/SkGlyphRun.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DSK_UNTIL_CRBUG_1187654_IS_FIXED -DSK_WIN_FONTMGR_NO_SIMULATIONS -DSK_CODEC_DECODES_PNG -DSK_CODEC_DECODES_WEBP -DSK_ENCODE_PNG -DSK_ENCODE_WEBP -DSK_ENABLE_SKSL -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_GL -DSK_CODEC_DECODES_JPEG -DSK_ENCODE_JPEG -DSK_HAS_WUFFS_LIBRARY -DSK_VULKAN=1 -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DSK_FREETYPE_MINIMUM_RUNTIME_VERSION=\(\(\(FREETYPE_MAJOR\)\ \*\ 0x01000000\)\ \|\ \(\(FREETYPE_MINOR\)\ \*\ 0x00010000\)\ \|\ \(\(FREETYPE_PATCH\)\ \*\ 0x00000100\)\) -DSKIA_IMPLEMENTATION=1 -DSK_GAMMA_EXPONENT=1.2 -DSK_GAMMA_CONTRAST=0.2 -DSK_DEFAULT_FONT_CACHE_LIMIT=20971520 -I../.. -Iyocto_native/gen -I../../third_party/skia -I../../third_party/wuffs/src/release/c -I../../third_party/vulkan/include -I../../third_party/vulkan-deps/vulkan-headers/src/include -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -m64 -march=x86-64 -msse3 -Xclang -fdebug-compilation-dir -Xclang . -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-psabi -Wno-ignored-pragma-optimize -Wno-builtin-assume-aligned-alignment -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-bitwise-instead-of-logical -std=c++17 -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -std=c++17 -isystem/home/bherrera/winYocto5.15/build/tmp/work/armv8a-mx8m-poky-linux/chromium-ozone-wayland/96.0.4664.110-r0/recipe-sysroot-native/usr/include -isysroot=/home/bherrera/winYocto5.15/build/tmp/work/armv8a-mx8m-poky-linux/chromium-ozone-wayland/96.0.4664.110-r0/recipe-sysroot-native -stdlib=libc++ -isystem/home/bherrera/winYocto5.15/build/tmp/work/armv8a-mx8m-poky-linux/chromium-ozone-wayland/96.0.4664.110-r0/recipe-sysroot-native/usr/include -isysroot=/home/bherrera/winYocto5.15/build/tmp/work/armv8a-mx8m-poky-linux/chromium-ozone-wayland/96.0.4664.110-r0/recipe-sysroot-native -stdlib=libc++ -O2 -pipe -c ../../third_party/skia/src/core/SkGlyphRun.cpp -o yocto_native/obj/skia/skia_core_and_effects/SkGlyphRun.o                                                                                                                  | FAILED: yocto_native/obj/skia/skia_core_and_effects/SkGlyphRun.o                                                                                                                                                | clang++ -MMD -MF yocto_native/obj/skia/skia_core_and_effects/SkGlyphRun.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DSK_UNTIL_CRBUG_1187654_IS_FIXED -DSK_WIN_FONTMGR_NO_SIMULATIONS -DSK_CODEC_DECODES_PNG -DSK_CODEC_DECODES_WEBP -DSK_ENCODE_PNG -DSK_ENCODE_WEBP -DSK_ENABLE_SKSL -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_GL -DSK_CODEC_DECODES_JPEG -DSK_ENCODE_JPEG -DSK_HAS_WUFFS_LIBRARY -DSK_VULKAN=1 -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DSK_FREETYPE_MINIMUM_RUNTIME_VERSION=\(\(\(FREETYPE_MAJOR\)\ \*\ 0x01000000\)\ \|\ \(\(FREETYPE_MINOR\)\ \*\ 0x00010000\)\ \|\ \(\(FREETYPE_PATCH\)\ \*\ 0x00000100\)\) -DSKIA_IMPLEMENTATION=1 -DSK_GAMMA_EXPONENT=1.2 -DSK_GAMMA_CONTRAST=0.2 -DSK_DEFAULT_FONT_CACHE_LIMIT=20971520 -I../.. -Iyocto_native/gen -I../../third_party/skia -I../../third_party/wuffs/src/release/c -I../../third_party/vulkan/include -I../../third_party/vulkan-deps/vulkan-headers/src/include -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -m64 -march=x86-64 -msse3 -Xclang -fdebug-compilation-dir -Xclang . -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno [... cut ...]                                                                               |          ^~~~~~~~~~~~                                                                                                                                                                                           | 1 error generated.        
                                                                                                                                                                                      `

kraj, 2022-10-18:
whats the clang compiler version in use here ?

bitglitcher, 2022-10-18:
I'm using 
```
revision="85d956d95401479ca666139e31f662f60c156d5f"
```

bitglitcher, 2022-10-18:
Updating my revision to the latest commit yields a different error. Seems to be related to the source code and not my configuration. I wonder if I need to update the clang version maybe.
`
ERROR: chromium-ozone-wayland-105.0.5195.102-r0 do_patch: Applying patch '0001-Disable-dri-for-imx-gpu.patch' on target directory '/home/bherrera/winYocto5.15/build/tmp/work/armv8a-mx8m-poky-linux/chromium-ozone-wayland/105.0.5195.102-r0/chromium-105.0.5195.102'                                                                                                                                                              CmdError('quilt --quiltrc /home/bherrera/winYocto5.15/build/tmp/work/armv8a-mx8m-poky-linux/chromium-ozone-wayland/105.0.5195.102-r0/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0001-Disable-dri-for-imx-gpu.patch                                                                                                                                                                                         patching file content/gpu/BUILD.gn                                                                                                                                                                                Hunk #1 succeeded at 8 (offset 1 line).                                                                                                                                                                           Hunk #2 FAILED at 141.                                                                                                                                                                                            1 out of 2 hunks FAILED -- rejects in file content/gpu/BUILD.gn                                                                                                                                                   Patch 0001-Disable-dri-for-imx-gpu.patch does not apply (enforce with -f)                                                                                                                                                                                                                                                                                                                                                           stderr: ')                                                                                                                                                                                                        ERROR: Logfile of failure stored in: /home/bherrera/winYocto5.15/build/tmp/work/armv8a-mx8m-poky-linux/chromium-ozone-wayland/105.0.5195.102-r0/temp/log.do_patch.3173797                                         ERROR: Task (/home/bherrera/winYocto5.15/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_105.0.5195.102.bb:do_patch) failed with exit code '1'                                 
`

kraj, 2022-10-18:
> 

that should yield clang 14.0.3, I wonder where its pulling its c++ headers from, it should get them from libcxx-native install in
native-sysroot.

what is your build host OS ?
perhaps try it on a different ( newer ) host and see if it helps.

kraj, 2022-10-18:
> 

`0001-Disable-dri-for-imx-gpu.patch` seems imx specific patch which perhaps is being added via another layer like meta-freescale so if you bump revs for meta-chromium and meta-clang you have to ensure the patches being applied to chromium from other layers work too.

bitglitcher, 2022-10-18:
I'm using an Ubuntu 20.04.3 LTS build server as host.

bitglitcher, 2022-10-18:
It's weird that it can't find the header files from the standard library, giving that it's clearly being specified in the compiler invocation. Maybe the `BBMASK` line to make qtwebengine work has something to do with it?
```
BBMASK += "/meta-browser/meta-chromium/recipes-browser/chromium/gn-native_.*\.bb"
```

bitglitcher, 2022-10-18:
Do you guys know of any project I can use as a reference point?

rakuco, 2022-10-18:
Perhaps start with something more basic, like adding Chromium to core-image-sato and building for e.g. x86-64? We've got many bug reports with iMX in them, so starting from something as simple as possible would help rule out something wrong in meta-clang in your setup, for example.

bitglitcher, 2022-10-18:
Basically, who knows why it was failing to compile. But I added Chromium through `local.conf`
```
IMAGE_INSTALL:append += \
    "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', ' chromium-ozone-wayland', \
        bb.utils.contains('DISTRO_FEATURES',     'x11', ' chromium-x11', \
                                                        '', d), d)}"
```
This worked well. Before I was adding `chromium-ozone-wayland ` to `CORE_IMAGE_EXTRA_INSTALL` inside of my `image.bb` file. Somehow that makes it not compile. Weird honestly. But it works now lol.

closed 2022-10-18 by bitglitcher