OSSystems/meta-browser #671: chromium x11 105.0.5195.102 cannot be started on stm32

nandra, 2022-10-25:
Hi,

I'm building chromium-x11 and when run on target platform I'm getting:
root@stm32mp1:~# chromium 
/usr/bin/chromium: error while loading shared libraries: libgbm.so: cannot open shared object file: No such file or directory

also to be able to build I need to add to bbappend file:
```
DEPENDS:append = " libxshmfence"
```

Any ideas please? Thanks.

rakuco, 2022-10-25:
@msivov do you know if the dependencies have changed recently? Do we need to pull in something that installs libgbm.so?

nandra, 2022-10-26:
FYI: tried chromium-ozone-wayland with weston manager and chromium can be started  fine.

nandra, 2022-11-02:
@rakuco seems to be related to stm32 layers only (build and run fine on sunxi though). Any ideas what can be checked and how libgbm is added or so? Thanks.

rakuco, 2022-11-02:
AFAICS libgbm.so is supposed to be provided by the platform's GLES implementation. On desktop Linux you'd get it from Mesa, for example.

@msisov does the above make sense? You know more about this part of the code base than I do.

msisov, 2022-11-02:
yes, on desktop it's provided by mesa for example

nandra, 2022-11-02:
@msisov so I should check if mesa provide such files? Files exists in final rootfs but looks like chromium is expecting something else. Any ideas what else can I check psl? Thanks.

msisov, 2022-11-02:
I've no idea what driver stm32 uses, but you can try to add virtual/libgbm to DEPENDS. Maybe, that helps (shrug).