OSSystems/meta-browser #674: Update to 107.0.5304.87

rwmacleod, 2022-11-06:
FYI, I'll be working on this update starting Nov 7th, 2022. I do the usual testing and try to also ensure that the musl patches work.

rwmacleod, 2022-11-09:
I have a 107 draft update https://github.com/rwmacleod/meta-browser
It works on qemux86-64 but not for qemuarm64 for both master and kirkstone. I confirmed that 106 works on kirkstone for qemuarm64.

I'll check if upstream knows about this and continue debugging.

rwmacleod, 2022-11-09:
I just noticed 107.0.5304.110 : https://chromereleases.googleblog.com/2022/11/stable-channel-update-for-desktop.html so I'll try that next.

rakuco, 2022-11-10:
Are you having a build issue or is something crashing at runtime?

rwmacleod, 2022-11-10:
It builds fine. The program loads, tries to talk to dbus, fails that as usual and then hangs. I have some strace data that I'll summarize later today and I've tried running chromium-bin under gdb but the result was inconclusive at least to me. I'll be doing that again today. I also just gather some logging data that I'll post in a minute from my desktop. Here's the default output:

[image]

rwmacleod, 2022-11-10:
and the log from /home/foo/.config/chromium/chrome_debug.log doesn't seem to be very helpful:

```
$ cat /folk/rmacleod/tmp/cr107//chrome_debug--qa64-kirkstone--107.0.5304.110.log
[637:637:1110/124421.267486:VERBOSE1:chrome_crash_reporter_client.cc(190)] GetCollectStatsConsent(): is_official_chrome_build is false so returning false
[637:637:1110/124421.323118:VERBOSE1:chrome_crash_reporter_client.cc(190)] GetCollectStatsConsent(): is_official_chrome_build is false so returning false
[1:1:1110/124423.345632:INFO:cpu_info.cc(53)] Available number of cores: 4
[646:646:1110/124423.346306:INFO:cpu_info.cc(53)] Available number of cores: 4
[646:646:1110/124423.351397:VERBOSE1:zygote_main_linux.cc(218)] ZygoteMain: initializing 0 fork delegates
[1:1:1110/124423.353255:VERBOSE1:zygote_main_linux.cc(218)] ZygoteMain: initializing 0 fork delegates
[637:637:1110/124427.143122:VERBOSE1:webrtc_event_log_manager.cc(96)] WebRTC remote-bound event logging enabled.
[637:637:1110/124427.333477:VERBOSE1:pref_proxy_config_tracker_impl.cc(187)] 0x2000326700: set chrome proxy config service to 0x2000313f00
[637:637:1110/124427.344898:VERBOSE1:google_api_keys.cc(314)] Using default value "invalid-client-id" for API key GOOGLE_CLIENT_ID_MAIN
[637:637:1110/124427.345466:VERBOSE1:google_api_keys.cc(314)] Using default value "invalid-client-secret" for API key GOOGLE_CLIENT_SECRET_MAIN
[637:637:1110/124427.345796:VERBOSE1:google_api_keys.cc(314)] Using default value "invalid-client-id" for API key GOOGLE_CLIENT_ID_REMOTING
[637:637:1110/124427.346119:VERBOSE1:google_api_keys.cc(314)] Using default value "invalid-client-secret" for API key GOOGLE_CLIENT_SECRET_REMOTING
[637:637:1110/124427.346440:VERBOSE1:google_api_keys.cc(314)] Using default value "invalid-client-id" for API key GOOGLE_CLIENT_ID_REMOTING_HOST
[637:637:1110/124427.346752:VERBOSE1:google_api_keys.cc(314)] Using default value "invalid-client-secret" for API key GOOGLE_CLIENT_SECRET_REMOTING_HOST
[637:662:1110/124427.849802:ERROR:bus.cc(399)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[637:662:1110/124427.858921:ERROR:bus.cc(399)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[637:662:1110/124427.859493:VERBOSE1:select_file_dialog_linux_portal.cc(248)] File chooser portal available: no
[637:665:1110/124500.097682:VERBOSE1:shutdown_signal_handlers_posix.cc(150)] Handling shutdown for signal 2.

```

rwmacleod, 2022-11-11:
FYI: qemuarm and the raspberrypi4 (32bit) and  from meta-raspberrypi with 107.0.5304.110 on kirkstone are fine BUT
raspberrypi4-64  hangs just as qa64 does.

Should we do the update despite arm64 being broken?
Any debugging tips? Today (Nov 11) is a holiday so I may not work on this again until Monday.

rakuco, 2022-11-11:
@msisov does this ring a bell by any chance?

msisov, 2022-11-14:
So you say it hangs, doesn't it? Can you paste bt of all threads when it hanged?

rwmacleod, 2022-11-14:
@msisov [image]

msisov, 2022-11-14:
@rwmacleod unfortunately, that's worthless without symbols :(

rwmacleod, 2022-11-14:
@msisov kids these days need symbols! ?  ;-)

Yeah, it's what I had on hand. Debugging in between meetings today so I'll post what I have later.

rwmacleod, 2022-11-16:
[image]

rakuco, 2022-11-16:
Hmm, there's still something wrong with the stack traces. If you launch Chromium with `--v=10 --single-process --no-sandbox` do you get anything else in the console?

msisov, 2022-11-17:
> [image]

it's maybe the very top frame that doesn't a have a symbol (an external lib?). Can you please check the stack trace of that particular thread for the "chromium-bin"? Thanks.

rwmacleod, 2022-11-23:
No news; I'm busy with other work. :-(  It might be a while so should I clean-up what I have now so that folks who aren't targeting arm64 will at least have a more secure version?

rwmacleod, 2022-11-28:
Updating to 107.0.5304.121 https://chromereleases.googleblog.com/2022/11/stable-channel-update-for-desktop_24.html

rwmacleod, 2022-12-10:
s/94/98/ Not sure if I misread the number of it changed here:
https://chromereleases.googleblog.com/2022/12/stable-channel-update-for-desktop_7.html

rwmacleod, 2022-12-22:
Finally back to this work.
The license script is broken so I'll work on that first.

MastaG, 2023-01-06:
Hi, any update on this?

rakuco, 2023-01-12:
@rwmacleod Thank you for working on this update!

> @kraj There's a build issue for TCLIBC=musl, shown below, that I'll look at tomorrow. Tips welcome.

You need the change to symbolize.h from https://chromium-review.googlesource.com/c/chromium/src/+/3997878

rwmacleod, 2023-01-12:
@kraj  Thanks, that was the last issue for musl. 

Chromium-108.x now builds and works well on master for both qemux86-64 with either glibc or musl. 
I'll test dunfell next and if that works, I'll go on to squashing patches, trying to write a sane commit log and then build testing for all the branches.

rwmacleod, 2023-01-12:
Well, qemux86-64 with glibc on dunfell (dunfell-clang12 for meta-clang), blows up with:

x86_64-poky-linux-clang++ -target x86_64-poky-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -rtlib=compiler-rt --unwindlib=libgcc -stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/ala-lpggp31/rmacleod/src/distro/yocto/b/ch108-dunfell/tmp/work/core2-64-poky-linux/chromium-x11/108.0.5359.124-r0/recipe-sysroot -MMD -MF obj/chrome/common/common_lib/chrome_manifest_url_handlers.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_40 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_40 -DWEBP_EXTERN=extern -DLIBYUV_DISABLE_NEON -DGL_GLEXT_PROTOTYPES -DUSE_GLX -DUSE_EGL -DVK_USE_PLATFORM_XCB_KHR -DON_FOCUS_PING_ENABLED -DTOOLKIT_VIEWS=1 -D_WTL_NO_AUTOMATIC_NAMESPACE -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_ENABLE_TRACING=1 -DU_ENABLE_RESOURCE_TRACING=0 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DGOOGLE_PROTOBUF_INTERNAL_DONATE_STEAL_INLINE=0 -DHAVE_PTHREAD -DSK_CODEC_DECODES_PNG -DSK_CODEC_DECODES_WEBP -DSK_ENCODE_PNG -DSK_ENCODE_WEBP -DSK_ENABLE_SKSL -DSK_UNTIL_CRBUG_1187654_IS_FIXED -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_WIN_FONTMGR_NO_SIMULATIONS -DSK_GL -DSK_CODEC_DECODES_JPEG -DSK_ENCODE_JPEG -DSK_HAS_WUFFS_LIBRARY -DSK_VULKAN=1 -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DV8_USE_EXTERNAL_STARTUP_DATA -DUSE_V8_CONTEXT_SNAPSHOT -DV8_CONTEXT_SNAPSHOT_FILENAME=\"v8_context_snapshot.bin\" -DWEBRTC_ENABLE_AVX2 -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_CHROMIUM_BUILD -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -DWEBRTC_USE_BUILTIN_ISAC_FIX=0 -DWEBRTC_USE_BUILTIN_ISAC_FLOAT=1 -DWEBRTC_USE_X11 -DLOGGING_INSIDE_WEBRTC -DLEVELDB_PLATFORM_CHROMIUM=1 -DATK_LIB_DIR=\"/usr/lib\" -DUSE_ATK_BRIDGE -DV8_COMPRESS_POINTERS -DV8_COMPRESS_POINTERS_IN_SHARED_CAGE -DV8_31BIT_SMIS_ON_64BIT_ARCH -DV8_ENABLE_SANDBOX -DV8_DEPRECATION_WARNINGS -DCPPGC_CAGED_HEAP -DCPPGC_YOUNG_GENERATION -DCPPGC_POINTER_COMPRESSION -DOPENSCREEN_TEST_DATA_DIR=\"third_party/openscreen/src/test/data/\" -I../.. -Igen -I../../third_party/perfetto/include -Igen/third_party/perfetto/build_config -Igen/third_party/perfetto -I../../third_party/libwebp/src/src -I../../third_party/libyuv/include -I../../third_party/jsoncpp/source/include -I../../third_party/vulkan-deps/vulkan-headers/src/include -I../../third_party/khronos -I../../gpu -Igen/third_party/dawn/include -I../../third_party/dawn/include -Igen/third_party/private_membership/src -Igen/third_party/shell-encryption/src -Igen/components/policy/proto -I../../third_party/wtl/include -Igen/shim_headers/flac_shim -I../../third_party/abseil-cpp -I../../third_party/boringssl/src/include -I../../third_party/protobuf/src -Igen/protoc_out -I../../third_party/ced/src -I../../third_party/icu/source/common -I../../third_party/icu/source/i18n -I../../net/third_party/quiche/overrides -I../../net/third_party/quiche/src/quiche/common/platform/default -I../../net/third_party/quiche/src -Igen/net/third_party/quiche/src -I../../third_party/skia -I../../third_party/wuffs/src/release/c -I../../third_party/vulkan/include -I../../third_party/webrtc_overrides -I../../third_party/webrtc -Igen/third_party/webrtc -I../../third_party/libwebm/source -I../../third_party/mesa_headers -I../../third_party/leveldatabase -I../../third_party/leveldatabase/src -I../../third_party/leveldatabase/src/include -I../../third_party/libaom/source/libaom -I../../v8/include -Igen/v8/include -Igen/third_party/metrics_proto -I../../third_party/zlib -I../../third_party/re2/src -I../../third_party/openscreen/src -Igen/third_party/openscreen/src -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables -m64 -msse3 -Xclang -fdebug-compilation-dir -Xclang . -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -I/ala-lpggp31/rmacleod/src/distro/yocto/b/ch108-dunfell/tmp/work/core2-64-poky-linux/chromium-x11/108.0.5359.124-r0/recipe-sysroot/usr/include/glib-2.0 -I/ala-lpggp31/rmacleod/src/distro/yocto/b/ch108-dunfell/tmp/work/core2-64-poky-linux/chromium-x11/108.0.5359.124-r0/recipe-sysroot/usr/lib/glib-2. [... cut ...]
/ala-lpggp31/rmacleod/src/distro/yocto/b/ch108-dunfell/tmp/work/core2-64-poky-linux/chromium-x11/108.0.5359.124-r0/recipe-sysroot/usr/include/c++/v1/map:982:9: note: candidate constructor template not viable: requires at most 3 arguments, but 5 were provided
        map(_InputIterator __f, _InputIterator __l,
        ^

I don't see a fix for that upstream and I left my c++ error secret decoder at my last job! ;-)
I have to do some other things for a bit so  I've added that last musl update to: https://github.com/rwmacleod/meta-browser/commits/chromium-108-2023-01-11 in case anyone wants to take a look or has their decoder ring handy.

kraj, 2023-01-12:
I wonder if chromium has started using some c++20 features from latest clang. Can you build it with `kirkstone` ?

rwmacleod, 2023-01-12:
I had already started that build so for a change, I was half a step ahead of you @kraj ;-) . 

For kirkstone, chromium/glibc builds fine. I've started musl just now so it'll build while I'm running errands.

rwmacleod, 2023-01-13:
It seems like we'll need a new dunfell branch: dunfell-clang1X in meta-clang, right @kraj ? 
Do you need me to do anything else? 

The commits that are associated with the error we're seeing on dunfell were not trivial but not huge either and while reverting it might work out, it's a risk and will take time. Since all branches other than dunfell-clang12 are working, this really points to the compiler version or default flags.

```
chromium.git on  main via 🐍 v2.7.18 
❯ git log --oneline -2 chrome/common/extensions/chrome_manifest_url_handlers.cc
cc15203c13bf7 Rename Manifest::available_values_dict() to available_values()
13306f0cb2d15 Migrate uses of Manifest::available_values() to available_values_dict()
❯ git log -1 --stat cc15203c13bf7053d943cb6015e2b2070eef6cd3
commit cc15203c13bf7053d943cb6015e2b2070eef6cd3
Author: Alex Turner <alexmt@chromium.org>
Date:   Tue Dec 20 20:54:06 2022

    Rename Manifest::available_values_dict() to available_values()
    
    Now that the deprecated available_values() has been fully removed, we
    can rename its replacmeent. Also migrates Manifest::available_values_ to
    base::Value::Dict.
    
    Bug: 1366865
    Change-Id: Ic9de64f98a7401629ca37d1f9df43c9d2bd9de3b
    Reviewed-on: https://chromium-review.googlesource.com/c/chromium/src/+/4098253
    Reviewed-by: David Bertoni <dbertoni@chromium.org>
    Reviewed-by: Devlin Cronin <rdevlin.cronin@chromium.org>
    Commit-Queue: Alex Turner <alexmt@chromium.org>
    Cr-Commit-Position: refs/heads/main@{#1085765}

 chrome/browser/extensions/webstore_installer.cc                           |  3 +--
 chrome/browser/ui/extensions/extensions_overrides/simple_overrides.cc     |  3 +--
 chrome/common/extensions/api/omnibox/omnibox_handler.cc                   |  3 +--
 chrome/common/extensions/api/side_panel/side_panel_info.cc                |  3 +--
 chrome/common/extensions/api/speech/tts_engine_manifest_handler.cc        |  2 +-
 chrome/common/extensions/api/system_indicator/system_indicator_handler.cc |  2 +-
 chrome/common/extensions/api/url_handlers/url_handlers_parser.cc          |  3 +--
 chrome/common/extensions/chrome_manifest_url_handlers.cc                  |  3 +--
 extensions/common/api/commands/commands_handler.cc                        |  2 +-
 extensions/common/api/declarative_net_request/dnr_manifest_handler.cc     |  3 +--
 extensions/common/extension.cc                                            |  2 +-
 extensions/common/manifest.cc                                             | 31 ++++++++++---------------------
 extensions/common/manifest.h                                              | 11 +++--------
 extensions/common/manifest_handlers/content_scripts_handler.cc            |  3 +--
 extensions/common/manifest_handlers/cross_origin_isolation_info.cc        |  3 +--
 extensions/common/manifest_handlers/extension_action_handler.cc           |  2 +-
 extensions/common/manifest_handlers/file_handler_info.cc                  |  3 +--
 extensions/common/manifest_handlers/file_handler_info_mv3.cc              |  3 +--
 extensions/common/manifest_handlers/icons_handler.cc                      |  2 +-
 extensions/common/manifest_handlers/incognito_info.cc                     |  3 +--
 extensions/common/manifest_handlers/oauth2_manifest_handler.cc            |  3 +--
 extensions/common/manifest_handlers/requirements_info.cc                  |  3 +--
 extensions/common/manifest_handlers/shared_module_info.cc                 |  3 +--
 extensions/common/manifest_handlers/web_accessible_resources_info.cc      |  6 ++----
 extensions/common/manifest_handlers/webview_info.cc                       |  2 +-
 extensions/common/manifest_unittest.cc                                    |  4 ++--
 26 files changed, 39 insertions(+), 72 deletions(-)

chromium.git on  main via 🐍 v2.7.18 
❯ git log -1 --stat 13306f0cb2d1547d3950c27844321a3e3fb21efd
commit 13306f0cb2d1547d3950c27844321a3e3fb21efd
Author: Alex Turner <alexmt@chromium.org>
Date:   Tue Dec 20 17:13:56 2022

    Migrate uses of Manifest::available_values() to available_values_dict()
    
    base::DictionaryValue is deprecated and should be replaced with
    base::Value::Dict. Two follow-ups are required: renaming
    available_values_dict() to available_values() and changing the type of
    available_values_.
    
[... 4 lines cut ...]
    Reviewed-by: David Bertoni <dbertoni@chromium.org>
    Reviewed-by: Devlin Cronin <rdevlin.cronin@chromium.org>
    Commit-Queue: Devlin Cronin <rdevlin.cronin@chromium.org>
    Cr-Commit-Position: refs/heads/main@{#1085665}

 chrome/browser/extensions/webstore_installer.cc                       | 2 +-
 chrome/browser/ui/extensions/extensions_overrides/simple_overrides.cc | 2 +-
 chrome/common/extensions/api/omnibox/omnibox_handler.cc               | 8 +++-----
 chrome/common/extensions/api/side_panel/side_panel_info.cc            | 2 +-
 chrome/common/extensions/chrome_manifest_url_handlers.cc              | 2 +-
 extensions/common/api/declarative_net_request/dnr_manifest_handler.cc | 2 +-
 extensions/common/extension.cc                                        | 2 +-
 extensions/common/manifest.h                                          | 8 ++------
 extensions/common/manifest_handlers/content_scripts_handler.cc        | 6 +++---
 extensions/common/manifest_handlers/cross_origin_isolation_info.cc    | 2 +-
 extensions/common/manifest_handlers/file_handler_info_mv3.cc          | 2 +-
 extensions/common/manifest_handlers/incognito_info.cc                 | 2 +-
 extensions/common/manifest_handlers/oauth2_manifest_handler.cc        | 2 +-
 extensions/common/manifest_handlers/requirements_info.cc              | 2 +-
 extensions/common/manifest_handlers/shared_module_info.cc             | 2 +-
 extensions/common/manifest_handlers/web_accessible_resources_info.cc  | 4 ++--
 extensions/common/manifest_unittest.cc                                | 8 ++++----
 17 files changed, 26 insertions(+), 32 deletions(-)

```

rakuco, 2023-01-13:
Note that neither of these commits is part of Chromium 108 though. You can use e.g. https://chromiumdash.appspot.com/commits?commit=13306f0cb2d1547d3950c27844321a3e3fb21efd to check it.

`git blame` says the lines causing the failures were introduced in https://chromium-review.googlesource.com/c/chromium/src/+/3843307 instead.

My uninformed guess is that this needs some bug fix in libc++ and, in the worst case, clang itself to recognize the new construct correctly.

rwmacleod, 2023-01-14:
@rakuco  Right, I was looking for potential fixes and clearly the coffee hadn't quite kicked in!

rwmacleod, 2023-01-17:
I updated to 109  because it needs to be done eventually and I wanted to see if perhaps the new language features had been reverted (unlikely, I know!). 
That built fine on kirkstone. I tried dunfell only to see:
    ... -o obj/base/third_party/symbolize/symbolize/symbolize.o
   clang-12: error: unknown argument: '-gsimple-template-names'
   
   patched that out, built dunfell and we're back to the previous error. :-(
   
      .../chrome_manifest_url_handlers.cc -o obj/chrome/common/common_lib/chrome_manifest_url_handlers.o
      ../../chrome/common/extensions/chrome_manifest_url_handlers.cc:101:7: 
      error: no matching constructor for initialization of 
      'std::map<const char *, std::reference_wrapper<const absl::optional<std::string>>>' (aka 'map<const char *, reference_wrapper<const optional<basic_string<char>>>>')
      std::map<const char*,
      ^

rakuco, 2023-01-17:
It looks like a bug in libc++. I've tried building the file with upstream clang 12 and 13 with Chromium's libc++ and both worked. If I try to use upstream libc++ 13, it also works. If I try libc++ 12 instead, I get the same error.

MastaG, 2023-01-17:
A lot of work goes into maintaining this layer as it's a massive code base.
Perhaps it would be an idea to forget about older clang releases and force users to use the latest instead.

rakuco, 2023-01-17:
I'd love to do that too, but if w do it means we'll stop supporting dunfell.

rwmacleod, 2023-01-17:
Well, I'm not an uber compiler geek but I back-ported the patch and it avoided the error!

Next up was a complaint that the expression in third_party/blink/renderer/platform/media/web_media_player_impl.cc
on this line:
     if constexpr (Flags & kEncrypted) {
wasn't a bool, so I cast it to such:
  if constexpr (bool(Flags & kEncrypted)) {
and that builds fine on dunfell/dunfell-clang12+patch. After I build sato and test it, it'll be onwards to kirkstone/master.

@rakuco Thanks for finding the libc patch. Was your bisecting outside of YP and done manually or via a magic script that you'd like to share?

kraj, 2023-01-17:
> I've done some bisecting here, and this is the libc++ commit that is not present in the 12.x releases: [llvm/llvm-project@eec0409](https://github.com/llvm/llvm-project/commit/eec04092d67b94f47439a9065b6bd4cd60165be2)
> 
> I just have no idea if it's safe to cherry-pick it into meta-clang's dunfell-clang12 branch, or if it's worth creating a dunfell-clang13 there instead.
> 
> @kraj any ideas?

It seems fine to backport, although one thing I worry about is that it got into llvm 13 during RC stages which is pretty late
that it might cause runtime side-effects with llvm-12, however, lets backport it for immediate concern. I think doing a dunfell-clang13 might be a better solution but it has its own work to do.

rakuco, 2023-01-18:
> @rakuco Thanks for finding the libc patch. Was your bisecting outside of YP and done manually or via a magic script that you'd like to share?

I did it manually:
* Checked out LLVM from GitHub, switched to the release/13.x branch
* In my upstream Chromium git checkout, I `touch`ed `chrome_manifest_url_handlers.cc`, ran `ninja -C out/Release -v -j1` and copied the compiler invocation to a separate file.
* Edited the command-line invocation:
  * Removed `-I../../buildtools/third_party/libc++`
  * Added `-D_LIBCPP_HAS_NO_VENDOR_AVAILABILITY_ANNOTATIONS` because some libc++ revisions were failing in the bisection
  * Replaced the inclusion of Chromium's bundled libc++ path with `-isystem/path/to/llvm-project/libcxx/include`
* Started bisecting `libcxx/include` between release/12.x and release/13.x.

rwmacleod, 2023-01-18:
Morning update: kirkstone and master both work fine for qemux86-64/glibc, as does rpi4-64 from meta-raspberrrypi. Next up: check the musl patches, clean-up the commit and finally send a freaking PR! ;-)

rwmacleod, 2023-01-19:
Submitted: 0001-chromium-update-to-109.0.5414.74.patch
more builds queued overnight.

rwmacleod, 2023-01-19:
PR: https://github.com/OSSystems/meta-browser/pull/688

closed 2023-01-21 by rwmacleod