OSSystems/meta-browser #679: chromium-ozone-wayland: No file chooser dialogs

MastaG, 2022-11-16:
Perhaps this is not the best place to ask this.
But I noticed that the chromium-ozone-wayland recipe sets: ```use_gtk=false```
I was wondering what's the reason for this.
Because without it, chromium will segfault when right-clicking anywhere on the page and selecting "Save as...".
Also when I try to attach a file in GMail, nothing happens when I click the attachment button.

So I wonder if chromium-ozone-wayland (without gtk support) doesn't feature a file-picker thing?
Or perhaps it doesn't work because I'm running it with: ```--in-process-gpu``` ?
My platform doesn't provide a complete DRI driver because it's running panfrost with the exynos-drm driver using dumb buffers, so --in-process-gpu is the only way for me to run it.

msisov, 2022-11-17:
That's definitely  a bug in our impl. There must be a stub instance for builds without gtk. Can you please file a bug to crbug  instead?

rakuco, 2022-11-17:
Aren't the file pickers always native though? I thought the Linux implementation first tried to use [Portals](https://flatpak.github.io/xdg-desktop-portal) before falling back to GTK, KDE and other implementations, but I haven't checked this in a while.

msisov, 2022-11-17:
We only have gtk implementation on Linux atm. And Wayland uses that. The reason why use_gtk is false in yocto meta-browser recipe is that we don't want to force users to depend on gtk stuff. "Save as" mustn't crash in that case.

MastaG, 2022-11-17:
Ah I see, I've opened a bug anyways: https://bugs.chromium.org/p/chromium/issues/detail?id=1385690
My platform is quite memory restrained, so I'd like to build without gtk support.
Also I think the native UI looks great.

MastaG, 2022-12-12:
I've tested the two MR's and I can confirm it fixes the freeze/crash when trying to open a file-picker window.
I couldn't get the xdg-desktop picker to work, but that's probably something related to my platform/desktop.
Too bad they wouldn't merge it for 109 though.

closed 2022-12-12 by MastaG