OSSystems/meta-browser #682: chromium compile add parameter option reported an error

Tracy-588, 2022-12-02:
I am compiling chromium browser, I need to make chromium support hardware acceleration decoding, so I opened v4l2, and added the option "use_v4lplugin=false" during the compilation. The following error occurred, may I ask what caused it

ERROR at the command-line "--args":1:1968: Invalid token.
|           use_cups=false          ffmpeg_branding="Chromium" proprietary_codecs=false  use_vaapi=false         is_component_build=false         use_gnome_keyring=false         use_kerberos=false         use_pulseaudio=true         use_system_freetype=true         use_system_libjpeg=true  enable_js_type_check=false host_pkg_config="pkg-config-native" is_debug=false is_official_build=true use_custom_libcxx=false use_lld=true use_gold=false symbol_level=0 enable_remoting=false enable_nacl=false use_sysroot=false treat_warnings_as_errors=false is_cfi=false fieldtrial_testing_like_official_build=true chrome_pgo_phase=0          google_api_key="invalid-api-key"         google_default_client_id="invalid-client-id"         google_default_client_secret="invalid-client-secret"           custom_toolchain="//build/toolchain/yocto:yocto_target"         gold_path=""         host_toolchain="//build/toolchain/yocto:yocto_native"         is_clang=true         clang_base_path="/home/demar2/yocoto-2.2.2/build-xwayland/tmp/work/cortexa53-crypto-mx8mm-poky-linux/chromium-ozone-wayland/91.0.4472.114-r0/recipe-sysroot-native/usr/bin"         clang_use_chrome_plugins=false         linux_use_bundled_binutils=false         target_cpu="arm64"         v8_snapshot_toolchain="//build/toolchain/yocto:yocto_target"           use_cups=false          ffmpeg_branding="Chromium" proprietary_codecs=false  use_vaapi=false         use_ozone=true         ozone_auto_platforms=false         ozone_platform_headless=true         ozone_platform_wayland=true         ozone_platform_x11=false         system_wayland_scanner_path="/home/demar2/yocoto-2.2.2/build-xwayland/tmp/work/cortexa53-crypto-mx8mm-poky-linux/chromium-ozone-wayland/91.0.4472.114-r0/recipe-sysroot-native/usr/bin/wayland-scanner"         use_system_wayland_scanner=true         use_xkbcommon=true         use_system_libwayland=true         use_system_minigbm=true         use_system_libdrm=true         use_gtk=false   use_v4lplugin=false  use_x11=false

rakuco, 2022-12-02:
It looks like you haven't pasted the entire command-line invocation so it's hard to tell. Did you add `use_v4lplugin=false` to `GN_ARGS`?

Tracy-588, 2022-12-05:
> It looks like you haven't pasted the entire command-line invocation so it's hard to tell. Did you add `use_v4lplugin=false` to `GN_ARGS`?

yes,I added 'use_v4lplugin=false' to 'GN_ARGS' and I got an error

rakuco, 2022-12-05:
Can you paste the entire command-line invocation that's causing the failure?

Tracy-588, 2022-12-06:
> Can you paste the entire command-line invocation that's causing the failure

Thanks for your reply. I have found the problem, which is the format of command line

rakuco, 2022-12-06:
OK, glad to hear it's working!

closed 2022-12-06 by rakuco