OSSystems/meta-browser #683: chromium-x11 DEPENDS is missing libx11-xcb

royaltm, 2022-12-06:
chromium-x11 dependency list should also include:

`libx11-xcb`

otherwise if the `libX11-xcb.so.1` is not present on the target, chromium won't start.

https://github.com/OSSystems/meta-browser/blob/master/meta-chromium/recipes-browser/chromium/chromium-x11_106.0.5249.119.bb#L5

A workaround is to explicitly add `libx11-xcb` to `IMAGE_INSTALL` in `local.conf`.

MastaG, 2022-12-19:
Well actually, DEPENDS lists the dependencies for building the recipe, these are staged into the build environment.
Apparently libx11-xcb is not a requirement for this.
As this package is only required for running chromium-x11, you might want to add it to RDEPENDS instead.
But I guess the developers assume that this package should be present by default for a typical X11 environment.

royaltm, 2022-12-19:
You are right. Nevertheless, I've managed to build an X11 system with chromium-x11 and without the libx11-xcb library. If you install only chromium and not much else, ie. for some kiosk configurations, you might hit a similar problem. 
I eventually found out what was missing only after using `strace` with the chromium binary.

MastaG, 2022-12-19:
Yes I could imagine it would be painful if you can't get it to start and not have many tools (such as ldd) to troubleshoot it.
@rakuco or @msisov perhaps you could add something like:
```
REDPENDS += "libx11-xcb"
```
to https://github.com/OSSystems/meta-browser/blob/master/meta-chromium/recipes-browser/chromium/chromium-x11_106.0.5249.119.bb ?

PR #690 (2023-01-26): chromium-x11: Add RDEPENDS on libx11-xcb.
libX11-xcb.so.1 is loaded at runtime by the X11 backend, but we had no
explicit dependency on the package that provides it (libx11-xcb).

It ended up being installed most of the time anyway, but make sure it is
listed as a runtime dependency to avoid the issue described in #683.

Fixes #683.
commit d5d3b82b (2023-01-26): chromium-x11: Add RDEPENDS on libx11-xcb.

libX11-xcb.so.1 is loaded at runtime by the X11 backend, but we had no
explicit dependency on the package that provides it (libx11-xcb).

It ended up being installed most of the time anyway, but make sure it is
listed as a runtime dependency to avoid the issue described in #683.

Fixes #683.
closed 2023-01-26 by rakuco