OSSystems/meta-browser #684: Unable to add chromium-ozone-wayland package in zeus version yocto

Ravikasyap, 2022-12-07:
Hi, 
I have downloaded meta-browser and meta-clang and switched to zeus version and added layer in bbappend file and able to build the image. But whenever i am trying to add chromium-ozone-wayland package, getting below error. 

ERROR: chromium-ozone-wayland-80.0.3987.132-r0 do_patch: Command Error: 'quilt --quiltrc /home/jofrartc/var-fsl-yocto/build_xwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/80.0.3987.132-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch Fix-build-breaks-on-non-gbm-machines.patch
patching file ui/ozone/platform/wayland/BUILD.gn
Hunk #1 succeeded at 260 (offset 32 lines).
Hunk #2 FAILED at 263.
Hunk #3 succeeded at 308 (offset 35 lines).
1 out of 3 hunks FAILED -- rejects in file ui/ozone/platform/wayland/BUILD.gn
Patch Fix-build-breaks-on-non-gbm-machines.patch does not apply (enforce with -f)

Kindly let me know how can i add this package.

rakuco, 2022-12-07:
You're using a recipe from early 2020, so it's very much possible that whatever is breaking has already been fixed. Can you try building the latest version of the recipe?

Ravikasyap, 2022-12-07:
I am new in yocto. Could you please guide me how to 
update the recipe

Ravikasyap, 2022-12-20:
Hi, 
I am getting attached error while building  chromium-ozone-wayland  packages with Yocto Krikstone. I have added meta-clang  layer  to build meta-chromium and using master branch. 

[log.txt](https://github.com/OSSystems/meta-browser/files/10267565/log.txt)

rakuco, 2023-01-26:
It'd help if you could compress and post your entire build log, it's weird that this symbol is not being found because the file where it is implemented should have been built, which is why I'm curious to see the whole build process.

Alternatively, can you check if the latest version of the recipe with Chromium 109 works for you?

Satyaprasad-GitHub, 2023-05-19:
@Ravikasyap,
Hi, Is this issue resolved. I'm also getting similar error when I try to compile chromium (master branch) with Yocto Kirkstone version.

Please share if you have any inputs.

Ravikasyap, 2023-05-22:
> @Ravikasyap, Hi, Is this issue resolved. I'm also getting similar error when I try to compile chromium (master branch) with Yocto Kirkstone version.
> 
> Please share if you have any inputs.
> 
> Build Configuration:
> 
> meta-chromium = "master:e8a756bb5ee28ee5adc668b78ff2011b0f3b62bd" meta-clang = "kirkstone-clang12:c2e89dc7e1dfcc1bbe432afca2dc43d6560cb007"

Please follow this link "https://github.com/OSSystems/meta-browser/issues/649"

Satyaprasad-GitHub, 2023-05-22:
Thank @Ravikasyap.

Compiling now with workaround.

closed 2024-03-15 by rakuco