OSSystems/meta-browser #685: Update sumo chromium recipe

mgonzalezg9, 2022-12-13:
Hi!
I have been researching ways of updating Chromium version from 64 stable to the latest one for a Sumo Yocto board. 
However I found out that chromium support was removed from master branch (although it was due to a firefox issue). See [https://github.com/OSSystems/meta-browser/commit/5a11120a6d54eb2d17a683a9c30a890a41008241](commit details).

Is there any way I could upgrade the recipe to get at least a newer version of Chromium? I have an IMX8 board with sumo yocto.

Thanks in advance!

rakuco, 2022-12-13:
I think you will run into other problems, such as a lack of support for sumo in meta-clang. sumo as a whole went EoL back in 2019 as far as I can see, so I am afraid you will have some work ahead if you want to do this. Perhaps you can try using the dunfell meta-clang branch and see how it goes, for example.

mgonzalezg9, 2022-12-13:
Thanks rakuco. The purpose on this is to install electron later, do you think I would be able to do that with the latest supported version for sumo? This version is Chromium 72,

rakuco, 2022-12-13:
Even if you did manage to build Chromium, I don't see how that would help with Electron, as AFAIR it ships its own Chromium copy at a specific version with extra patches.

rakuco, 2022-12-13:
I am not very familiar with Electron myself, all I know is that it ships with its own Chromium copy (or used to, at least). Poor performance can be caused by a lot of different things depending on what exactly you are experiencing, from bad graphics drivers to specific hardware you are using to some bug in Chromium. I can only offer general advice, like trying to see if a more recent Yocto branch works better, or if you can run an official Chrome build on your hardware to compare the performance data.

closed 2022-12-14 by mgonzalezg9