OSSystems/meta-browser #689: firefox on riscv

kush930, 2023-01-24:
Hi all,

I want to `port firefox on riscv64`, I cloned **meta-browser** layer from link:- https://github.com/OSSystems/meta-browser

using yocto build environmement i am building firefox, i am facing issue:- 

```
ERROR: firefox-68.9.0esr-r0 do_configure: Execution of '/home/kush/khu/sources/build/tmp-glibc/work/riscv64-oe-linux/firefox/68.9.0esr-r0/temp/run.do_configure.3619' failed with exit code 1:
checking for vcs source checkout... no
Adding configure options from /home/kush/khu/sources/build/tmp-glibc/work/riscv64-oe-linux/firefox/68.9.0esr-r0/firefox-68.9.0/mozconfig
  --enable-application=browser
  --enable-application=browser
  --enable-official-branding
  --enable-release
  --disable-gold
  --disable-strip
  --disable-install-strip
  --with-system-zlib
  --with-system-bz2
  --with-system-libevent
  --enable-system-ffi
  --enable-system-pixman
  --enable-mobile-optimize
  --enable-startup-notification
  --enable-pulseaudio
  --disable-tests
  --disable-necko-wifi
  --disable-gconf
  --disable-crashreporter
  --disable-updater
  --disable-debug-symbols
  --target=riscv64-oe-linux
  --host=x86_64-linux
  --with-toolchain-prefix=riscv64-oe-linux-
  --prefix=/usr
  --libdir=/usr/lib
  --enable-alsa
  --enable-default-toolkit=cairo-gtk3
  --disable-webrtc
  --enable-optimize=-Os -fsigned-char -fno-strict-aliasing
checking for host system type... x86_64-pc-linux-gnu
checking for target system type... 
ERROR: Unknown CPU type: riscv64

WARNING: exit code 1 from a shell command.
```

Can anyone help me to port firefox on riscv64.

T&R
Kush

kraj, 2023-01-24:
```
ERROR: Unknown CPU type: riscv64
```

thats clearly stating whats missing. Perhaps start with tip of trunk and see what all support went into latest firefox and then enhance it. 

see
https://bugzilla.mozilla.org/show_bug.cgi?id=1717203

lyctw, 2023-12-24:
Hi, has the issue been resolved in the latest version?