OSSystems/meta-browser #693: Chromium 109.0.5414.74 build Issue with Arm® Cortex® A35 based on i.MX8X

SilvanTeufel, 2023-02-15:
Building chromium-ozone-wayland on i.MX8X leads to following Issue for me:

silvan@ubuntu:~/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build$ bitbake chromium-ozone-wayland
Loading cache: 100% |#######################################################################################| Time: 0:00:02
Loaded 4536 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
NOTE: Multiple providers are available for runtime kernel-module-imx-gpu-viv (kernel-module-imx-gpu-viv, linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match kernel-module-imx-gpu-viv
NOTE: Multiple providers are available for runtime linux-mfgtool-image (linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match linux-mfgtool-image
NOTE: Multiple providers are available for runtime linux-mfgtool-image-image (linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match linux-mfgtool-image-image
NOTE: Multiple providers are available for runtime linux-mfgtool-base (linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match linux-mfgtool-base

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-tq-linux"
MACHINE              = "tqma8xqp-flowcomp"
DISTRO               = "dumpling-wayland-nxp"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa35 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:f98db02718944865811d3daf6404ab23bbb655b8"
meta-clang           
meta-oe              
meta-python          
meta-multimedia      
meta-webserver       
meta-networking      
meta-filesystems     = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-chromium        = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-freescale       = "HEAD:3747641f1e71d8e4edd5b587b49d09dc2d243942"
meta-freescale-distro 
meta-rust            = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-qt5             = "HEAD:5b71df60e523423b9df6793de9387f87a149ac42"
meta-tq              
meta-dumpling        = "HEAD:90ec60075fa72be93212ada3369e7a0021b37f3b"
meta-bsp             
meta-sdk             
meta-ml              = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-flowcomp        = "kirkstone-next:afb1ab4eac90d7d5487448700f0cc4422f32a414"

WARNING: /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_109.0.5414.74.bb:do_compile is tainted from a forced run
WARNING: /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_109.0.5414.74.bb:do_build is tainted from a forced run
Initialising tasks: 100% |##################################################################################| Time: 0:00:04
Sstate summary: Wanted 110 Local 103 Mirrors 0 Missed 7 Current 610 (93% match, 99% complete)
NOTE: Executing Tasks
ERROR: chromium-ozone-wayland-109.0.5414.74-r0 do_patch: Applying patch '0003-Disable-dri-for-imx-gpu.patch' on target directory '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build/tmp/work/cortexa35-mx8-tq-linux/chromium-ozone-wayland/109.0.5414.74-r0/chromium-109.0.5414.74'
CmdError('quilt --quiltrc /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build/tmp/work/cortexa35-mx8-tq-linux/chromium-ozone-wayland/109.0.5414.74-r0/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0003-Disable-dri-for-imx-gpu.patch
patching file content/gpu/BUILD.gn
Hunk #1 succeeded at 8 (offset 1 line).
Hunk #2 FAILED at 141.
1 out of 2 hunks FAILED -- rejects in file content/gpu/BUILD.gn
Patch 0003-Disable-dri-for-imx-gpu.patch does not apply (enforce with -f)

stderr: ')
ERROR: Logfile of failure stored in: /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build/tmp/work/cortexa35-mx8-tq-linux/chromium-ozone-wayland/109.0.5414.74-r0/temp/log.do_patch.466951
ERROR: Task (/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_109.0.5414.74.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2550 tasks of which 2549 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_109.0.5414.74.bb:do_patch
Summary: There were 2 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

__________________________________________________________________________

"Patch 0003-Disable-dri-for-imx-gpu.patch does not apply (enforce with -f)"

I am able to disable this Patch?
Any Ideas how to fix? :)

rakuco, 2023-02-15:
> "Patch 0003-Disable-dri-for-imx-gpu.patch does not apply (enforce with -f)"
> 
> I am able to disable this Patch? Any Ideas how to fix? :)

Hi. We've never had a patch with this name in this repository, so I'm guessing it's coming from your meta-chromium layer. You need to file an issue there, as the patch probably needs to be updated for this Chromium release.

SilvanTeufel, 2023-02-15:
Ah ok,

I could find it. I guess you can close here now.

closed 2023-02-15 by rakuco