OSSystems/meta-browser #694: Chromium-Ozone-Wayland Package "nss" missing

SilvanTeufel, 2023-02-16:
Building chromium-ozone-wayland for i.MX8X leads to following Issue for me:
Build Machine is a VM Ubuntu 20.04 64bit. 16GB Ram.

silvan@ubuntu:~/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build$ bitbake chromium-ozone-wayland
Loading cache: 100% |###########################################################################################| Time: 0:00:04
Loaded 4536 entries from dependency cache.
Parsing recipes: 100% |#########################################################################################| Time: 0:00:02
Parsing of 3030 .bb files complete (3029 cached, 1 parsed). 4536 targets, 482 skipped, 2 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: Multiple providers are available for runtime kernel-module-imx-gpu-viv (kernel-module-imx-gpu-viv, linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match kernel-module-imx-gpu-viv
NOTE: Multiple providers are available for runtime linux-mfgtool-image (linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match linux-mfgtool-image
NOTE: Multiple providers are available for runtime linux-mfgtool-base (linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match linux-mfgtool-base
NOTE: Multiple providers are available for runtime linux-mfgtool-image-image (linux-fslc-lts-mfgtool, linux-fslc-mfgtool, linux-imx-mfgtool)
Consider defining a PREFERRED_RPROVIDER entry to match linux-mfgtool-image-image

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-tq-linux"
MACHINE              = "tqma8xqp-flowcomp"
DISTRO               = "dumpling-wayland-nxp"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa35 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:f98db02718944865811d3daf6404ab23bbb655b8"
meta-clang           
meta-oe              
meta-python          
meta-multimedia      
meta-webserver       
meta-networking      
meta-filesystems     = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-chromium        = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-freescale       = "HEAD:3747641f1e71d8e4edd5b587b49d09dc2d243942"
meta-freescale-distro 
meta-rust            = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-qt5             = "HEAD:5b71df60e523423b9df6793de9387f87a149ac42"
meta-tq              
meta-dumpling        = "HEAD:90ec60075fa72be93212ada3369e7a0021b37f3b"
meta-bsp             
meta-sdk             
meta-ml              = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-flowcomp        = "kirkstone-next:afb1ab4eac90d7d5487448700f0cc4422f32a414"

WARNING: /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_109.0.5414.74.bb:do_compile is tainted from a forced run
WARNING: /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_109.0.5414.74.bb:do_build is tainted from a forced run
Initialising tasks: 100% |######################################################################################| Time: 0:00:05
Sstate summary: Wanted 108 Local 103 Mirrors 0 Missed 5 Current 622 (95% match, 99% complete)
NOTE: Executing Tasks
ERROR: chromium-ozone-wayland-109.0.5414.74-r0 do_configure: ExecutionError('/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build/tmp/work/cortexa35-mx8-tq-linux/chromium-ozone-wayland/109.0.5414.74-r0/temp/run.do_configure.497979', 1, None, None)
ERROR: Logfile of failure stored in: /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build/tmp/work/cortexa35-mx8-tq-linux/chromium-ozone-wayland/109.0.5414.74-r0/temp/log.do_configure.497979
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/gn-native_109.0.5414.74.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-imx/meta-bsp/recipes-graphics/drm/libdrm_2.4.109.imx.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-imx/meta-bsp/recipes-graphics/imx-gpu-viv/imx-gpu-viv_6.4.3.p4.6-aarch64.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-openembedded/meta-clang/recipes-devtools/clang/clang-cross_git.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-openembedded/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-openembedded/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-openembedded/meta-oe/recipes-support/nspr/nspr_4.29.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-bsp/pciutils/pciutils_3.7.0.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-core/expat/expat_2.4.9.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-core/glib-2.0/glib-2.0_2.72.3.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-devtools/desktop-file-utils/desktop-file-utils_0.26.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-devtools/qemu/qemu-native_6.2.0.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-gnome/gdk-pixbuf/gdk-pixbuf_2.42.9.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.34.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-gnome/hicolor-icon-theme/hicolor-icon-theme_0.17.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-graphics/cairo/cairo_1.16.0.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-graphics/jpeg/libjpeg-turbo_2.1.4.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-graphics/mesa/mesa_22.0.3.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-graphics/pango/pango_1.50.4.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-graphics/wayland/wayland_1.20.0.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-graphics/xorg-lib/libxkbcommon_1.4.1.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-multimedia/alsa/alsa-lib_1.2.6.1.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-multimedia/flac/flac_1.3.4.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-multimedia/pulseaudio/pulseaudio_15.0.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-multimedia/webp/libwebp_1.2.4.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-support/atk/at-spi2-atk_2.38.0.bb:do_populate_sysroot', '/home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/poky/meta/recipes-support/atk/atk_2.38.0.bb:do_populate_sysroot', [... cut ...]
Summary: There were 2 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

________________________________________________________________________________________________________________________

How can i add the missing nss package? Where does it come from "normaly". I am using core setup from tq-system. And added meta-layers from openembedded. Got everything fixed so far but Chromium, does not want to build cause of nss.

Hope for some help :)

best regards

Silvan Teufel

rakuco, 2023-02-16:
nss is part of meta-oe: https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-support/nss?h=kirkstone

meta-oe is a dependency of this layer and I see it's in your logs, so it's weird that it's not being found.

SilvanTeufel, 2023-02-16:
Do i have to add it anywhere else? Or can i check the Path where it searching for anyhow?

rakuco, 2023-02-16:
Is meta-oe in your bblayers.conf? What happens if you run `bitbake nss`?

SilvanTeufel, 2023-02-16:
Yes it is. This is how my Layer-Conf looks like:
___________________________________________________________________________________________
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BSPDIR := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)) + '/../..')}"

BBFILES ?= ""
BBLAYERS = "\
    ${BSPDIR}/sources/poky/meta \
    ${BSPDIR}/sources/poky/meta-poky \
    \
    ${BSPDIR}/sources/meta-openembedded/meta-clang \
    ${BSPDIR}/sources/meta-openembedded/meta-oe \
    ${BSPDIR}/sources/meta-openembedded/meta-python \
    ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
    ${BSPDIR}/sources/meta-openembedded/meta-webserver \
    ${BSPDIR}/sources/meta-openembedded/meta-networking \
    ${BSPDIR}/sources/meta-openembedded/meta-filesystems \
    ${BSPDIR}/sources/meta-browser/meta-chromium \
    \
    ${BSPDIR}/sources/meta-freescale \
    ${BSPDIR}/sources/meta-freescale-distro \
    \
    ${BSPDIR}/sources/meta-rust \
    ${BSPDIR}/sources/meta-qt5 \
    \
    ${BSPDIR}/sources/meta-tq/meta-tq \
    ${BSPDIR}/sources/meta-tq/meta-dumpling \
    \
    ${BSPDIR}/sources/meta-imx/meta-bsp  \
    ${BSPDIR}/sources/meta-imx/meta-sdk \
    ${BSPDIR}/sources/meta-imx/meta-ml \
    \
    ${BSPDIR}/sources/meta-flowcomp \
"
__________________________________________________________________

If i bitbake nss it is successful:
___________________________________________________________________________________________________________________
silvan@ubuntu:~/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build$ bitbake nss
Loading cache: 100% |###########################################################################################| Time: 0:00:08
Loaded 4536 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-tq-linux"
MACHINE              = "tqma8xqp-flowcomp"
DISTRO               = "dumpling-wayland-nxp"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa35 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:f98db02718944865811d3daf6404ab23bbb655b8"
meta-clang           
meta-oe              
meta-python          
meta-multimedia      
meta-webserver       
meta-networking      
meta-filesystems     = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-chromium        = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-freescale       = "HEAD:3747641f1e71d8e4edd5b587b49d09dc2d243942"
meta-freescale-distro 
meta-rust            = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-qt5             = "HEAD:5b71df60e523423b9df6793de9387f87a149ac42"
meta-tq              
meta-dumpling        = "HEAD:90ec60075fa72be93212ada3369e7a0021b37f3b"
meta-bsp             
meta-sdk             
meta-ml              = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-flowcomp        = "kirkstone-next:afb1ab4eac90d7d5487448700f0cc4422f32a414"

Initialising tasks: 100% |######################################################################################| Time: 0:00:04
Sstate summary: Wanted 16 Local 13 Mirrors 0 Missed 3 Current 197 (81% match, 98% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 859 tasks of which 855 didn't need to be rerun and all succeeded.
___________________________________________________________________________________________________________________

SilvanTeufel, 2023-02-16:
Second Time it got 100%:

_____________________________________________________

silvan@ubuntu:~/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/tqma8xqp_build$ bitbake nss
Loading cache: 100% |#####################################################################################################| Time: 0:00:04
Loaded 4536 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-tq-linux"
MACHINE              = "tqma8xqp-flowcomp"
DISTRO               = "dumpling-wayland-nxp"
DISTRO_VERSION       = "4.0.5"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa35 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:f98db02718944865811d3daf6404ab23bbb655b8"
meta-clang           
meta-oe              
meta-python          
meta-multimedia      
meta-webserver       
meta-networking      
meta-filesystems     = "HEAD:50d4a8d2a983a68383ef1ffec2c8e21adf0c1a79"
meta-chromium        = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-freescale       = "HEAD:3747641f1e71d8e4edd5b587b49d09dc2d243942"
meta-freescale-distro 
meta-rust            = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-qt5             = "HEAD:5b71df60e523423b9df6793de9387f87a149ac42"
meta-tq              
meta-dumpling        = "HEAD:90ec60075fa72be93212ada3369e7a0021b37f3b"
meta-bsp             
meta-sdk             
meta-ml              = "kirkstone-tqma8xxs:23d8ade6f37b2a33909e886e3889500ff5eac35f"
meta-flowcomp        = "kirkstone-next:afb1ab4eac90d7d5487448700f0cc4422f32a414"

Initialising tasks: 100% |################################################################################################| Time: 0:00:10
Sstate summary: Wanted 13 Local 13 Mirrors 0 Missed 0 Current 200 (100% match, 100% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 859 tasks of which 859 didn't need to be rerun and all succeeded.

rakuco, 2023-02-16:
Does `bitbake nss-native` also work? If it also does, the only thing I can think of is that some downstream changes are removing them from `DEPENDS` (just like #693 looked like a downstream issue).

SilvanTeufel, 2023-02-16:
bitbake nss-native -  is working.

I added nss and nss-native to the DEPENDS on the chromkum-*.74.bb. Then it went through the issue but throwed the same line for dri:

_______________________________________________________________________________________________________________________

ARNING: chromium-ozone-wayland-109.0.5414.74-r0 do_patch: Fuzz detected:

Applying patch 0107-V4L2VDA-Add-function-IsMultiQueue-for-S_FMT-and-G_FM.patch
patching file media/gpu/v4l2/v4l2_device.cc
Hunk #1 succeeded at 62 (offset 4 lines).
Hunk #2 succeeded at 501 (offset 18 lines).
Hunk #3 succeeded at 1476 (offset 31 lines).
patching file media/gpu/v4l2/v4l2_device.h
Hunk #1 succeeded at 469 with fuzz 2 (offset 45 lines).
patching file media/gpu/v4l2/v4l2_video_decode_accelerator.cc
Hunk #1 succeeded at 2155 (offset 2 lines).
Hunk #2 succeeded at 2284 (offset 3 lines).
Hunk #3 succeeded at 2302 (offset 3 lines).
Hunk #4 succeeded at 2319 (offset 3 lines).
Hunk #5 succeeded at 2361 (offset 3 lines).

Applying patch 0115-V4L2VDA-Add-V4L2_PIX_FMT_NV12M_8L128-format-for-amph.patch
patching file media/gpu/chromeos/fourcc.cc
Hunk #2 succeeded at 162 (offset 12 lines).
patching file media/gpu/chromeos/fourcc.h
patching file media/gpu/v4l2/generic_v4l2_device.cc
Hunk #1 succeeded at 44 with fuzz 2 (offset -2 lines).
Hunk #2 succeeded at 55 (offset -2 lines).
patching file media/gpu/v4l2/v4l2_video_decode_accelerator.cc
Hunk #1 succeeded at 1358 (offset 2 lines).
Hunk #2 succeeded at 1502 (offset 2 lines).

Applying patch 0116-V4L2VDA-Support-tile-to-linear-transform-for-amphion.patch
patching file media/gpu/v4l2/BUILD.gn
Hunk #1 succeeded at 81 (offset -5 lines).
patching file media/gpu/v4l2/v4l2_device.cc
Hunk #1 succeeded at 44 (offset 4 lines).
Hunk #2 succeeded at 192 (offset 18 lines).
Hunk #3 succeeded at 209 (offset 18 lines).
Hunk #4 succeeded at 264 (offset 18 lines).
Hunk #5 succeeded at 279 (offset 18 lines).
Hunk #6 succeeded at 388 (offset 18 lines).
Hunk #7 succeeded at 457 (offset 18 lines).
Hunk #8 succeeded at 1171 (offset 21 lines).
Hunk #9 succeeded at 1195 with fuzz 1 (offset 23 lines).
Hunk #10 succeeded at 1236 (offset 23 lines).
Hunk #11 succeeded at 1547 (offset 31 lines).
patching file media/gpu/v4l2/v4l2_device.h
Hunk #1 succeeded at 503 (offset 46 lines).

Applying patch 0118-V4L2VDA-Use-dlopen-to-dynamically-use-g2d-api.patch
patching file media/gpu/v4l2/BUILD.gn
Hunk #1 succeeded at 81 (offset -5 lines).
patching file media/gpu/v4l2/v4l2_device.cc
Hunk #1 succeeded at 44 (offset 4 lines).
Hunk #2 succeeded at 162 (offset 18 lines).
Hunk #3 succeeded at 433 (offset 18 lines).
Hunk #4 succeeded at 571 (offset 18 lines).
Hunk #5 succeeded at 592 (offset 18 lines).
Hunk #6 succeeded at 1332 (offset 21 lines).
Hunk #7 succeeded at 1354 with fuzz 1 (offset 23 lines).
Hunk #8 succeeded at 1763 (offset 31 lines).
Hunk #9 succeeded at 1949 (offset 42 lines).

The context lines in the patches can be updated with devtool:

    devtool modify chromium-ozone-wayland
    devtool finish --force-patch-refresh chromium-ozone-wayland <layer_path>

[... 35 lines cut ...]
Summary: 1 task failed:
  /home/silvan/workspace/kirkstone.TQMa8.BSP.SW.0087/ci-meta-tq/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_109.0.5414.74.bb:do_configure
Summary: There were 4 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

____________________________________________________________________________

I have updated my /content/gpu/BUILD.gn

______________________________________________________________________________

--- a/content/gpu/BUILD.gn      2021-07-30 16:02:54.298106046 -0700
+++ b/content/gpu/BUILD.gn      2021-07-30 16:03:19.318535373 -0700
@@ -135,7 +135,7 @@
   }
 
   # Use DRI on desktop Linux builds.
-  if (current_cpu != "s390x" && current_cpu != "ppc64" && is_linux &&
+  if (current_cpu != "s390x" && current_cpu != "ppc64" && is_linux && ozone_platform_x11 &&
       !is_castos) {
     configs += [ "//build/config/linux/dri" ]
   }
____________________________________________________________________________________

Compile is not finished yet. Not sure if it will work. I will make an Update here if it worked or not. :)

rakuco, 2023-02-16:
nss and nss-native are already in `DEPENDS`: https://github.com/OSSystems/meta-browser/blob/master/meta-chromium/recipes-browser/chromium/chromium-gn.inc#L79

This, plus the fact that none of the patches that caused issues above exist in this repository, makes me think that your issues are caused by downstream changes like #693 was.

SilvanTeufel, 2023-02-16:
Yes i think you are right. But i think its compiling now.

SilvanTeufel, 2023-02-24:
I tryed a lower Version and successfully got compiled and run chromium. Runs fast and stable if i start as root with --no-sandbox.

But as soon as i switch to a user it throws.

SIOCSIFFLAGS: Operation not permitted
SIOCSIFFLAGS: Operation not permitted
SIOCSIFFLAGS: Operation not permitted
SIOCSIFFLAGS: Operation not permitted
[677:677:0224/120430.251998:ERROR:wayland_connection.cc(209)] Failed to connect to Wayland display
[677:677:0224/120430.252175:ERROR:ozone_platform_wayland.cc(226)] Failed to initialize Wayland platform
[677:677:0224/120430.252219:ERROR:env.cc(225)] The platform failed to initialize.  Exiting.

i tried to add user to same group as root .

rakuco, 2023-02-24:
It would be great if you could try the latest version since the code in Chromium changes quite frequently.

This looks like some platform-specific issue, but @MaxIhlenfeldt probably knows more about this area. What hardware are you using? Can you update the downstream recipes you're using and build/run Chromium 110?

MaxIhlenfeldt, 2023-02-24:
I suspect this could be a permission issue on the Wayland socket, or that Chromium doesn't know the correct socket path.

From a quick look at `meta-dumpling` it looks like the Wayland socket should be at `/run/wayland-0`. What are the owner, group and permissions for it? Have you tried running Chromium with `WAYLAND_DISPLAY=/run/wayland-0`?

SilvanTeufel, 2023-02-28:
I got it. Solution is Discussed here: https://community.nxp.com/t5/i-MX-Processors/Cannot-find-wayland-0-socket-under-non-root-user/m-p/1071842

This can be closed :). Thx for your help Guys.

MaxIhlenfeldt, 2023-02-28:
Happy to hear it works now!

closed 2023-02-28 by MaxIhlenfeldt