OSSystems/meta-browser #699: HW acceleration on stm32mp1

sedlalu2, 2023-03-07:
Hi, i am using stm32mp DK2 kit. I have built their latest ecosystem with Yocto and added chromium for testing. Compilation went just fine. 

Weston interface boots up correctly after start (desktop-shell.so). There are a few applications for GPU testing, which running without any problems either (weston-simple-egl,...).

Chromium on this setup also starts. But HW acceleration is not working correctly. I am lauching chromium with these arguments
```
--use-gl=egl --ozone-platform=wayland --no-sandbox
```
When I check chrome://gpu everything is disabled. So I added argument `--in-process-gpu`, then chrome://gpu shows that chromium is HW accelerated but I believe it is not really - when I monitor GPU during some CSS animated pages there is almost no load (1-3 % permanently), webgl is not working at all, and so on... 

During the start error shows up:
```
[10152:10179:0307/111631.034638:ERROR:gl_display.cc(892)] EGL display query failed with error EGL_SUCCESS
[10152:10179:0307/111631.034893:ERROR:gl_display.cc(917)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[10152:10179:0307/111631.035007:ERROR:gl_ozone_egl.cc(23)] GLDisplayEGL::Initialize failed.
...
[10152:10179:0307/111632.263693:ERROR:gpu_channel_manager.cc(883)] Failed to create GLES3 context, fallback to GLES2.
[10152:10179:0307/111632.264127:ERROR:gpu_channel_manager.cc(894)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[10152:10179:0307/111632.264374:ERROR:shared_image_stub.cc(475)] SharedImageStub: unable to create context
[10152:10179:0307/111632.264765:ERROR:gpu_channel.cc(589)] GpuChannel: Failed to create SharedImageStub
```

Am I missing something?

Also tried to add `--ignore-gpu-blocklist --ignore-gpu-blacklist` but none of these helped.

rakuco, 2023-03-07:
Tentatively cc'ing @MaxIhlenfeldt for GPU-related stuff

MaxIhlenfeldt, 2023-03-08:
I'm not too familiar with HW acceleration, sorry. Maybe @msisov still has some time to take a look?

In any case, if `chrome://gpu` reports that HW acceleration is available, but it looks like it isn't, this may be worth to report upstream?

sedlalu2, 2023-03-09:
I am starting chromium with these arguments:
```
chromium --no-sandbox --in-process-gpu --ignore-gpu-blacklist --ignore-gpu-blocklist --enable-features=CanvasOopRasterization,UseOzonePlatform --enable-accelerated-2d-canvas --enable-zero-copy --enable-gpu-compos
iting --enable-raw-draw --enable-native-gpu-memory-buffers --enable-gpu-rasterization --use-gl=egl --ozone-platform=wayland
```
This is chrome://gpu report
```
Graphics Feature Status
Canvas: Hardware accelerated
Canvas out-of-process rasterization: Enabled
Direct Rendering Display Compositor: Disabled
Compositing: Software only. Hardware acceleration disabled
Multiple Raster Threads: Disabled
OpenGL: Enabled
Rasterization: Hardware accelerated on all pages
Raw Draw: Disabled
Video Decode: Hardware accelerated
Video Encode: Software only. Hardware acceleration disabled
Vulkan: Disabled
WebGL: Hardware accelerated but at reduced performance
WebGL2: Hardware accelerated but at reduced performance
WebGPU: Disabled
Driver Bug Workarounds
clear_uniforms_before_first_program_use
enable_webgl_timer_query_extensions
exit_on_context_lost
disabled_extension_GL_KHR_blend_equation_advanced
disabled_extension_GL_KHR_blend_equation_advanced_coherent
disabled_extension_GL_MESA_framebuffer_flip_y
Problems Detected
WebGPU has been disabled via blocklist or the command line.
Disabled Features: webgpu
Raster is using a single thread.
Disabled Features: multiple_raster_threads
Accelerated video encode has been disabled, either via blocklist, about:flags or the command line.
Disabled Features: video_encode
Gpu compositing has been disabled, either via blocklist, about:flags or the command line. The browser will fall back to software compositing and hardware acceleration will be unavailable.
Disabled Features: gpu_compositing
Clear uniforms before first program use on all platforms: 124764, 349137
Applied Workarounds: clear_uniforms_before_first_program_use
Disable KHR_blend_equation_advanced until cc shaders are updated: 661715
Applied Workarounds: disable(GL_KHR_blend_equation_advanced), disable(GL_KHR_blend_equation_advanced_coherent)
Expose WebGL's disjoint_timer_query extensions on platforms with site isolation: 808744, 870491
Applied Workarounds: enable_webgl_timer_query_extensions
Some drivers can't recover after OUT_OF_MEM and context lost: 893177
Applied Workarounds: exit_on_context_lost
Disable GL_MESA_framebuffer_flip_y for desktop GL: 964010
Applied Workarounds: disable(GL_MESA_framebuffer_flip_y)
Version Information
Data exported
2023-03-08T10:50:00.381Z
Chrome version
Chrome/110.0.5481.77
Operating system
Linux 5.15.67
Software rendering list URL
https://chromium.googlesource.com/chromium/src/+/65ed616c6e8ee3fe0ad64fe83796c020644d42af/gpu/config/software_rendering_list.json
Driver bug list URL
https://chromium.googlesource.com/chromium/src/+/65ed616c6e8ee3fe0ad64fe83796c020644d42af/gpu/config/gpu_driver_bug_list.json
ANGLE commit id
unknown hash
2D graphics backend
Skia/110 aab9fb4100da797d25fe340e9a2fcb2ae30fc2e1
Command Line
/usr/bin/chromium --no-sandbox --in-process-gpu --ignore-gpu-blacklist --ignore-gpu-blocklist --enable-features=CanvasOopRasterization,UseOzonePlatform --enable-accelerated-2d-canvas --enable-zero-copy --enable-gpu-compositing --enablel
Driver Information
Initialization time
3563
In-process GPU
true
[... 250 lines cut ...]
Display[19] bounds=[0,0 800x480], workarea=[0,0 800x480], scale=1, rotation=270, panel_rotation=270 external.
Color space (all)
{primaries:BT709, transfer:SRGB, matrix:RGB, range:FULL}
Buffer format (all)
RGBA_8888
Color volume
{name:'srgb', r:[0.6400, 0.3300], g:[0.3000, 0.6000], b:[0.1500, 0.3300], w:[0.3127, 0.3290]}
SDR white level in nits
203
HDR relative maximum luminance
1
Bits per color component
8
Bits per pixel
24
Video Acceleration Information
Decoding
Encoding
Vulkan Information
Device Performance Information
Log Messages
```
Is looks like HW acceleration is partly working. But I am not so sure. WebGL is not working at all for example. In comparsion with WPEWebkit and Cog chromium's performance is still laggy. 

When I try to monitor GPU load it is around 5-10 % with chromium. On the same page with COG load is around 80 %. Monitoring with this command: https://wiki.st.com/stm32mpu/wiki/How_to_monitor_the_GCNANO_GPU_load

msisov, 2023-03-09:
First of all, you have to use --in-process-gpu because you system doesn't support wl_drm or zwp_linux_dmabuf, which require certain extensions.

```
Interfaces exposed by the Wayland compositor
wl_compositor:4 wl_subcompositor:1 wp_viewporter:1 zxdg_output_manager_v1:2 wp_presentation:1 zwp_relative_pointer_manager_v1:1 zwp_pointer_constraints_v1:1 zwp_input_timestamps_manager_v1:1 wl_data_device_manager:3 wl_shm:1 weston_tou2
```

Secondly, I think the main problem here is

```
[10152:10179:0307/111631.034638:ERROR:gl_display.cc(892)] EGL display query failed with error EGL_SUCCESS
[10152:10179:0307/111631.034893:ERROR:gl_display.cc(917)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[10152:10179:0307/111631.035007:ERROR:gl_ozone_egl.cc(23)] GLDisplayEGL::Initialize failed.
```

In case of --in-process-gpu, a display is initialised as [1]. I can't know why GL doesn't understand this display type on your system. It feels like you have to figure that out on your own (or ask your provider for help) as it doesn't look like a mainstream device and we haven't worked on the upstream support for that unlike WPE, which knows how to deal with that. Probably, there are driver specifics that we don't know and don't support, but which works with desktop devices.

[1] https://source.chromium.org/chromium/chromium/src/+/main:ui/ozone/platform/wayland/gpu/wayland_surface_factory.cc;l=162?q=wayland_surface_factory.cc&ss=chromium%2Fchromium%2Fsrc

sedlalu2, 2023-03-09:
@msisov Thank you. I will investigate a little more further.

Anyway, I also deleted original libEGL.so and libGLESv2.so in /usr/lib/chromium. Couldn't this be the problem why HW acceleration is not working as expected? Before I deleted it, I have got a lot of API based errors. Chromium should then use system installed libEGL and libGLESv2, shouldn't it?

Error log while using original libraries located in /usr/lib/chromium:
```
[1697:1715:0309/143440.935660:ERROR:bus.cc(399)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security.
[1697:1715:0309/143440.945154:ERROR:bus.cc(399)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security.
[     1] Failed EGL API GetProcAddress: eglBindWaylandDisplayWL !
[     2] Failed EGL API GetProcAddress: eglUnbindWaylandDisplayWL !
[     3] Failed EGL API GetProcAddress: eglQueryWaylandBufferWL !
[     4] Failed EGL API GetProcAddress: eglQueryDevicesEXT !
[     5] Failed EGL API GetProcAddress: eglCreateWaylandBufferFromImageWL !
[     6] Failed EGL API GetProcAddress: eglSwapBuffersWithDamageEXT !
[     7] Failed EGL API GetProcAddress: eglPatchID !
[     8] Failed ES_Common API GetProcAddress: forward_glActiveTexture !
[     9] Failed ES_Common API GetProcAddress: forward_glBindBuffer !
[    10] Failed ES_Common API GetProcAddress: forward_glBindTexture !
[    11] Failed ES_Common API GetProcAddress: forward_glBlendFunc !
[    12] Failed ES_Common API GetProcAddress: forward_glBufferData !
[    13] Failed ES_Common API GetProcAddress: forward_glBufferSubData !
[    14] Failed ES_Common API GetProcAddress: forward_glClear !
[    15] Failed ES_Common API GetProcAddress: forward_glClearColor !
[    16] Failed ES_Common API GetProcAddress: forward_glClearDepthf !
[    17] Failed ES_Common API GetProcAddress: forward_glClearStencil !
[    18] Failed ES_Common API GetProcAddress: forward_glColorMask !
[    19] Failed ES_Common API GetProcAddress: forward_glCompressedTexImage2D !
[    20] Failed ES_Common API GetProcAddress: forward_glCompressedTexSubImage2D !
[    21] Failed ES_Common API GetProcAddress: forward_glCopyTexImage2D !
[    22] Failed ES_Common API GetProcAddress: forward_glCopyTexSubImage2D !
[    23] Failed ES_Common API GetProcAddress: forward_glCullFace !
[    24] Failed ES_Common API GetProcAddress: forward_glDeleteBuffers !
[    25] Failed ES_Common API GetProcAddress: forward_glDeleteTextures !
[    26] Failed ES_Common API GetProcAddress: forward_glDepthFunc !
[    27] Failed ES_Common API GetProcAddress: forward_glDepthMask !
[    28] Failed ES_Common API GetProcAddress: forward_glDepthRangef !
[    29] Failed ES_Common API GetProcAddress: forward_glDisable !
[    30] Failed ES_Common API GetProcAddress: forward_glDrawArrays !
[    31] Failed ES_Common API GetProcAddress: forward_glDrawElements !
[    32] Failed ES_Common API GetProcAddress: forward_glEnable !
[    33] Failed ES_Common API GetProcAddress: forward_glFinish !
[    34] Failed ES_Common API GetProcAddress: forward_glFlush !
[    35] Failed ES_Common API GetProcAddress: forward_glFrontFace !
[    36] Failed ES_Common API GetProcAddress: forward_glGenBuffers !
[    37] Failed ES_Common API GetProcAddress: forward_glGenTextures !
[    38] Failed ES_Common API GetProcAddress: forward_glGetBooleanv !
[    39] Failed ES_Common API GetProcAddress: forward_glGetBufferParameteriv !
[    40] Failed ES_Common API GetProcAddress: forward_glGetError !
[    41] Failed ES_Common API GetProcAddress: forward_glGetFloatv !
[    42] Failed ES_Common API GetProcAddress: forward_glGetIntegerv !
[    43] Failed ES_Common API GetProcAddress: forward_glGetPointerv !
[    44] Failed ES_Common API GetProcAddress: forward_glGetString !
[    45] Failed ES_Common API GetProcAddress: forward_glGetTexParameterfv !
[    46] Failed ES_Common API GetProcAddress: forward_glGetTexParameteriv !
[    47] Failed ES_Common API GetProcAddress: forward_glHint !
[    48] Failed ES_Common API GetProcAddress: forward_glIsBuffer !
[    49] Failed ES_Common API GetProcAddress: forward_glIsEnabled !
[    50] Failed ES_Common API GetProcAddress: forward_glIsTexture !
[    51] Failed ES_Common API GetProcAddress: forward_glLineWidth !
[    52] Failed ES_Common API GetProcAddress: forward_glPixelStorei !
[    53] Failed ES_Common API GetProcAddress: forward_glPolygonOffset !
[    54] Failed ES_Common API GetProcAddress: forward_glReadPixels !
[    55] Failed ES_Common API GetProcAddress: forward_glSampleCoverage !
[    56] Failed ES_Common API GetProcAddress: forward_glScissor !
[    57] Failed ES_Common API GetProcAddress: forward_glStencilFunc !
[    58] Failed ES_Common API GetProcAddress: forward_glStencilMask !
[    59] Failed ES_Common API GetProcAddress: forward_glStencilOp !
[    60] Failed ES_Common API GetProcAddress: forward_glTexImage2D !
[    61] Failed ES_Common API GetProcAddress: forward_glTexParameterf !
[    62] Failed ES_Common API GetProcAddress: forward_glTexParameterfv !
[... 21 lines cut ...]
[    84] Failed GLES32 API GetProcAddress: glTexDirectWithTileStatusMapVIV !
[    85] Failed GLES32 API GetProcAddress: glGetTexImage !
[    86] Failed ES Common GLES2X API GetProcAddress: glMultiDrawArraysEXT !
[1697:1725:0309/143441.139478:ERROR:gl_display.cc(892)] EGL display query failed with error EGL_SUCCESS
[... cut ...]
[1697:1725:0309/143442.401767:ERROR:gpu_channel.cc(589)] GpuChannel: Failed to create SharedImageStub
```

Btw, this error still remains:
```
[1697:1725:0309/143441.139478:ERROR:gl_display.cc(892)] EGL display query failed with error EGL_SUCCESS
[1697:1725:0309/143441.139650:ERROR:gl_display.cc(917)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[1697:1725:0309/143441.139748:ERROR:gl_ozone_egl.cc(23)] GLDisplayEGL::Initialize failed.
```
Does it mean that I should investigate more on GPU drivers level?

msisov, 2023-03-13:
> Why are these libraries in chromium folder anyway

They are there because of ANGLE usage by Chromium

> [1697:1725:0309/143441.139478:ERROR:gl_display.cc(892)] EGL display query failed with error EGL_SUCCESS
[1697:1725:0309/143441.139650:ERROR:gl_display.cc(917)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[1697:1725:0309/143441.139748:ERROR:gl_ozone_egl.cc(23)] GLDisplayEGL::Initialize failed.

Check your driver's logs. Maybe, that can give you a hint why it happens

sedlalu2, 2023-03-27:
@msisov 

No solution yet. I have tried to switch driver to Etnaviv instead of gcnano. Now when I launch chromium with `--in-process-gpu` flag no window shows up. Without the flag chromium starts successfully, but HW acceleration is not working at all.

Log when I launch chromium with `--in-process-gpu`:
```
Error sending request: Bad file descriptor
[1509:1536:0327/073206.219348:ERROR:gl_display.cc(504)] EGL Driver message (Error) eglInitialize: eglInitialize
[1509:1536:0327/073206.219729:ERROR:gl_display.cc(917)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[1509:1536:0327/073206.219899:ERROR:gl_ozone_egl.cc(23)] GLDisplayEGL::Initialize failed.
[1509:1536:0327/073207.279205:ERROR:gpu_channel_manager.cc(883)] Failed to create GLES3 context, fallback to GLES2.
[1509:1536:0327/073207.281692:ERROR:gpu_channel_manager.cc(894)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[1509:1536:0327/073207.281966:ERROR:shared_image_stub.cc(475)] SharedImageStub: unable to create context
[1509:1536:0327/073207.282176:ERROR:gpu_channel.cc(589)] GpuChannel: Failed to create SharedImageStub
[1509:1536:0327/073207.283101:ERROR:gpu_channel_manager.cc(883)] Failed to create GLES3 context, fallback to GLES2.
[1509:1536:0327/073207.283328:ERROR:gpu_channel_manager.cc(894)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[1509:1536:0327/073207.283514:ERROR:shared_image_stub.cc(475)] SharedImageStub: unable to create context
[1509:1536:0327/073207.283709:ERROR:gpu_channel.cc(589)] GpuChannel: Failed to create SharedImageStub
Error: Couldn't load Vulkan. Searched /usr/lib/chromium/libvulkan.so.1, /usr/lib/chromium/libvulkan.so.1, libvulkan.so.1.
    at operator() (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:248)
    at Initialize (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:253)
    at Create (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:215)
    at operator() (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:474)

Error: Couldn't load Vulkan. Searched /usr/lib/chromium/libvk_swiftshader.so, /usr/lib/chromium/libvk_swiftshader.so, libvk_swiftshader.so.
    at operator() (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:248)
    at Initialize (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:259)
    at Create (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:215)
    at operator() (../../third_party/dawn/src/dawn/native/vulkan/BackendVk.cpp:474)

Error: eglGetDisplay
    at Create (../../third_party/dawn/src/dawn/native/opengl/ContextEGL.cpp:32)

Error: eglGetDisplay
    at Create (../../third_party/dawn/src/dawn/native/opengl/ContextEGL.cpp:32)
```
Any idea why there is `Error sending request: Bad file descriptor` error printed? Tried also launch with `--enable-logging=stderr --v=1` but none useful information shows up.

msisov, 2023-03-27:
> No solution yet. I have tried to switch driver to Etnaviv instead of gcnano. Now when I launch chromium with --in-process-gpu flag no window shows up. Without the flag chromium starts successfully, but HW acceleration is not working at all.

It'd help if you can share your driver's log. There is no clue why there is no display as [eglInitialize](https://registry.khronos.org/EGL/sdk/docs/man/html/eglInitialize.xhtml) states that EGL_NOT_INITIALIZED is generated if display cannot be initialized.

sedlalu2, 2023-03-28:
This is all I can get when I set environmental variables for Mesa (22.0.3) debug
```
LIBGL_DEBUG=1
MESA_DEBUG=1
EGL_LOG_LEVEL=debug
```
```
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libEGL debug: Native platform type: wayland (environment)
Error sending request: Socket operation on non-socket
[3698:3726:0327/171458.895816:ERROR:gl_display.cc(504)] EGL Driver message (Error) eglInitialize: eglInitialize
libEGL debug: EGL user error 0x3001 (EGL_NOT_INITIALIZED) in eglInitialize

[3698:3726:0327/171458.896360:ERROR:gl_display.cc(917)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[3698:3726:0327/171458.896567:ERROR:gl_ozone_egl.cc(23)] GLDisplayEGL::Initialize failed.
...
```

No error messages in dmesg. Also no error messages when I turn on DRM debug in kernel debugfs. 

Is that what you meant by driver's log?

When I try to launch `weston-simple-eql`  example (which works fines), log is as follows:
```
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libEGL debug: using driver stm for 5

libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libEGL debug: MESA-LOADER: device is not located on the PCI bus

libEGL debug: using driver stm for 5

libEGL debug: MESA-LOADER: dlopen(/usr/lib/dri/stm_dri.so)

libEGL debug: found extension `DRI_Core'
libEGL info: found extension DRI_Core version 2
libEGL debug: found extension `DRI_IMAGE_DRIVER'
libEGL debug: found extension `DRI_DRI2'
libEGL info: found extension DRI_DRI2 version 4
libEGL debug: found extension `DRI_ConfigOptions'
libEGL debug: found extension `DRI_Core'
libEGL debug: found extension `DRI_IMAGE_DRIVER'
libEGL debug: found extension `DRI_DRI2'
libEGL debug: found extension `DRI_ConfigOptions'
libEGL info: found extension DRI_ConfigOptions version 2
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
...
```

msisov, 2023-03-30:
Thanks. It's probably a problem with how it gets the display handle. We store it here https://source.chromium.org/chromium/chromium/src/+/main:ui/ozone/platform/wayland/gpu/wayland_surface_factory.cc;l=164 and then get the EGLDisplay type somewhere here in the switch case https://source.chromium.org/chromium/chromium/src/+/main:ui/gl/gl_display.cc;drc=48340c1e35efad5fb0253025dcc36b3a9573e258;bpv=1;bpt=1;l=276

So, it's worth checking what path it uses. If it doesn't use angle, it must use https://source.chromium.org/chromium/chromium/src/+/main:ui/gl/gl_display.cc;l=284;drc=48340c1e35efad5fb0253025dcc36b3a9573e258;bpv=1;bpt=1

sedlalu2, 2023-03-30:
@msisov Thank you.

I'm starting chromium with argument `--use-gl=egl`. I suppose that is the corrent way to force chromium to use system GL libraries (mesa). With that argument the above error is printed out.

When I switch to `--use-gl=angle` then this warning occures:
```
[1063:1092:0328/145713.692315:ERROR:gl_factory.cc(128)] Requested GL implementation (gl=egl-angle,angle=default) not found in allowed implementations.
```

So it looks like angle is not allowed. Despite that warning chromium window at least shows up. Suprisingly, it shouldn't work with angle, should it? And what GL implementation will be used on fallback?  

When I open `chrome://gpu` page (`--use-gl=angle`) it shows:

[image]

From this report it looks like some of the features are HW accelerated, but I dont think they really are. For example WebGL is "HW accelerated at reduced performance" - but even the simpliest WebGL 1.0 example does not work at all.

nandra, 2023-04-03:
@sedlalu2 which poky release do you use? I'm using kirkstone and on same platform I see lot of crashes. Thanks.

sedlalu2, 2023-04-03:
@nandra I'm using kirkstone too. And still facing the same EGL based crashes I described above :)

nandra, 2023-04-03:
@sedlalu2 do you use chromium-x11 or ozone wayland please? Thanks.

sedlalu2, 2023-04-03:
@nandra Wayland ozone platform.

msisov, 2024-11-07:
> [3176:3202:1106/202331.027560:ERROR:gl_factory.cc(135)] Requested GL implementation (gl=egl-angle,angle=default) not found in allowed implementations: [(gl=egl
> -gles2,angle=none),(gl=egl-angle,angle=swiftshader)].

One of the possible problems here is that --use-gl=angle is requested. Try --use-gl=egl-gles2