OSSystems/meta-browser #700: new release:  111.0.5563.64

rwmacleod, 2023-03-07:
https://chromereleases.googleblog.com/2023/03/stable-channel-update-for-desktop.html
I'm not sure if I'll have time to work on this. Is anyone else interested?

MaxIhlenfeldt, 2023-03-08:
I'll give it a shot :+1:

MaxIhlenfeldt, 2023-03-08:
@rwmacleod I wanted to use `devtool` to automatically rebase the patches, as you mentioned in your 109 update commit:
> and finally rebase patches in a single bound using super-devtool now with SRC_URI:append support!

However, I haven't been able to get it working.

`devtools upgrade -V  111.0.5563.64 chromium-ozone-wayland` didn't work at all at first because it couldn't find the licenses. I was able to fix this by removing the `${S}/` prefix for all the entries in `LIC_FILES_CHECKSUM` (which shouldn't be needed as the default is to use `S` as the base directory for license files during the build). Then it threw an error when it couldn't cleanly apply a patch, and instead of giving me the chance to fix this manually and continue, it just went on to generate the license checksums for the new version, which failed because a version got removed. Maybe I need to update the `LIC_FILES_CHECKSUM` variable before running `devtool upgrade`?

`devtool modify chromium-ozone-wayland` after manually upgrading the version and `LIC_FILES_CHECKSUM` seems to almost do what I want, but it also just errors out when it can't cleanly apply a patch, and on the next run it does the complete unpacking again, which takes ages, so it would be faster to manually rebase the patches.

Could you please tell me how you managed to automatically rebase the patches?

rwmacleod, 2023-03-08:
It was a while ago and I do remember that it wasn't straightforward. As you have found, devtool was only useful for chromium to rebase patches after manually updating the recipe. As you state, it still takes a while but for me getting the patch rebase done automatically in one pass was a step forward.

rwmacleod, 2023-03-15:
Did devtool help after all?