OSSystems/meta-browser #702: [Feature request] widevine support on aarch64

redfast00, 2023-03-12:
Currently, it's not possible to play DRM protected content (like for example Spotify or Netflix). This is possible on aarch64, see https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/248 and https://aur.archlinux.org/cgit/aur.git/tree/?h=widevine-aarch64. However, this must be enabled while building Chromium (https://github.com/rpmfusion/chromium-freeworld/blob/d2165a468332661df1439208eb930661caaa9f1e/chromium-enable-widevine.patch#L7-L8).

MastaG, 2023-03-23:
I second this.
It would be nice to enable it for all architectures though and let people decide for themselves whether and how they're going to obtain the libwidevinecdm.so module.
I'm building chromium-ozone-wayland with widevine enabled for armv7 and it works great for watching Netflix.

rwmacleod, 2023-04-06:
Can you send a PR with widevine enabled as a PACKAGECONFIG  and a (vague?) README about enabling the feature?