OSSystems/meta-browser #707: Error building chromium-ozone-wayland for Kirkstone on i.MX7

thgbe, 2023-04-20:
I'm trying to build chromium-ozone-wayland for the `imx7dsabresd` machine. I'm using the current i.MX Yocto Kirkstone release [5.15.71-2.2.0](https://github.com/nxp-imx/imx-manifest/blob/imx-linux-kirkstone/imx-5.15.71-2.2.0.xml) with distro `fsl-imx-xwayland`.

I get the following error during compilation of the dependencies, `compiler-rt_git` and `libcxx_git`:

```
/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/hosttools/ld: cannot find -lstdc++: No such file or directory
```

Some more context:

```
| -- Check for working CXX compiler: /data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/bin/clang++ - broken
| CMake Error at /data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/share/cmake-3.22/Modules/CMakeTestCXXCompiler.cmake:62 (message):
|   The C++ compiler
| 
|     "/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/bin/clang++"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|     Change Dir: /data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/build/CMakeFiles/CMakeTmp
| 
|     Run Build Command(s):ninja cmTC_3692e && [1/2] Building CXX object CMakeFiles/cmTC_3692e.dir/testCXXCompiler.cxx.o
|     [2/2] Linking CXX executable cmTC_3692e
|     FAILED: cmTC_3692e
|     : && /data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/bin/clang++ -target x86_64-linux   -isystem/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/include -O2 -pipe   -stdlib=libstdc++ -target x86_64-linux   -isystem/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/include -O2 -pipe   -stdlib=libstdc++  -L/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/lib                         -L/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/lib                         -Wl,-O1 -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++ -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -rtlib=libgcc --unwindlib=libgcc  -L/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/lib                         -L/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/work/x86_64-linux/compiler-rt-native/14.0.3-r0/recipe-sysroot-native/lib                         -Wl,-O1 -unwindlib=libgcc -rtlib=libgcc -stdlib=libstdc++ -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -rtlib=libgcc --unwindlib=libgcc CMakeFiles/cmTC_3692e.dir/testCXXCompiler.cxx.o -o cmTC_3692e   && :
|     /data/user/imx/imx-yocto-bsp/imx7dsabresd_fsl-imx-xwayland/tmp/hosttools/ld: cannot find -lstdc++: No such file or directory
|     clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
|     ninja: build stopped: subcommand failed.

```

The complete output is here:  [error.txt](https://github.com/OSSystems/meta-browser/files/11285031/error.txt)

The same happens if I update meta-browser to the current master (https://github.com/OSSystems/meta-browser/commit/4b98e0655f2766a69667c95068a89775f352dcc4).

The same happens with chromium-x11.

rakuco, 2023-04-20:
This looks like an issue with the meta-clang dependencies (cc @kraj). Do you have the libstdc++ development libraries installed on your host?

thgbe, 2023-04-20:
Thanks for the quick reply, this fixed the issue.

I had checked before, but it seems I had the wrong version. My host is a Ubuntu 22.04 machine. I only had `libstdc++-11-dev` installed. After installing `libstdc++-12-dev`, the issue is gone.

closed 2023-04-20 by thgbe