OSSystems/meta-browser #710: chromium: Update to 113.0.5672.*

rwmacleod, 2023-05-04:
Just as we are close to finishing 112, along comes 113!
https://chromereleases.googleblog.com/2023/05/stable-channel-update-for-desktop.html

MaxIhlenfeldt, 2023-05-18:
If no one else wants to take this, I can tackle this next week.

nrpt-m, 2023-05-23:
@MaxIhlenfeldt, I am looking forward to your chromium-113 update and will test it once PR is available.
Let us know, if there are any problems.

MaxIhlenfeldt, 2023-05-30:
I see, makes sense. I'll continue to try and make it work with clang 12, then, and we can look into using clang 14 separately in #716.

rwmacleod, 2023-06-01:
@MaxIhlenfeldt  are you waiting to be able to use clang-14 on dunfell or should we try to fix the compile issues with clang-12?

If you send a PR, @nrpt-m  and I could test it and perhaps help fix any compilation issues.

MaxIhlenfeldt, 2023-06-01:
I'm still working on this. I now have something that compiles on kirkstone and langdale, dunfell still needs some work. Not sure if it makes sense to upload the WIP update, as changes to the patches or entirely new ones would probably mean we'd need to test the non-dunfell configurations again, right?

rwmacleod, 2023-06-01:
> I'm still working on this. I now have something that compiles on kirkstone and langdale, dunfell still needs some work. Not sure if it makes sense to upload the WIP update, as changes to the patches [...]

Sure, if you are making progress, we'll just wait to test.

If you want some input on the compilation problems, we can spend time on that if you are stuck.

MaxIhlenfeldt, 2023-06-02:
> If you want some input on the compilation problems, we can spend time on that if you are stuck.

Thanks! I think I now have everything patched that needs patching. Luckily, it's mostly just backports. I'll upload everything on Monday!

MaxIhlenfeldt, 2023-06-05:
PR is up at #721!

closed 2023-06-12 by MaxIhlenfeldt