OSSystems/meta-browser #713: Getting Compilation Issue with Chromium and Yocto kirkstone

Satyaprasad-GitHub, 2023-05-19:
Hi,
I'm using Yocto kirkstone with chromium (master) branch and getting below compilation issues.

5615.165-r0=/usr/src/debug/chromium-ozone-wayland/112.0.5615.165-r0                      -fdebug-prefix-map=/home/toradex/imx8mp_620/oe-core/build/tmp/work/cortexa53-mx8mp-tdx-linux/chromium-ozone-wayland/112.0.5615.165-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/toradex/imx8mp_620/oe-core/build/tmp/work/cortexa53-mx8mp-tdx-linux/chromium-ozone-wayland/112.0.5615.165-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -latk-1.0 -latk-bridge-2.0 -lgio-2.0 -ldbus-1 -latspi -lresolv -ljpeg -lexpat -luuid -ldrm -lxkbcommon -lpci -l:libffi_pic.a -lgbm -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lasound -lFLAC -lm -lz -lxslt -lxml2
| aarch64-tdx-linux-ld.lld: error: undefined symbol: gl::NativeViewGLSurfaceEGL::NativeViewGLSurfaceEGL(gl::GLDisplayEGL*, void*, std::__1::unique_ptr<gfx::VSyncProvider, std::__1::default_delete<gfx::VSyncProvider> >)
| >>> referenced by gl_surface_wayland.cc:35 (../../ui/ozone/platform/wayland/gpu/gl_surface_wayland.cc:35)
| >>>               thinlto-cache/llvmcache-5EE66D699A8852C77C1B02DAA33A977E95EE8846:(ui::GLSurfaceWayland::GLSurfaceWayland(gl::GLDisplayEGL*, std::__1::unique_ptr<wl_egl_window, ui::EGLWindowDeleter>, ui::WaylandWindow*))
| clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/toradex/imx8mp_620/oe-core/build/../layers/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_112.0.5615.165.bb:do_compile) failed with exit code '1'

Build Configuration:
meta-chromium        = "master:e8a756bb5ee28ee5adc668b78ff2011b0f3b62bd"
meta-clang           = "kirkstone-clang12:c2e89dc7e1dfcc1bbe432afca2dc43d6560cb007"

Satyaprasad-GitHub, 2023-05-22:
@nrpt-m, Thanks for the response. 

"kirkstone-clang12" branch was mentioned as per my SOC vendor documentation. Checking with vendor as per your comment and yet to get response.

LuisCarlosRojasTorres, 2023-06-06:
I also had an issue with meta-chromium in kirkstone branch , but the log is different:

```
DEBUG: Executing shell function do_compile
[1/44094] python3 ../../mojo/public/tools/mojom/mojom_parser.py --input-root ../.. --input-root gen --output-root gen --mojom-file-list=gen/components/services/storage/privileged/mojom/mojom__parser.rsp --check-imports gen/components/services/storage/privileged/mojom/mojom.build_metadata --enable-feature is_posix --enable-feature is_linux
[2/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/public/mojom/mojom__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
[3/44094] touch obj/components/services/storage/public/mojom/mojom__generate_message_ids.stamp
[4/44094] touch obj/components/services/storage/privileged/mojom/mojom__parser.stamp
[5/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/public/mojom/mojom_shared__generator.rsp --generate_non_variant_code -g c++
[6/44094] touch obj/components/services/storage/public/mojom/mojom_shared__generator.stamp
[7/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/privileged/mojom/mojom_shared__generator.rsp --generate_non_variant_code -g c++
[8/44094] touch obj/components/services/storage/privileged/mojom/mojom_shared__generator.stamp
[9/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/privileged/mojom/mojom__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
[10/44094] touch obj/components/services/storage/privileged/mojom/mojom__generate_message_ids.stamp
[11/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/content/common/mojo_bindings__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
FAILED: gen/content/common/agent_scheduling_group.mojom-shared-message-ids.h gen/content/common/aggregatable_report.mojom-shared-message-ids.h gen/content/common/associated_interfaces.mojom-shared-message-ids.h gen/content/common/child_process.mojom-shared-message-ids.h gen/content/common/dom_automation_controller.mojom-shared-message-ids.h gen/content/common/download/mhtml_file_writer.mojom-shared-message-ids.h gen/content/common/field_trial_recorder.mojom-shared-message-ids.h gen/content/common/frame.mojom-shared-message-ids.h gen/content/common/frame_messages.mojom-shared-message-ids.h gen/content/common/histogram_fetcher.mojom-shared-message-ids.h gen/content/common/input/input_injector.mojom-shared-message-ids.h gen/content/common/media/media_log_records.mojom-shared-message-ids.h gen/content/common/native_types.mojom-shared-message-ids.h gen/content/common/navigation_client.mojom-shared-message-ids.h gen/content/common/private_aggregation_host.mojom-shared-message-ids.h gen/content/common/render_message_filter.mojom-shared-message-ids.h gen/content/common/renderer.mojom-shared-message-ids.h gen/content/common/renderer_host.mojom-shared-message-ids.h gen/content/common/renderer_variations_configuration.mojom-shared-message-ids.h gen/content/common/shared_storage_worklet_service.mojom-shared-message-ids.h gen/content/common/web_ui.mojom-shared-message-ids.h gen/content/common/pepper_plugin.mojom-shared-message-ids.h 
python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/content/common/mojo_bindings__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
Traceback (most recent call last):
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 418, in <module>
    ret = main()
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 413, in main
    return args.func(args, remaining_args)
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 275, in _Generate
    processor._GenerateModule(
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 207, in _GenerateModule
[... cut ...]
meta-python          
meta-networking      
meta-perl            
meta-multimedia      = "kirkstone:9e539aaed608a03b1128a7361426703d8f9ba705"
meta-intel           = "kirkstone:1edf26e5b90371c9acf7bd6ac7155000de85f133"
meta-dotnet-core     = "master:8cf4fc8963168d4e1f06d19e520089b1d9a41151"
meta-clang           = "kirkstone:b9df71259272cc4e3078e92614720756ccb05845"
meta-chromium        = "master:633dbeecfe0db3e4ac9750d871ad213149728d26"

Have you ever had this kind of problem?

Thank you in advance!