OSSystems/meta-browser #716: chromium: use clang 14 on dunfell

MaxIhlenfeldt, 2023-05-30:
dunfell using clang 12 is responsible for a lot of our downstream patches, and using clang 14 would get rid of many of them. Maybe it's possible to do this in a similar way to the recently introduced possibility of using Node 14 on dunfell?

kraj, 2023-06-06:
using clang16 just for chromium might be fine for dunfell and kirkstone however, this would need work since we wont be able to build rest of OE packages with clang16 without major surgery.

MaxIhlenfeldt, 2023-06-15:
> we wont be able to build rest of OE packages with clang16 without major surgery.

I would have expected using a newer clang version to be backwards compatible, what kind of problems can arise from updating clang?

> using clang16 just for chromium might be fine

Is it possible to use clang16 just for chromium and use the default clang version otherwise?

kraj, 2023-06-15:
> > we wont be able to build rest of OE packages with clang16 without major surgery.
> 
> I would have expected using a newer clang version to be backwards compatible, what kind of problems can arise from updating clang?

Think of this, we are trying to use a compiler released in 2023 to compile packages which were predominantly released
in 2020, clang-16 was not around back then, so these packages may have code which new clang compiler will find errors
in or expose errors which older compiler was incapable of. This is commonly seen issue when mixing newer compiler with older source code of packages.

> 
> > using clang16 just for chromium might be fine
> 
> Is it possible to use clang16 just for chromium and use the default clang version otherwise?

I think yes. If we create `dunfell_clang16` then this branch may only be used to compile chromium alone.
this will be troublesome for folks who are using clang as default system compiler with dunfell and also using
chromium and want to upgrade to latest chromium. They will have to switch system compiler to gcc and select
clang only for chromium for best results.

MaxIhlenfeldt, 2023-06-20:
> This is commonly seen issue when mixing newer compiler with older source code of packages.

I had feared that my expectation regarding backwards compatibility might be a bit optimistic, thanks for confirming.

> this will be troublesome for folks who are using clang as default system compiler with dunfell and also using
chromium and want to upgrade to latest chromium. They will have to switch system compiler to gcc and select
clang only for chromium for best results.

Why is that? Is it not possible to use clang 12 as the default system compiler and clang 16 for Chromium only? Either way, I suppose it'd be a reasonable restriction, given that the alternative is not supporting Chromium on dunfell at all.

MaxIhlenfeldt, 2023-07-24:
Updating to m115 gives more errors that presumably wouldn't happen with clang 16:
```
| In file included from ../../chrome/test/chromedriver/capabilities.cc:5:
| In file included from ../../chrome/test/chromedriver/capabilities.h:10:
| In file included from /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/map:541:
| In file included from /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/__node_handle:63:
| In file included from /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/memory:846:
| In file included from /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/__memory/allocate_at_least.h:13:
| /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/__memory/allocator_traits.h:298:9: error: no matching function for call to 'construct_at'
|         _VSTD::construct_at(__p, _VSTD::forward<_Args>(__args)...);
|         ^~~~~~~~~~~~~~~~~~~
| /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/__config:658:17: note: expanded from macro '_VSTD'
| #  define _VSTD std
|                 ^
| /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/vector:818:21: note: in instantiation of function template specialization 'std::allocator_traits<std::allocator<BrandVersion>>::construct<BrandVersion, const std::string &, const std::string &, void, void>' requested here
|     __alloc_traits::construct(this->__alloc(), _VSTD::__to_address(__tx.__pos_),
|                     ^
| /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/vector:1631:9: note: in instantiation of function template specialization 'std::vector<BrandVersion>::__construct_one_at_end<const std::string &, const std::string &>' requested here
|         __construct_one_at_end(_VSTD::forward<_Args>(__args)...);
|         ^
| ../../chrome/test/chromedriver/capabilities.cc:358:16: note: in instantiation of function template specialization 'std::vector<BrandVersion>::emplace_back<const std::string &, const std::string &>' requested here
|         brands.emplace_back(*brand, *version);
|                ^
| /home/gitlab-runner/yocto-chromium/builds/ci-chromium-wayland-qemuarm64-mickledore/tmp-glibc/work/cortexa57-oe-linux/chromium-ozone-wayland/115.0.5790.102-r0/recipe-sysroot/usr/include/c++/v1/__memory/construct_at.h:33:38: note: candidate template ignored: substitution failure [with _Tp = BrandVersion, _Args = <const std::string &, const std::string &>]: no matching constructor for initialization of 'BrandVersion'
| _LIBCPP_HIDE_FROM_ABI constexpr _Tp* construct_at(_Tp* __location, _Args&&... __args) {
|                                      ^
| 1 error generated.
```
(I suspect the substitution error mentioned at the end doesn't happen with clang  >= 16.)

As already mentioned, there will be more errors as slowly the code base uses more and more C++20 features (this might not even be the only error of this kind in m115). What's the plan for moving to using clang 16 on dunfell, kirkstone and mickledore?

rakuco, 2023-07-28:
FWIW, I've been toying with a dunfell-clang14 branch that's basically cherry-picking a ton of commits from meta-clang's kirkstone branch. chromium-x11 and chromium-ozone-wayland built fine, but I haven't tried running the binaries yet.

rakuco, 2023-07-28:
That's probably true. At this point I'm trying to do this as a learning experience; if this works, we can have both dunfell and kirkstone using the same compiler version, in which case moving both to a newer version.

With that said, clang 15 is less than a year old, so my only fear is that we'll always spend a non-trivial amount of time chasing compiler failures simply because Chromium's always using LLVM's main branch. Building the Chromium recipe with Chromium's LLVM is probably a no-go :/

MaxIhlenfeldt, 2023-09-27:
How should we proceed here now that `meta-clang` has a `dunfell-clang14` branch? The update to 117 once more needs many build fixes that wouldn't be needed with clang 16. And while these fixes usually are trivial, they still take up a lot of my time :slightly_frowning_face:

MaxIhlenfeldt, 2023-09-27:
Sounds reasonable!

@kraj would you have time for that work? If not, I suppose I could give it a try.

kraj, 2023-09-27:
> Sounds reasonable!
> 
> @kraj would you have time for that work? If not, I suppose I could give it a try.

we use clang16 on `mickledore` and moved master to clang17,  dunfell will be EOL in Apr 24 so lets see if we can live with clang14

MaxIhlenfeldt, 2023-09-28:
> we use clang16 on mickledore

https://github.com/kraj/meta-clang/blob/mickledore/recipes-devtools/clang/clang.inc#L7 shows clang 15 being used, is this still WIP?

> lets see if we can live with clang14

kirkstone is also still on clang 14 and will be supported until April 2026, so we definitely need to update to clang 16. And iiuc, `meta-clang`'s `dunfell-clang14` and `kirkstone` branches are quite similar, so if we get clang 16 to work on kirkstone, we might also get it to work on dunfell? There will be another seven Chromium releases until April 2024, I think that's enough to warrant at least trying.

kraj, 2023-09-28:
I meant to say clang15 not 16.

MaxIhlenfeldt, 2023-09-28:
Thanks for the clarification!

Do you have time to look into updating mickledore and kirkstone to clang 16?

MaxIhlenfeldt, 2023-11-07:
@kraj do you think you'll have time for this soon?

MaxIhlenfeldt, 2023-11-14:
[It was just announced](https://groups.google.com/a/chromium.org/g/chromium-dev/c/qjveLqape0I/m/Hrfc-vw0BgAJ) that many C++20 features are now allowed in Chromium, so maintaining support for clang 14 will probably get very hard very soon.

MaxIhlenfeldt, 2024-02-07:
It seems like with Chromium 121 it's not really feasible to make it work with clang < 16. I will focus on getting nanbield and master to compile, but the amount of patches needed for kirkstone seems to high. (e.g. just look at https://crrev.com/c/5027747 and its relation chain)

@kraj if possible, can you please take a look at providing a `meta-clang/kirkstone-clang16` branch?

kraj, 2024-02-07:
@MaxIhlenfeldt I guess we have to bite the bullet, perhaps its better to move to `clang-18` in one go. So perhaps we can live with that for a while.

MaxIhlenfeldt, 2024-02-08:
> @MaxIhlenfeldt I guess we have to bite the bullet, perhaps its better to move to `clang-18` in one go. So perhaps we can live with that for a while.

You're right, probably longer than with clang 16. If clang 18 is possible, that would be great of course!

rwmacleod, 2024-03-26:
@MaxIhlenfeldt  @kraj  Is there something  that either @rjanani-p or I can do to help this along?

adalessandro, 2024-10-21:
Any updates on this issue?

kraj, 2024-10-21:
@adalessandro yeah, if someone can help backporting clang-19 to kirkstone, that will be awesome. I have not had anytime lately to do it myself, I am not sure if I will get a chance to do it until December either as I will be off for good part of November.

adalessandro, 2024-10-21:
> @adalessandro yeah, if someone can help backporting clang-19 to kirkstone, that will be awesome. I have not had anytime lately to do it myself, I am not sure if I will get a chance to do it until December either as I will be off for good part of November.

@kraj thanks for the quick reply. Sure, I need to get upstream chromium built on kirkstone, so I take a look on this one as part of that task. Just in case, if you happen to have a wip/sketch branch or any early comments that could help this move along faster, please let me know.

MaxIhlenfeldt, 2024-10-21:
Thanks for looking into this! Note that clang 16 is the bare minimum we need to build Chromium, but that might change again (or might have changed already since the last time we built kirkstone). Thus, having clang 17 or 18 (scarthgap uses the 18) would be even better. Not sure how much more work it'd be to backport clang 18?

adalessandro, 2024-10-21:
> Thanks for looking into this! Note that clang 16 is the bare minimum we need to build Chromium, but that might change again (or might have changed already since the last time we built kirkstone). Thus, having clang 17 or 18 (scarthgap uses 18) would be even better. Not sure how much more work it'd be to backport clang 18?

I can't really estimate the effort right now, will probably have a better idea in the next days. As always, @kraj feel free to jump in as you may have a clear overview on this.

I've just created https://github.com/OSSystems/meta-browser/issues/842 to track this effort, maybe we can move the discussion there.