OSSystems/meta-browser #718: chromium: build error for dunfell

micky7, 2023-05-30:
Hi,

i get the following error when building the chromium for dunfell.

internal/modules/cjs/loader.js:818
|   throw err;
|   ^
| 
| Error: Cannot find module 'node:process'
| Require stack:
| - /home/data/build/atlas/yocto-msd/build-msd-imx/tmp/work/aarch64-mx8mm-msd-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/third_party/node/node_modules/rollup/dist/bin/rollup
|     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:815:15)
|     at Function.Module._load (internal/modules/cjs/loader.js:667:27)
|     at Module.require (internal/modules/cjs/loader.js:887:19)
|     at require (internal/modules/cjs/helpers.js:74:18)
|     at Module.<anonymous> (/home/data/build/atlas/yocto-msd/build-msd-imx/tmp/work/aarch64-mx8mm-msd-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/third_party/node/node_modules/rollup/dist/bin/rollup:16:19)
|     at Module._compile (internal/modules/cjs/loader.js:999:30)
|     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
|     at Module.load (internal/modules/cjs/loader.js:863:32)
|     at Function.Module._load (internal/modules/cjs/loader.js:708:14)
|     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12) {
|   code: 'MODULE_NOT_FOUND',
|   requireStack: [
|     '/home/data/build/atlas/yocto-msd/build-msd-imx/tmp/work/aarch64-mx8mm-msd-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/third_party/node/node_modules/rollup/dist/bin/rollup'
|   ]
| }
[log.do_compile.58645.zip](https://github.com/OSSystems/meta-browser/files/11601384/log.do_compile.58645.zip)

rakuco, 2023-05-30:
It looks like you're still using nodejs 12 for the build. Please see the end of the commit message in https://github.com/OSSystems/meta-browser/commit/e8a756bb5ee28ee5adc668b78ff2011b0f3b62bd for instructions for enabling the use of nodejs 14 in dunfell.

rwmacleod, 2023-06-17:
@micky7 Can you confirm that if you use the right version of Nodejs, the problem goes away?

micky7, 2023-06-20:
Hi,

yes i can confirm. Thanks

closed 2023-06-20 by micky7