OSSystems/meta-browser #722: chromium: Update to 114.0.5735.*

MaxIhlenfeldt, 2023-06-05:
Once again, 114 is out before we have finished updating to 113.

I can start this update as soon as we're done with 113, which should hopefully be soon.

MaxIhlenfeldt, 2023-06-12:
Now that the update to 113 is done, I'll start working on this tomorrow!

MaxIhlenfeldt, 2023-06-20:
It seems I have the build working for mickledore, and if we're lucky, the current build will be the first one to work for kirkstone as well.

However, I started getting the weird `use of undeclared identifier '__ops'; did you mean '__gnu_cxx::__ops'?` errors on dunfell again that I already mentioned [here](https://github.com/OSSystems/meta-browser/pull/706#issuecomment-1516234777) before when working on 112.

Not sure why it reappeared again, it was gone when I was working on 113. As it seems this is a local problem (no one else saw this for 112), I'll just upload the patch if that's the only remaining issue. Of course, there may still be other errors on dunfell that I'm unable to see as this one happens relatively early in the build, if that's the case you'd have to relay these errors to me so I can fix them :sweat_smile:

MaxIhlenfeldt, 2023-06-20:
Hm, I can reproduce the dunfell error on another machine. I'll try one more build on a third machine that's using a different distro than the other two.

@kraj I think we started looking into this during the m112 update (see comments after [this one](https://github.com/OSSystems/meta-browser/pull/706#issuecomment-1516234777), but never got anywhere as I didn't have time and it didn't reappear until know. I'd appreciate any ideas you might have :smile:

MaxIhlenfeldt, 2023-06-20:
The third machine gives me an error when trying to compile nodejs-native, now I'm properly confused... I feel like I'm missing something obvious.
```
| In file included from ../deps/v8/src/base/macros.h:12,
|                  from ../deps/v8/src/libplatform/worker-thread.h:12,
|                  from ../deps/v8/src/libplatform/worker-thread.cc:5:
| ../deps/v8/src/base/logging.h: In function ‘typename std::enable_if<(std::is_enum<_Tp>::value && (! v8::base::has_output_operator<T>::value)), std::__cxx11::basic_string<char> >::type v8::base::PrintCheckOperand(T)’:
| ../deps/v8/src/base/logging.h:170:34: error: ‘uint8_t’ was not declared in this scope
|   170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
|       |                                  ^~~~~~~
| ../deps/v8/src/base/logging.h:16:1: note: ‘uint8_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
|    15 | #include "src/base/template-utils.h"
|   +++ |+#include <cstdint>
|    16 |
| ../deps/v8/src/base/logging.h:170:41: error: template argument 2 is invalid
|   170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
|       |                                         ^
| ../deps/v8/src/base/logging.h:170:51: error: ‘uint16_t’ was not declared in this scope
|   170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
|       |                                                   ^~~~~~~~
| ../deps/v8/src/base/logging.h:170:51: note: ‘uint16_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
| ../deps/v8/src/base/logging.h:172:61: error: template argument 1 is invalid
|   172 |                                 int16_t, underlying_t>::type>::type;
|       |                                                             ^
| ../deps/v8/src/base/logging.h:172:61: error: template argument 2 is invalid
| ../deps/v8/src/base/logging.h:172:62: error: ‘<declaration error>’ is not a template [-fpermissive]
|   172 |                                 int16_t, underlying_t>::type>::type;
|       |                                                              ^~
| ../deps/v8/src/base/logging.h:173:40: error: ‘int_t’ does not name a type; did you mean ‘wint_t’?
|   173 |   return PrintCheckOperand(static_cast<int_t>(static_cast<underlying_t>(val)));
|       |                                        ^~~~~
|       |                                        wint_t
| ../deps/v8/src/base/macros.h: At global scope:
| ../deps/v8/src/base/macros.h:338:25: error: ‘intptr_t’ has not been declared
|   338 | inline T RoundDown(T x, intptr_t m) {
|       |                         ^~~~~~~~
| ../deps/v8/src/base/macros.h:344:11: error: ‘intptr_t’ has not been declared
|   344 | template <intptr_t m, typename T>
|       |           ^~~~~~~~
| ../deps/v8/src/base/macros.h: In function ‘constexpr T RoundDown(T)’:
| ../deps/v8/src/base/macros.h:348:17: error: ‘m’ was not declared in this scope
|   348 |   STATIC_ASSERT(m != 0 && ((m & (m - 1)) == 0));
|       |                 ^
| ../deps/v8/src/base/macros.h:200:43: note: in definition of macro ‘STATIC_ASSERT’
|   200 | #define STATIC_ASSERT(test) static_assert(test, #test)
|       |                                           ^~~~
| ../deps/v8/src/base/macros.h: At global scope:
| ../deps/v8/src/base/macros.h:354:23: error: ‘intptr_t’ has not been declared
|   354 | inline T RoundUp(T x, intptr_t m) {
|       |                       ^~~~~~~~
| ../deps/v8/src/base/macros.h:358:11: error: ‘intptr_t’ has not been declared
|   358 | template <intptr_t m, typename T>
|       |           ^~~~~~~~
| ../deps/v8/src/base/macros.h: In function ‘constexpr T RoundUp(T)’:
| ../deps/v8/src/base/macros.h:361:20: error: ‘m’ was not declared in this scope
|   361 |   return RoundDown<m, T>(static_cast<T>(x + (m - 1)));
|       |                    ^
| ../deps/v8/src/base/macros.h: In function ‘void* AlignedAddress(void*, size_t)’:
| ../deps/v8/src/base/macros.h:372:51: error: ‘uintptr_t’ does not name a type
|   372 |   return reinterpret_cast<void*>(reinterpret_cast<uintptr_t>(address) &
|       |                                                   ^~~~~~~~~
| ../deps/v8/src/base/macros.h:13:1: note: ‘uintptr_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
|    12 | #include "src/base/logging.h"
|   +++ |+#include <cstdint>
|    13 |
| ../deps/v8/src/base/macros.h:373:47: error: ‘uintptr_t’ does not name a type
|   373 |                                  ~static_cast<uintptr_t>(alignment - 1));
|       |                                               ^~~~~~~~~
| ../deps/v8/src/base/macros.h:373:47: note: ‘uintptr_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
```

closed 2023-06-26 by MaxIhlenfeldt

russdill, 2023-06-26:
Note that enabling use_v4l2_codec=true requires linux-libc-headers >= 6.1 (not yet available in kirkstone and older)

MaxIhlenfeldt, 2023-06-26:
Was there a change related to this in 114? It seems to me that the default is `use_v4l2_codec=true` for non-Lacros platforms [1].

[1] https://source.chromium.org/chromium/chromium/src/+/main:media/gpu/args.gni;l=21-22;drc=1fe3ac087197f01f95332f9cd9eab25fa6b20b09

russdill, 2023-06-26:
Yes, the default is false. Just a note for those trying to build with true. For reference https://source.chromium.org/chromium/chromium/src/+/2a8f8a7b357b895108598ebee922c3a5fb877072 in 101.0.4933.0 introduced a requirement for v5.17 and https://source.chromium.org/chromium/chromium/src/+/a5c4a9aeb274b02f9caaf71ff3c3eed30bc8dc91 in 114.0.5720.0 introduced a requirement for v6.1