OSSystems/meta-browser #724: Getting compilation error with Kirkstone

LuisCarlosRojasTorres, 2023-06-06:
Good afternoon!
I had an issue with meta-chromium in kirkstone branch , but the log is different:

```
DEBUG: Executing shell function do_compile
[1/44094] python3 ../../mojo/public/tools/mojom/mojom_parser.py --input-root ../.. --input-root gen --output-root gen --mojom-file-list=gen/components/services/storage/privileged/mojom/mojom__parser.rsp --check-imports gen/components/services/storage/privileged/mojom/mojom.build_metadata --enable-feature is_posix --enable-feature is_linux
[2/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/public/mojom/mojom__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
[3/44094] touch obj/components/services/storage/public/mojom/mojom__generate_message_ids.stamp
[4/44094] touch obj/components/services/storage/privileged/mojom/mojom__parser.stamp
[5/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/public/mojom/mojom_shared__generator.rsp --generate_non_variant_code -g c++
[6/44094] touch obj/components/services/storage/public/mojom/mojom_shared__generator.stamp
[7/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/privileged/mojom/mojom_shared__generator.rsp --generate_non_variant_code -g c++
[8/44094] touch obj/components/services/storage/privileged/mojom/mojom_shared__generator.stamp
[9/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/components/services/storage/privileged/mojom/mojom__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
[10/44094] touch obj/components/services/storage/privileged/mojom/mojom__generate_message_ids.stamp
[11/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/content/common/mojo_bindings__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
FAILED: gen/content/common/agent_scheduling_group.mojom-shared-message-ids.h gen/content/common/aggregatable_report.mojom-shared-message-ids.h gen/content/common/associated_interfaces.mojom-shared-message-ids.h gen/content/common/child_process.mojom-shared-message-ids.h gen/content/common/dom_automation_controller.mojom-shared-message-ids.h gen/content/common/download/mhtml_file_writer.mojom-shared-message-ids.h gen/content/common/field_trial_recorder.mojom-shared-message-ids.h gen/content/common/frame.mojom-shared-message-ids.h gen/content/common/frame_messages.mojom-shared-message-ids.h gen/content/common/histogram_fetcher.mojom-shared-message-ids.h gen/content/common/input/input_injector.mojom-shared-message-ids.h gen/content/common/media/media_log_records.mojom-shared-message-ids.h gen/content/common/native_types.mojom-shared-message-ids.h gen/content/common/navigation_client.mojom-shared-message-ids.h gen/content/common/private_aggregation_host.mojom-shared-message-ids.h gen/content/common/render_message_filter.mojom-shared-message-ids.h gen/content/common/renderer.mojom-shared-message-ids.h gen/content/common/renderer_host.mojom-shared-message-ids.h gen/content/common/renderer_variations_configuration.mojom-shared-message-ids.h gen/content/common/shared_storage_worklet_service.mojom-shared-message-ids.h gen/content/common/web_ui.mojom-shared-message-ids.h gen/content/common/pepper_plugin.mojom-shared-message-ids.h 
python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/content/common/mojo_bindings__generate_message_ids.rsp --generate_non_variant_code --generate_message_ids -g c++ --scrambled_message_id_salt_path ../../chrome/VERSION
Traceback (most recent call last):
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 418, in <module>
    ret = main()
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 413, in main
    return args.func(args, remaining_args)
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 275, in _Generate
    processor._GenerateModule(
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/out/Release/../../mojo/public/tools/bindings/mojom_bindings_generator.py", line 207, in _GenerateModule
    module = Module.Load(f)
  File "/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11/112.0.5615.165-r0/chromium-112.0.5615.165/mojo/public/tools/mojom/mojom/generate/module.py", line 1524, in Load
    result = pickle.load(f)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8c in position 5: invalid start byte
[12/44094] python3 ../../mojo/public/tools/bindings/mojom_bindings_generator.py --use_bundled_pylibs -o gen generate -d ../../ -I ../../ --bytecode_path gen/mojo/public/tools/bindings --filelist=gen/content/common/mojo_bindings_shared__generator.rsp --generate_non_variant_code -g c++ --export_attribute COMPONENT_EXPORT\(MOJOM_SHARED_CONTENT_EXPORT\) --export_header base/component_export.h
FAILED: gen/content/common/agent_scheduling_group.mojom-params-data.h gen/content/common/agent_scheduling_group.mojom-shared-internal.h gen/content/common/agent_scheduling_group.mojom-shared.cc gen/content/common/agent_scheduling_group.mojom-shared.h gen/content/common/aggregatable_report.mojom-params-data.h gen/content/common/aggregatable_report.mojom-shared-internal.h gen/content/common/aggregatable_report.mojom-shared.cc gen/content/common/aggregatable_report.mojom-shared.h gen/content/common/associated_interfaces.mojom-params-data.h gen/content/common/associated_interfaces.mojom-shared-internal.h gen/content/common/associated_interfaces.mojom-shared.cc gen/content/common/associated_interfaces.mojom-shared.h gen/content/common/child_process.mojom-params-data.h gen/content/common/child_process.mojom-shared-internal.h gen/content/common/child_process.mojom-shared.cc gen/content/common/child_process.mojom-shared.h gen/content/common/dom_automation_controller.mojom-params-data.h gen/content/common/dom_automation_controller.mojom-shared-internal.h gen/content/common/dom_automation_controller.mojom-shared.cc gen/content/common/dom_automation_controller.mojom-shared.h gen/content/common/download/mhtml_file_writer.mojom-params-data.h gen/content/common/download/mhtml_file_writer.mojom-shared-internal.h gen/content/common/download/mhtml_file_writer.mojom-shared.cc gen/content/common/download/mhtml_file_writer.mojom-shared.h gen/content/common/field_trial_recorder.mojom-params-data.h gen/content/common/field_trial_recorder.mojom-shared-internal.h gen/content/common/field_trial_recorder.mojom-shared.cc gen/content/common/field_trial_recorder.mojom-shared.h gen/content/common/frame.mojom-params-data.h gen/content/common/frame.mojom-shared-internal.h gen/content/common/frame.mojom-shared.cc gen/content/common/frame.mojom-shared.h gen/content/common/frame_messages.mojom-params-data.h gen/content/common/frame_messages.mojom-shared-internal.h gen/content/common/frame_messages.mojom-shared.cc gen/content/common/frame_messages.mojom-shared.h gen/content/common/histogram_fetcher.mojom-params-data.h gen/content/common/histogram_fetcher.mojom-shared-internal.h gen/content/common/histogram_fetcher.mojom-shared.cc gen/content/common/histogram_fet [... cut ...]
meta-yocto-bsp = "kirkstone:4cc0e9438b450b43749730e128b6b9adb30f9663"
meta-oe
meta-python
meta-networking
meta-perl
meta-multimedia = "kirkstone:9e539aaed608a03b1128a7361426703d8f9ba705"
meta-intel = "kirkstone:1edf26e5b90371c9acf7bd6ac7155000de85f133"
meta-dotnet-core = "master:8cf4fc8963168d4e1f06d19e520089b1d9a41151"
meta-clang = "kirkstone:b9df71259272cc4e3078e92614720756ccb05845"
meta-chromium = "master:633dbeecfe0db3e4ac9750d871ad213149728d26"
```

Have somebody ever had this kind of problem?

Thank you in advance!

MaxIhlenfeldt, 2023-06-12:
Is that from a clean build? If not, you could try deleting `/home/redtower01/build-yocto/tmp/work/corei7-64-poky-linux/chromium-x11` and rebuilding.

If that doesn't work, it might be fixed in the 113 version of meta-chromium, which should be released today or tomorrow.

rwmacleod, 2023-07-11:
@LuisCarlosRojasTorres  Any update? Have you tried version 113 on kirkstone? Googling a bit picks up:
  https://stackoverflow.com/questions/22216076/unicodedecodeerror-utf8-codec-cant-decode-byte-0xa5-in-position-0-invalid-s
  
  What locale are you using?
  ✦ ❯ locale
LANG=en_CA.UTF-8
LANGUAGE=en_CA:en
LC_CTYPE="en_CA.UTF-8"
LC_NUMERIC="en_CA.UTF-8"
LC_TIME="en_CA.UTF-8"
LC_COLLATE="en_CA.UTF-8"
LC_MONETARY="en_CA.UTF-8"
LC_MESSAGES="en_CA.UTF-8"
LC_PAPER="en_CA.UTF-8"
LC_NAME="en_CA.UTF-8"
LC_ADDRESS="en_CA.UTF-8"
LC_TELEPHONE="en_CA.UTF-8"
LC_MEASUREMENT="en_CA.UTF-8"
LC_IDENTIFICATION="en_CA.UTF-8"
LC_ALL=