OSSystems/meta-browser #731: chromium: Update to 114.0.5735.198

rakuco, 2023-07-14:
See https://chromereleases.googleblog.com/2023/06/stable-channel-update-for-desktop_26.html

This is probably the last M114 stable release, and a first M115 stable release has been pushed to the Early Stable channel: https://chromereleases.googleblog.com/2023/07/early-stable-update-for-desktop.html

I'm filing this bug because it will probably be a lot easier to update to a new M114 release than to switch to M115.

MaxIhlenfeldt, 2023-07-14:
Didn't we already update to that version with #729?

rakuco, 2023-07-14:
Doh, that's what happens when you look at the version number in an outdated checkout rather than the GitHub repository page :-) I'll take this as motivation to grab more coffee.

closed 2023-07-14 by rakuco