OSSystems/meta-browser #732: Chromium: build 114.0.5735.198 fails for kirkstone with vaapi related errors

AndreasUfert, 2023-07-15:
Hi, when building chromium-x11 114.0.5735.198 in Yocto 4.0 (kirkstone) with 

`
PACKAGECONFIG += "proprietary-codecs"
PACKAGECONFIG += "use-vaapi"
`

the following errors occur:

`
 % grep error: 114.0.5735.198-r0/temp/log.do_compile.140516.txt
ALWAYS_INLINE error::Error GLES2DecoderImpl::DoMultiDrawArrays(
ALWAYS_INLINE error::Error GLES2DecoderImpl::DoMultiDrawElements(
../../media/gpu/vaapi/av1_vaapi_video_encoder_delegate.cc:497:30: error: no member named 'bit_depth_minus8' in '_VAEncSequenceParameterBufferAV1::(unnamed struct at /builds/yocto/build-linux-for-ices/build/tmp/work/corei7-64-poky-linux/chromium-x11/114.0.5735.198-r0/recipe-sysroot/usr/include/va/va_enc_av1.h:279:9)'
../../media/gpu/vaapi/av1_vaapi_video_encoder_delegate.cc:498:30: error: no member named 'subsampling_x' in '_VAEncSequenceParameterBufferAV1::(unnamed struct at /builds/yocto/build-linux-for-ices/build/tmp/work/corei7-64-poky-linux/chromium-x11/114.0.5735.198-r0/recipe-sysroot/usr/include/va/va_enc_av1.h:279:9)'
../../media/gpu/vaapi/av1_vaapi_video_encoder_delegate.cc:499:30: error: no member named 'subsampling_y' in '_VAEncSequenceParameterBufferAV1::(unnamed struct at /builds/yocto/build-linux-for-ices/build/tmp/work/corei7-64-poky-linux/chromium-x11/114.0.5735.198-r0/recipe-sysroot/usr/include/va/va_enc_av1.h:279:9)'
`

Last known working version of chromium-x11 was 113.0.5672.126

Is there anything we can change to get chromium-x11 built and running using this PACKAGECONFIG?

rakuco, 2023-07-15:
The problem seems to be that the libva version in kirkstone is too old: https://github.com/intel/libva/commit/395e502d162fe04ccec7700bd1a58dca40328605 was added in 2.15.0 and kirkstone has 2.14.0.

AndreasUfert, 2023-07-15:
Hmm, you're right. Any idea how to solve this issue? The recipes for libva_2.14.0 et al are in poky and meta-intel (kirkstone branch). Given that kirkstone is the current LTS version of Yocto, there's little hope the version will be upgraded in poky, is there? What about meta-intel? Does it make sense to ask over there?

rakuco, 2023-07-17:
@kraj do you have more information about this? Would it be acceptable to try to cherry-pick https://git.openembedded.org/openembedded-core/commit/?id=b15a278764fdac47d6db4bed116c7104db578d49 and https://git.openembedded.org/openembedded-core/commit/?id=59524db2b9361384240e362e214c0a85ef3dc414 to kirkstone?

AndreasUfert, 2023-07-18:
> The problem seems to be that the libva version in kirkstone is too old: [intel/libva@395e502](https://github.com/intel/libva/commit/395e502d162fe04ccec7700bd1a58dca40328605) was added in 2.15.0 and kirkstone has 2.14.0.

After your reply I picked the recipes for langdale from https://layers.openembedded.org/layerindex/branch/langdale/recipes/?q=libva (meta-openembedded and meta-intel) and chromium could be built successfully.

Works for me until a new Yocto LTS is released. Would drop those langdale recipes if the commits mentioned in https://github.com/OSSystems/meta-browser/issues/732#issuecomment-1637563829 are cherry-picked.