OSSystems/meta-browser #735: Chromium optimisation on imx6

hiten4612, 2023-07-22:
On my imx6dl custom board, I'm running the chromium-ozone-wayland_75.0.3770.142.bb web browser. 

 I ran Chrome with the following flags
`/usr/lib/chromium/chromium-bin --no-sandbox --reduce-gpu-sandbox --gpu-sandbox-start-early --ignore-gpu-blocklist --enable-gpu-rasterization --enable-features=RawDraw,CanvasOopRasterization,WebRTCPipeWireCapturer --in-process-gpu --use-gl=egl --ozone-platform=wayland` 

And use the sram command to monitor memory use.
  499 0        /usr/lib/chromium/chromium-        0   776.0K     8.1M    25.0M 
  522 0        /usr/lib/chromium/chromium-        0     6.2M    14.5M    33.8M 
  538 0        /usr/lib/chromium/chromium-        0    35.6M    48.9M    70.9M 
  497 0        /usr/lib/chromium/chromium-        0    55.3M    70.4M    97.5M 

As was already noted, it uses a lot of RAM. and I'd like to make that memory as small as possible.

MaxIhlenfeldt, 2023-07-24:
> I'm running the chromium-ozone-wayland_75.0.3770.142.bb web browser

Do I understand correctly that you are running Chromium 75? If so, updating the browser might already give some RAM usage improvements, as I believe they are continuously working on this issue (e.g., in 110 the "Memory Saver" was introduced, which might help with your use case).

hiten4612, 2023-07-26:
@MaxIhlenfeldt Thanks for your support, When i run chrome://gpu/ some features use HW Acceleration and some.
[image]

MaxIhlenfeldt, 2023-07-27:
I'm not sure I can help with that a lot, I can only recommend experimenting with the CLI flags you're using to see what effects they have in this regard.

Rutvij-dev, 2023-08-09:
Hi @MaxIhlenfeldt,

Any advise on the porting/migration ?

--
BR

MaxIhlenfeldt, 2023-08-09:
I don't recall anything special about 110. If you're facing concrete problems, I'd be happy to try and help. If they're not related to this issue (RAM usage / HW acceleration on imx6), please open a separate issue.

palaniya, 2023-08-10:
Hi @MaxIhlenfeldt  and @hiten4612 ,

I could achieve it using CGROUP (with memory limit option) with addition of swap space to control. It saved lot of mem.
I did the test with [https://webglsamples.org/aquarium/aquarium.html](url) and found same performance but less RAM usage, you can try this as well. Hope this helps.

@MaxIhlenfeldt, as you mentioned may v110 help.But similar to @hiten4612 I am also having 70+ version (older build).
To try this newer version I am planning to migrate on v110 (or perhaps another version having mem option), from your any past experience can you share the Blockers or steps?

I can sense huge difference from 70+ version to v110 though, can we cherry pick commits ?

MaxIhlenfeldt, 2023-08-10:
I don't think I have any suggestions other than just trying out the newer version. If anything comes up, feel free to open a separate issue.