OSSystems/meta-browser #740: chromium-ozone-wayland 114 - FAILED: yocto_native/obj/skia/skia/SkSLType.o

barndoor2276, 2023-08-05:
Just recently attempted to upgrade from chromium v90 to latest in master and am getting the below compilation error. Figured I'd start here and move the issue to meta-clang if needed. I'm pretty sure I had a newer version building earlier this year so I may step backwards a few versions of chromium and see what results I get.

I set preferred version of nodejs/nodejs-native to 14.% and and am using dunfell-clang12 branch of meta-clang.
I have no modifications to the chromium-ozone-wayland recipe other than `DEPENDS += "at-spi2-atk"` in a bbappend.

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "imx8mp"
DISTRO               = "seco-imx-xwayland"
DISTRO_VERSION       = "Yocto Dunfell"
TUNE_FEATURES        = "aarch64 cortexa53 crc crypto"
TARGET_FPU           = ""

poky                    = "dunfell:df86cc15d0a39d8d85747f7acc2c887cccfd9fa7"
meta-openembedded       = "dunfell:b8b0b06821d4d4df0cce4f07fa31a8ca1dd38f46"
meta-yocto-bsp          = "dunfell:df86cc15d0a39d8d85747f7acc2c887cccfd9fa7"
meta-chromium           = "master:0f2de2d3ede1388b841642ad5454831db6bda315"
meta-clang              = "dunfell-clang12:8acc0b9c2a3942077d43f0baefc2841cc993fae1"
meta-freescale          = "dunfell:f66fdef1c2dc735690a7ec7dd44a534d69ea3918"
meta-freescale-3rdparty = "dunfell:4c0a166fa2809f74c93bd4c56d9a0f7ccb7818c8"
meta-freescale-distro   = "dunfell:5d882cdf079b3bde0bd9869ce3ca3db411acbf3b"
```

```
| FAILED: yocto_native/obj/skia/skia/SkSLType.o
| clang++ -MMD -MF yocto_native/obj/skia/skia/SkSLType.o.d -DSK_CODEC_DECODES_JPEG_GAINMAPS -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DSK_CODEC_DECODES_PNG -DSK_CODEC_DECODES_WEBP -DSK_ENCODE_PNG -DSK_ENCODE_WEBP -DSK_ENABLE_SKSL -DSK_UNTIL_CRBUG_1187654_IS_FIXED -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_WIN_FONTMGR_NO_SIMULATIONS -DSK_DISABLE_LEGACY_IMAGE_FACTORIES -DSK_DISABLE_LEGACY_MAKE_TEXTURE_IMAGE -DSK_DISABLE_LEGACY_IMAGE_FLUSH -DSK_DISABLE_LEGACY_GET_BACKEND_TEXTURE -DSK_DISABLE_LEGACY_IMAGE_ENCODE_METHODS -DSK_DISABLE_LEGACY_IMAGE_ENCODER -DSK_CODEC_DECODES_JPEG -DSK_ENCODE_JPEG -DSK_HAS_WUFFS_LIBRARY -DSK_GANESH -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DSK_GL -DSK_VULKAN=1 -DVK_USE_PLATFORM_WAYLAND_KHR -DIS_SKIA_IMPL=1 -DSKIA_IMPLEMENTATION=1 -DSK_FREETYPE_MINIMUM_RUNTIME_VERSION=\(\(\(FREETYPE_MAJOR\)\ \*\ 0x01000000\)\ \|\ \(\(FREETYPE_MINOR\)\ \*\ 0x00010000\)\ \|\ \(\(FREETYPE_PATCH\)\ \*\ 0x00000100\)\) -DSK_TYPEFACE_FACTORY_FREETYPE -DSK_GAMMA_EXPONENT=1.2 -DSK_GAMMA_CONTRAST=0.2 -DSK_DEFAULT_FONT_CACHE_LIMIT=20971520 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_56 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_56 -DBASE_USE_PERFETTO_CLIENT_LIBRARY=1 -DWEBP_EXTERN=extern -DFT_CONFIG_MODULES_H=\"freetype-custom/freetype/config/ftmodule.h\" -DFT_CONFIG_OPTIONS_H=\"freetype-custom/freetype/config/ftoption.h\" -DPDFIUM_REQUIRED_MODULES -DCHROMIUM_RESTRICT_VISIBILITY -DUSE_SYSTEM_LIBJPEG -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_ENABLE_TRACING=1 -DU_ENABLE_RESOURCE_TRACING=0 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -I../.. -Iyocto_native/gen -I../../third_party/skia -I../../third_party/wuffs/src/release/c -I../../third_party/vulkan/include -I../../third_party/vulkan-deps/vulkan-headers/src/include -I../../third_party/wayland/src/src -I../../third_party/wayland/include/src -I../../third_party/perfetto/include -Iyocto_native/gen/third_party/perfetto/build_config -Iyocto_native/gen/third_party/perfetto -I../../third_party/libwebp/src/src -I../../third_party/abseil-cpp -I../../third_party/boringssl/src/include -I../../third_party/protobuf/src -Iyocto_native/gen/protoc_out -I../../third_party/libpng -I../../third_party/zlib -I../../third_party/freetype/include -I../../third_party/freetype/include/freetype-custom -I../../third_party/freetype/src/include -I../../third_party/harfbuzz-ng/src/src -I../../third_party/fontconfig/src -I../../third_party/icu/source/common -I../../third_party/icu/source/i18n -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -m64 -msse3 -Xclang -fdebug-compilation-dir -Xclang . -ftrivial-auto-var-init=pattern -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wno-redundant-parens -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-unused-but-set-variable -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -I/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include -I/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/glib-2.0 -I/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib/glib-2.0/include -I/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include -DPROTOBUF_ALLOW_DEPRECATED=1 -std=c++20 -Wno-trigraphs -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -isystem/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot-native/usr/include   -isystem/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot-native/usr/include   -O2 -pipe -c ../../third_party/skia/src/sksl/ir/SkSLType.cpp -o yocto_native/obj/skia/skia/SkSLType.o
| ../../third_party/skia/src/sksl/ir/SkSLType.cpp:1042:50: error: no viable constructor or deduction guide for deduction of template arguments of 'vector'
|                                                  std::vector(fieldSpan.begin(), fieldSpan.end()),
|                                                  ^
| /usr/lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/stl_vector.h:295:7: note: candidate function template not viable: no known conversion from 'const SkSL::Type::Field *' to 'size_type' (aka 'unsigned long') for 1st argument
|       vector(size_type __n, const value_type&amp; __value,
|       ^
| /usr/lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/stl_vector.h:283:7: note: candidate template ignored: couldn't infer template argument '_Tp'
|       vector(size_type __n, const allocator_type&amp; __a = allocator_type())
|       ^
| /usr/lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/stl_vector.h:411:2: note: candidate template ignored: couldn't infer template argument '_Tp'
|         vector(_InputIterator __first, _InputIterator __last,
|         ^
| /usr/lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/stl_vector.h:348:7: note: candidate template ignored: could not match 'vector&lt;_Tp, _Alloc&gt;' against 'const SkSL::Type::Field *'
|       vector(const vector&amp; __x, const allocator_type&amp; __a)
|       ^
| /usr/lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/stl_vector.h:358:7: note: candidate template ignored: could not match 'vector&lt;_Tp, _Alloc&gt;' against 'const SkSL::Type::Field *'
|       vector(vector&amp;&amp; __rv, const allocator_type&amp; __m)
|       ^
| /usr/lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/stl_vector.h:383:7: note: candidate template ignored: could not match 'initializer_list&lt;type-parameter-0-0&gt;' against 'const SkSL::Type::Field *'
[... cut ...]
| WARNING: /workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/temp/run.do_compile.10644:1 exit 1 from 'ninja -v -j 62 chrome chrome_sandbox chromedriver.unstripped'
| ERROR: Execution of '/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/temp/run.do_compile.10644' failed with exit code 1
NOTE: recipe chromium-ozone-wayland-114.0.5735.198-r0: task do_compile: Failed
```

rakuco, 2023-08-06:
This looks like #726 -- the build's failing when building some bits on the host using your distro's libstdc++, which is too old for what Chromium needs.

We're trying to solve that by working on a dunfell-clang14 branch in meta-clang that would use meta-clang's libc++ rather than the distro's libstdc++, but in the meantime you could try using a more recent Ubuntu release on your host (18.04 is EOL anyway) or using a newer GCC release on 18.04.

barndoor2276, 2023-08-09:
Took all day for the build to run. I'm not sure if it really exited properly in the end. The log file was over 6GB. But I did manage to pull the below snippet using your method of searching.

../../gpu/command_buffer/service/gles2_cmd_decoder.cc:11533:1: warning: unknown attribute 'always_inline' ignored [-Wunknown-attributes]
ALWAYS_INLINE error::Error GLES2DecoderImpl::DoMultiDrawElements(

In file included from ../../gpu/command_buffer/service/gles2_cmd_decoder.cc:61:
In file included from ../../gpu/command_buffer/service/gles2_cmd_decoder_passthrough.h:26:
../../gpu/command_buffer/service/client_service_map.h:85:3: warning: unknown attribute 'always_inline' ignored [-Wunknown-attributes]
  ALWAYS_INLINE bool HasClientID(ClientType client_id) const {
  ^~~~~~~~~~~~~
../../base/compiler_specific.h:55:25: note: expanded from macro 'ALWAYS_INLINE'
#define ALWAYS_INLINE [[clang::always_inline]] inline
                        ^~~~~~~~~~~~~~~~~~~~
../../gpu/command_buffer/service/gles2_cmd_decoder.cc:11165:1: warning: unknown attribute 'always_inline' ignored [-Wunknown-attributes]
ALWAYS_INLINE bool GLES2DecoderImpl::CheckMultiDrawArraysVertices(
^~~~~~~~~~~~~
../../base/compiler_specific.h:55:25: note: expanded from macro 'ALWAYS_INLINE'
#define ALWAYS_INLINE [[clang::always_inline]] inline
                        ^~~~~~~~~~~~~~~~~~~~
../../gpu/command_buffer/service/gles2_cmd_decoder.cc:11226:1: warning: unknown attribute 'always_inline' ignored [-Wunknown-attributes]
ALWAYS_INLINE bool GLES2DecoderImpl::CheckTransformFeedback(
^~~~~~~~~~~~~
../../base/compiler_specific.h:55:25: note: expanded from macro 'ALWAYS_INLINE'
#define ALWAYS_INLINE [[clang::always_inline]] inline
                        ^~~~~~~~~~~~~~~~~~~~
../../gpu/command_buffer/service/gles2_cmd_decoder.cc:11268:1: warning: unknown attribute 'always_inline' ignored [-Wunknown-attributes]
ALWAYS_INLINE error::Error GLES2DecoderImpl::DoMultiDrawArrays(
^~~~~~~~~~~~~
../../base/compiler_specific.h:55:25: note: expanded from macro 'ALWAYS_INLINE'
#define ALWAYS_INLINE [[clang::always_inline]] inline
                        ^~~~~~~~~~~~~~~~~~~~
../../gpu/command_buffer/service/gles2_cmd_decoder.cc:11464:1: warning: unknown attribute 'always_inline' ignored [-Wunknown-attributes]
ALWAYS_INLINE bool GLES2DecoderImpl::CheckMultiDrawElementsVertices(
^~~~~~~~~~~~~
../../base/compiler_specific.h:55:25: note: expanded from macro 'ALWAYS_INLINE'
#define ALWAYS_INLINE [[clang::always_inline]] inline
                        ^~~~~~~~~~~~~~~~~~~~
../../gpu/command_buffer/service/gles2_cmd_decoder.cc:11533:1: warning: unknown attribute 'always_inline' ignored [-Wunknown-attributes]
ALWAYS_INLINE error::Error GLES2DecoderImpl::DoMultiDrawElements(
^~~~~~~~~~~~~
../../base/compiler_specific.h:55:25: note: expanded from macro 'ALWAYS_INLINE'
#define ALWAYS_INLINE [[clang::always_inline]] inline
                        ^~~~~~~~~~~~~~~~~~~~
223 warnings generated.

barndoor2276, 2023-08-09:
Here we go, ran it again, got a faster result. And a better looking error.

```
[4270/4278] rm -f obj/chrome/browser/libbrowser.a && "aarch64-oe-linux-llvm-ar" -T -r -c -s -D obj/chrome/browser/libbrowser.a @"obj/chrome/browser/libbrowser.a.rsp"
[4271/4278] python3 ../../build/gn_run_binary.py ./v8-qemu-wrapper.sh ./v8_context_snapshot_generator --output_file=v8_context_snapshot.bin
[4272/4278] touch obj/tools/v8_context_snapshot/generate_v8_context_snapshot.stamp
[4273/4278] touch obj/tools/v8_context_snapshot/v8_context_snapshot.stamp
[4274/4278] rm -f obj/chrome/renderer/librenderer.a && "aarch64-oe-linux-llvm-ar" -T -r -c -s -D obj/chrome/renderer/librenderer.a @"obj/chrome/renderer/librenderer.a.rsp"
[4275/4278] touch obj/chrome/dependencies.stamp
[4276/4278] rm -f obj/headless/libheadless_non_renderer.a && "aarch64-oe-linux-llvm-ar" -T -r -c -s -D obj/headless/libheadless_non_renderer.a @"obj/headless/libheadless_non_renderer.a.rsp"
[4277/4278] rm -f obj/headless/libheadless_shell_lib.a && "aarch64-oe-linux-llvm-ar" -T -r -c -s -D obj/headless/libheadless_shell_lib.a @"obj/headless/libheadless_shell_lib.a.rsp"
[4278/4278] "python3" "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- aarch64-oe-linux-clang++ -target aarch64-oe-linux  -mcpu=cortex-a53+crc+crypto -rtlib=compiler-rt --unwindlib=libgcc -stdlib=libc++ -mlittle-endian -Qunused-arguments --sysroot=/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -flto=thin -Wl,--thinlto-jobs=14 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -Wl,-mllvm,-import-instr-limit=30 -fwhole-program-vtables -Wl,--undefined-version -Wl,--no-call-graph-profile-sort -Wl,-mllvm,-enable-machine-outliner=never -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-z,defs -Wl,--as-needed -pie -Wl,--disable-new-dtags -Wl,--lto-O2 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -latk-1.0 -latk-bridge-2.0 -lgio-2.0 -ldbus-1 -lresolv -lexpat -ljpeg -luuid -ldrm -lxkbcommon -latspi -lpci -lffi -lgbm -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lasound -lFLAC -lm -lz -lxslt -lxml2 
FAILED: chrome 
"python3" "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- aarch64-oe-linux-clang++ -target aarch64-oe-linux  -mcpu=cortex-a53+crc+crypto -rtlib=compiler-rt --unwindlib=libgcc -stdlib=libc++ -mlittle-endian -Qunused-arguments --sysroot=/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -flto=thin -Wl,--thinlto-jobs=14 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -Wl,-mllvm,-import-instr-limit=30 -fwhole-program-vtables -Wl,--undefined-version -Wl,--no-call-graph-profile-sort -Wl,-mllvm,-enable-machine-outliner=never -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-z,defs -Wl,--as-needed -pie -Wl,--disable-new-dtags -Wl,--lto-O2 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -latk-1.0 -latk-bridge-2.0 -lgio-2.0 -ldbus-1 -lresolv -lexpat -ljpeg -luuid -ldrm -lxkbcommon -latspi -lpci -lffi -lgbm -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lasound -lFLAC -lm -lz -lxslt -lxml2 
aarch64-oe-linux-ld.lld: error: undefined symbol: gl::NativeViewGLSurfaceEGL::NativeViewGLSurfaceEGL(gl::GLDisplayEGL*, void*, std::__1::unique_ptr<gfx::VSyncProvider, std::__1::default_delete<gfx::VSyncProvider> >)
>>> referenced by gl_surface_wayland.cc:35 (./../../ui/ozone/platform/wayland/gpu/gl_surface_wayland.cc:35)
>>>               thinlto-cache/Thin-955ab7.tmp.o:(ui::GLSurfaceWayland::GLSurfaceWayland(gl::GLDisplayEGL*, std::__1::unique_ptr<wl_egl_window, ui::EGLWindowDeleter>, ui::WaylandWindow*))
clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
WARNING: /workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/temp/run.do_compile.261:1 exit 1 from 'ninja -v -j 14 chrome chrome_sandbox chromedriver.unstripped'
ERROR: Execution of '/workdir/build-image-ross/yoctobuild/tmp-glibc/work/aarch64-mx8mp-oe-linux/chromium-ozone-wayland/114.0.5735.198-r0/temp/run.do_compile.261' failed with exit code 1
```

MaxIhlenfeldt, 2023-08-09:
Not sure why that might happen... my only idea would be to inspect the object files and see what's defined/exported and compare that to what's being referenced. Although I haven't figured out how to do so on my machine yet, because apparently the .o files that make up `out/WaylandRelease/obj/ui/gl/libgl_wrapper.a` - including `gl_surface_egl.o`, where the symbol in question symbol should be defined - are LLVM IR bitcode, and I don't know how to use that[^1]. It might be different for your build, though, because I only have a regular Linux desktop release build (I don't have a Yocto build ready right now).

Alternatively, I think you might have success using the already mentioned dunfell-clang14 branch (`rakuco/meta-clang/tree/dunfell-clang14` instead of `kraj/meta-clang/tree/dunfell-clang12`), but of course I understand if you don't want to use that experimental branch.

[^1]: `llvm-dis` gives me this error: `Unknown attribute kind (86) (Producer: 'LLVM17.0.0git' Reader: 'LLVM 15.0.7')`, and I don't want to compile it from source.

rakuco, 2023-08-09:
Uninformed guess here: every now and then people file issues related to undefined GL symbols, and IIRC most of the time they seem related to imx and different GL implementations. I wonder if that's the case here?

barndoor2276, 2023-08-09:
Could be. I'm building for an imx8 device. I'll give dunfell-clang14 a try and see. Not sure I would be willing to proceed with that in a production setting.

barndoor2276, 2023-08-15:
We're going to stick with the tried and true version 90 until we either change hardware or migrate our BSP to a newer version of Yocto (currently on dunfell). We've actually been testing version 83 with better results for hardware acceleration. I'm wondering if this just might be the boat we're stuck in due to our custom hardware.

I'll close this issue.

closed 2023-08-15 by barndoor2276