OSSystems/meta-browser #747: firefox segmentation fault

chrisvollorenew, 2023-08-29:
Hello, I am trying to run firefox headless and I get a segmentation fault with not much more info. 

```
root@r16-renew-vista-e:~# firefox -headless --safe-mode localhost
*** You are running in headless mode.
Segmentation fault
root@r16-renew-vist
```
Here's some info about firefox and my system: 
```
root@r16-renew-vista-e:~# firefox --version
Mozilla Firefox 68.9.0esr
root@r16-renew-vista-e:~# uname -a
Linux r16-renew-vista-e 5.17.0-rc8 #1 SMP Thu May 4 23:20:23 UTC 2023 armv7l armv7l armv7l GNU/Linux
```

I ran a strace to get more info with `strace  firefox -headless --safe-mode localhost 2> strace.out`
[strace.log](https://github.com/OSSystems/meta-browser/files/12467886/strace.log)

The last lines show:
```
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
unlink("/home/root/.mozilla/firefox/md9k5kcw.default-default/lock") = 0
close(4)                                = 0
rt_sigaction(SIGSEGV, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6bd0711}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [SEGV], NULL, 8) = 0
gettid()                                = 4834
getpid()                                = 4834
tgkill(4834, 4834, SIGSEGV)             = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_TKILL, si_pid=4834, si_uid=0} ---
+++ killed by SIGSEGV +++
```

My yocto configuration is kirkstone: 
```
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    refspec: fa7dd8ac75a98c04a4ca2cf90882c2b1c27b9679
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:

  # updated to kirkstone commit
  meta-openembedded:
    url: "https://github.com/openembedded/meta-openembedded.git"
    refspec: 4da92ed9be41734f6ced46b981958e2e868cbff2
    layers:
      meta-oe:
      meta-python:
      meta-networking:
      meta-filesystems:
      meta-webserver:
      meta-perl:

  # required by meta-webkit, and requires meta-perl
  # updated to kirkstone commit
  meta-security:
    url: "https://git.yoctoproject.org/git/meta-security"
    refspec: 06c240c3c235fa06fe724e278d53463ac67cc26b

  # updated to kirkstone commit
  # rolled back to cog 0.12.4 (WPE WebKit 2.36.3) 
  # -- b/c 0.14.1 (WPE WebKit 2.38.2) had graphics artifacts
  meta-webkit:
    url: "https://github.com/Igalia/meta-webkit.git"
    refspec: f707dbab77cc0f7f5abc2c13c3f3fe6db4e569c4

  meta-browser:
    url: "https://github.com/OSSystems/meta-browser.git"
    refspec: 0f2de2d3ede1388b841642ad5454831db6bda315
    layers:
      meta-firefox:

  meta-clang:
    url: "https://github.com/kraj/meta-clang.git"
    refspec: 2d08d6bf376a1e06c53164fd6283b03ec2309da4
```

Please help!

chrisvollorenew, 2023-08-29:
I noticed the firefox recipe has not been updated in a year ... maybe that has something to do with the issue here. Is firefox not well supported here?

OldManYellsAtCloud, 2023-09-03:
If interested, I have a fork with a more up to date Firefox (though I can't offer n awful lot more support than you get here). You can find it here: https://github.com/OldManYellsAtCloud/meta-browser

Regarding strace, it could be more readable if you could run something like `strace -s 500 -ff -o /tmp/fflog -v -y firefox -headless --safe-mode localhost` - this will create a bunch of files in /tmp folder, one for each thread.

otavio, 2023-09-03:
If desired, a PR would be nice :-)

otavio, 2023-09-03:
@OldManYellsAtCloud you could maintain firefox here, so we avoid the fork

chrisvollorenew, 2023-09-03:
Thank you @OldManYellsAtCloud - it am very interested in Firefox 116 because if it’s support for v4l2 (as I understand earlier versions dropped support for v4l2 and gstreamer) - do you plan to bump to version 116 when it is released?

OldManYellsAtCloud, 2023-09-03:
@otavio - will try to get back on this topic during the week. Principally I would be quite interested in that.
@chrisvollorenew - Firefox is currently at version 117, and I have updated my repo a few days ago to this version. (In the commit history you can also find v116, if that would be better for whatever reason).

otavio, 2025-03-28:
There is a much newer version now available in the layer, so give it a try, and if you have any issues, please open a new issue with the details regarding that

closed 2025-03-28 by otavio