OSSystems/meta-browser #756: Please submit meta-chromium and meta-firefox to layerindex

moto-timo, 2023-09-29:
The layerindex has been unable to  perform updates since the split to having meta-chromium and meta-firefox subdirectories.

https://layers.openembedded.org/layerindex/updates/37925/
ERROR: conf/layer.conf not found for layer meta-browser - is subdirectory set correctly?

Updates are now disabled for the meta-browser entry, please submit meta-chromium and meta-firefox as new layers.

layerindex is not psychic.

kraj, 2023-09-29:
done

moto-timo, 2023-09-29:
meta-chromium "Published" so it will hopefully be in the next update within the next hour (roughly three hours between update runs)

moto-timo, 2023-09-29:
meta-firefox needs to be submitted separately?

moto-timo, 2023-09-29:
Nevermind. Also "Published".

moto-timo, 2023-09-29:
I'll report back and close this issue when the updates are complete and no errors.

moto-timo, 2023-09-29:
Passed updates (with a typical annoying warning that is mostly universal right now)
https://layers.openembedded.org/layerindex/updates/37926/

https://layers.openembedded.org/layerindex/branch/master/layer/meta-chromium/
https://layers.openembedded.org/layerindex/branch/master/layer/meta-firefox/

Thank you @kraj

closed 2023-09-29 by moto-timo