OSSystems/meta-browser #761: Update to Chromium 118

MaxIhlenfeldt, 2023-10-12:
Release notes: https://chromereleases.googleblog.com/2023/10/stable-channel-update-for-desktop_10.html

MaxIhlenfeldt, 2023-10-12:
I'll work on this once #757 is merged. I probably won't have time next week because I'll attend BlinkOn.

MaxIhlenfeldt, 2023-10-27:
Hm, fixing the RPi issue currently present in #757 is taking longer than I'd hoped... @nrpt-m do you have an estimate when you might be able to get a stack trace?

nrpt-m, 2023-10-27:
> Hm, fixing the RPi issue currently present in #757 is taking longer than I'd hoped... @nrpt-m do you have an estimate when you might be able to get a stack trace?

@MaxIhlenfeldt, 
Sorry, I won't able to provide rpi4 board stack trace soon as it is remotely available for me and none of the methods are working for me to boot my custom image remotely :-(
Please proceed with merging #757 if it ok to merge without rpi4 test results.
cc: @rwmacleod

nrpt-m, 2023-10-31:
@MaxIhlenfeldt 
Is Chromium-118 PR coming soon or still some time is there ? If it is coming soon then, I can wait for chromium-118 upgrade in our product.

wrightleft, 2023-11-01:
Looks like 119 is the latest stable release for Linux now.

https://chromiumdash.appspot.com/releases?platform=Linux

MaxIhlenfeldt, 2023-11-02:
> Looks like 119 is the latest stable release for Linux now.

Indeed, but I'm not sure whether it makes sense to update two versions at once. Personally, I'd still first update to 118 and then to 119. @kraj @rakuco wdyt?

MaxIhlenfeldt, 2023-11-02:
> @MaxIhlenfeldt Is Chromium-118 PR coming soon or still some time is there ? If it is coming soon then, I can wait for direct chromium-118 upgrade in our product instead of Chromium-117.

Unfortunately I haven't started yet, but I plan to do so today. I hope to send the PR some time next week.

rakuco, 2023-11-02:
Moving one milestone at a time is much easier IMO, so I'd go with that option.