OSSystems/meta-browser #763: chromium tarball is much too big: split source?

rwmacleod, 2023-10-15:
@nrpt-m  mentioned that with upstream adding Rust in chromium 117, the tarball is ~3 GB.
Even before this, chromium-116.0.5845.187.tar.xz is 1.5 GB and 13 GB unpacked. 
It seemed hard to believe that  all of that data would be required to build chromium.
Using ncdu, the top level directories that are > 100 MiB are:

    9.7 GiB [##########] /third_party                                                                                                                                                         
  822.8 MiB [          ] /chrome
  420.8 MiB [          ] /components
  288.4 MiB [          ] /buildtools 
  199.6 MiB [          ] /tools
  126.0 MiB [          ] /ios
  110.9 MiB [          ] /net
  104.3 MiB [          ] /ui
  101.0 MiB [          ] /ash  

and the largest third-party componets are:
    1.6 GiB [##########] /llvm  
    1.5 GiB [######### ] /angle
    1.0 GiB [######    ] /swiftshader
  649.2 MiB [###       ] /dawn                                                                                                                                                                
  454.5 MiB [##        ] /devtools-frontend
  373.9 MiB [##        ] /catapult

Is it possible to use llvm and rust from YP/OE and does anyone think that it would be a good idea to look at other large directories and remove them or split them into a seperate tarball if not needed. Does anyone object to such a change assuming that it's not too painful to implement and maintain?

rakuco, 2023-10-17:
The [tarball publishing process upstream](https://source.chromium.org/chromium/chromium/tools/build/+/main:recipes/recipes/publish_tarball.py;l=116;drc=0aa0d6544f37d294001834d295d79056df373a74) already creates a tarball called "chromium-lite" which removes [some directories](https://source.chromium.org/chromium/chromium/tools/build/+/main:recipes/recipe_modules/chromium/resources/export_tarball.py;l=124;drc=b8b94bc7142da33a1102685d85c56c946800b522), but this list is not very up-to-date and actively maintained.

As a first step one could see if this tarball is considerably smaller than the one we use and whether one can actually build Chromium from it.

As for using LLVM and Rust from YP/OE and the rest of the suggestion: do you mean creating a tarball of our own?

MaxIhlenfeldt, 2024-12-04:
https://groups.google.com/a/chromium.org/g/chromium-packagers/c/oE60kVFFMyQ/m/xxF0DCnkAQAJ seems interesting:

> Since distributions link against their own system libraries, and have
> their own toolchains, we can exclude:
> - Debian sysroot(s)
> - Toolchain components (Rust, Clang, LLVM etc.)
> - third_party/instrumented_libs 

I assume we don't want to actually use these third-party tarballs in our recipe, but maybe the `-lite` tarballs will get a little more attention again:
> We are hoping that these changes will be accepted into the '-lite'
> tarballs upstream, as these are no longer significantly smaller
> than the standard Chromium tarball.

MaxIhlenfeldt, 2024-12-18:
https://chromium-review.googlesource.com/c/chromium/tools/build/+/6072713 has landed, and this already affect the latest releases. e.g. https://storage.cloud.google.com/chromium-browser-official/chromium-131.0.6778.139-lite.tar.xz is only 1.9 GB. I'll give building with that a try. @rakuco if that works, should we switch to the lite tarballs?

MaxIhlenfeldt, 2024-12-24:
CL landed and seems to have fixed the issue, but there's no new 131 tarball yet. We should try again once there is, or maybe as part of the next update.

ashie, 2026-10-02:
Although I'm not yet check for latest versions, I'm trying to use lite tarballs.

e.g.) https://github.com/clear-code/meta-browser/tree/chromium-138-lite

In this branch, I extract missing files from the original full tarball then repack them into a tarball in this branch:
https://github.com/clear-code/meta-browser/blob/chromium-138-lite/meta-chromium/recipes-browser/chromium/files/chromium-138.0.7204.232-lite-missing.tar.xz

v132 lacks various libraries (icu, freetype, libxml, snappy) so that the tarball is large.

  * https://github.com/clear-code/meta-browser/tree/chromium-132-lite/meta-chromium
    * https://github.com/clear-code/meta-browser/blob/chromium-132-lite/meta-chromium/recipes-browser/chromium/files/chromium-132.0.6834.83-lite-missing.tar.xz

On the other hand v138 lacks only prebuilt `bindgen`binary at least for my build.
It would be happy if latest versions resolve it so that we don't need to repack missing files.
I'll check it later.

ashie, 2026-10-23:
An update on using the lite tarball with Chromium 150.

My Scarthgap example is available here: [clear-code/meta-browser: scarthgap-chromium-150-lite](https://github.com/clear-code/meta-browser/tree/scarthgap-chromium-150-lite)

In my Scarthgap setup, I still need to supplement the lite tarball with the original prebuilt `bindgen` and, unlike Chromium 138, `rustfmt`. Chromium 150 explicitly runs `rustfmt` after generating bindings.

The files I need from the matching full tarball are:

```text
third_party/rust-toolchain/bin/bindgen
third_party/rust-toolchain/bin/rustfmt
third_party/rust-toolchain/lib/librustc_driver-<hash>.so
```

I tried replacing these with Yocto-built tools, but the `bindgen` 0.71.1 provided by my meta-clang version generates identifiers named `gen` without escaping them. Chromium's Rust formatting configuration uses edition 2024, where `gen` is reserved, so the subsequent `rustfmt` invocation fails. This is a [known bindgen issue](https://github.com/rust-lang/rust-bindgen/issues/3093).

Wrynose and master look more promising:

- [Wrynose provides bindgen 0.72.1](https://github.com/openembedded/openembedded-core/blob/wrynose/meta/recipes-devtools/bindgen-cli/bindgen-cli_0.72.1.bb), with exactly the same source revision as Chromium 150 (`d874de8d646d9b8a3e7ba2db2bcd52f2fba8f1f5`).
- [master currently provides bindgen 0.73.2](https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-devtools/bindgen-cli/bindgen-cli_0.73.2.bb). It also handles `gen` and retains the CLI options Chromium uses.

Using these may eliminate the need to supplement the tarball, but some recipe/build-system changes are still needed. For example, in `chromium-gn.inc`:

```bitbake
DEPENDS += "bindgen-cli-native clang-native"

GN_ARGS:append = ' rust_bindgen_root="${STAGING_DIR_NATIVE}/usr"'
```

This makes Chromium use the native sysroot's `bin/bindgen`, `bin/rustfmt`, and `lib/libclang.so`.

The Rust recipes in both branches do not install `rustfmt` for native builds by default, so an additional `rust_%.bbappend` would be needed, for example:

```bitbake
do_install:append:class-native() {
    rust_runx install rustfmt
    patchelf --set-rpath '$ORIGIN/../lib' ${D}${bindir}/rustfmt
}
```

There is also a separate issue with hardcoded Ninja input dependencies on `//third_party/rust-toolchain/bin/rustc`, even though the commands already use Yocto's compiler. These need to reference `${rust_sysroot}/bin/rustc` instead, in:

```text
build/config/rust.gni
build/rust/std/BUILD.gn
build/rust/cargo_crate.gni
```

With that fixed, the bundled `rustc` and `rustlib` directory should not be needed; compilation continues to use Yocto's Rust compiler and standard libraries.

I have only inspected the Wrynose/master approach statically, so the snippets above are proposed changes, not a verified build recipe. In particular, compatibility with Yocto's libclang and the resulting generated bindings still needs build testing.