OSSystems/meta-browser #772: chromium: "Illegal instruction" error on RPi4/dunfell

MaxIhlenfeldt, 2023-11-22:
See the discussions at #757 and #771.

MaxIhlenfeldt, 2023-11-22:
@nrpt-m, I've tried to open the new coredump you uploaded, but it's still failing with the same warnings as last time. What output do you get if you run `gdb chromium-bin core` and then run `bt` inside gdb?

If that doesn't give you a useful backtrace, you might need to do a build with `DEBUG_BUILD = "1"`.

nrpt-m, 2023-11-22:
> @nrpt-m, I've tried to open the new coredump you uploaded, but it's still failing with the same warnings as last time. What output do you get if you run `gdb chromium-bin core` and then run `bt` [...]

@MaxIhlenfeldt 
Actually gdb was not installed on that rpi4/dunfell image so, will add the gdb and test it again as you suggested.

nrpt-m, 2023-11-23:
@MaxIhlenfeldt ,
My rpi4 unit is remotely available so, couldn't login as root and with some user I have tried running gdb on chromium-bin. Please find the attached gdb logs.
[chromium-119-dunfell-rpi4-gdb-logs.txt](https://github.com/OSSystems/meta-browser/files/13446720/chromium-119-dunfell-rpi4-gdb-logs.txt)
 
Please let me know what next to be done for further debugging on this.