OSSystems/meta-browser #773: Chromium: build 119.0.6045.159 fails for kirkstone with vaapi enabled

AndreasUfert, 2023-11-26:
Hi,

we need to enable the following PACKAGECONFIG for chromium-x11 in Yocto 4.0 (kirkstone):

```
PACKAGECONFIG += "proprietary-codecs"
PACKAGECONFIG += "use-vaapi"
```

For this to work, since https://github.com/OSSystems/meta-browser/issues/732 we had to add the following recipes picked from langdale to our local layer:

```
$ find recipes-*
recipes-graphics
recipes-graphics/libva
recipes-graphics/libva/libva_2.15.0.bb
recipes-graphics/libva/libva-utils_2.15.0.bb
recipes-graphics/libva/libva-initial_2.15.0.bb
recipes-graphics/libva/libva-intel-utils_2.16.0.bb
recipes-graphics/libva/libva-intel_2.16.0.bb
recipes-graphics/libva/libva.inc
recipes-multimedia
recipes-multimedia/libva
recipes-multimedia/libva/intel-media-driver_22.5.4.bb
recipes-multimedia/libva/files
recipes-multimedia/libva/files/0001-Fix-uClibc-build.patch
```

This has been working for months up until chromium-x11 118.0.5993.117.

Now, when building chromium-x11 119.0.6045.159 again, errors occur:

```
../../media/gpu/chromeos/image_processor_backend.cc:37:36: error: defaulting this default constructor would delete it after its first declaration
ImageProcessorBackend::PortConfig::PortConfig() = default;
                                   ^
../../media/gpu/chromeos/image_processor_backend.h:77:18: note: default constructor of 'PortConfig' is implicitly deleted because field 'fourcc' of const-qualified type 'const media::Fourcc' would not be initialized
    const Fourcc fourcc;
                 ^
1 error generated.
```

However, it builds successfully

- *without* use-vaapi using kirkstone and
- *with* use-vaapi but using mickledore branches of poky, meta-openembedded, meta-clang etc.

So again, there's some package in kirkstone that has been updated in mickledore (or langdale) but is too old in kirkstone. Can anyone tell which package that might be?

Hint: The commit in chromium that might have triggered the problem is https://github.com/chromium/chromium/commit/da9b8a600ac8ff5c03b39043267bf3a8b7be3afc

Thanks in advance!

Regards, Andreas

fwitk, 2023-12-01:
I can confirm this, i have exact the same error.

MaxIhlenfeldt, 2023-12-05:
I suspect this has the same root cause as the error fixed by https://github.com/OSSystems/meta-browser/blob/master/meta-chromium/recipes-browser/chromium/files/0020-Fix-implicitly-deleted-default-constructor-build-err.patch.

It would be great if you could try doing a similar approach to that patch (add explicit in-class initializations for const members) and let me know if that fixes the issue. Else I can check myself after the 120 update is done.

bkauler, 2023-12-10:
I'm also building chromium-x11 with kirkstone, and have the libva update, and getting the failure with 119.0.6045.159.

My knowledge of C++ is just about nothing; anyway, had a go, created a patch to change the assignment from "default" to "0":

`ImageProcessorBackend::PortConfig::PortConfig() = 0;`

But get this error:

`error: initializer on function does not look like a pure-specifier`

MaxIhlenfeldt, 2023-12-19:
Can you please give the (currently WIP) 120 version of the recipe at #777 a try?

AndreasUfert, 2023-12-19:
Adjusted my subrepos and triggered a build. Will report here in a few hours... Thanks in advance!

bkauler, 2023-12-19:
120.0.6099.109 build, stops here:

```
FAILED: obj/media/gpu/chromeos/common/image_processor_backend.o 
x86_64-poky-linux-clang++ -target x86_64-poky-linux  -m64 -march=nocona -mtune=nocona -mno-sse3 -mfpmath=sse -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/mnt/build/builds/oe/kirkstone/oe-quirky/build-amd64/tmp/work/nocona-64-poky-linux/chromium-x11/120.0.6099.109-r6/recipe-sysroot -MMD -MF obj/media/gpu/chromeos/common/image_processor_backend.o.d -DMEDIA_GPU_IMPLEMENTATION -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_56 -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_56 -DBASE_USE_PERFETTO_CLIENT_LIBRARY=1 -DSK_ENABLE_SKSL -DSK_UNTIL_CRBUG_1187654_IS_FIXED -DSK_USER_CONFIG_HEADER=\"../../skia/config/SkUserConfig.h\" -DSK_WIN_FONTMGR_NO_SIMULATIONS -DSK_DISABLE_LEGACY_SKSURFACE_METHODS -DSK_DISABLE_LEGACY_GRAPHITE_IMAGE_FACTORIES -DSK_DISABLE_LEGACY_GRAPHITE_IMAGE_METHODS -DSK_DISABLE_LEGACY_SKSURFACE_FACTORIES -DSK_DISABLE_LEGACY_SKSURFACE_FLUSH -DSK_DISABLE_LEGACY_SKSURFACE_AS_IMAGE -DSK_DISABLE_LEGACY_SKSURFACE_DISPLAYLIST -DSK_DISABLE_LEGACY_IMAGE_SUBSET_METHODS -DSK_DISABLE_LEGACY_IMAGE_COLORSPACE_METHODS -DSK_DISABLE_LEGACY_IMAGE_RELEASE_PROCS -DSK_DISABLE_LEGACY_GL_BACKEND_SURFACE -DSK_DISABLE_LEGACY_INIT_DECODERS -DSK_DISABLE_LEGACY_GRDIRECTCONTEXT_FLUSH -DSK_DISABLE_LEGACY_GRDIRECTCONTEXT_BOOLS -DSK_DISABLE_LEGACY_GL_GRDIRECTCONTEXT_FACTORIES -DSK_DISABLE_LEGACY_PNG_WRITEBUFFER -DSK_SLUG_DISABLE_LEGACY_DESERIALIZE -DSK_DISABLE_LEGACY_VK_GRDIRECTCONTEXT_FACTORIES -DSK_CODEC_DECODES_JPEG -DSK_ENCODE_JPEG -DSK_ENCODE_PNG -DSK_ENCODE_WEBP -DSK_GANESH -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DSK_GL -DSK_VULKAN=1 -DSK_GRAPHITE -DVK_USE_PLATFORM_XCB_KHR -DWEBP_EXTERN=extern -DUSE_EGL -DLIBYUV_DISABLE_NEON -DCRASHPAD_ZLIB_SOURCE_EXTERNAL -DU_USING_ICU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DUSE_CHROMIUM_ICU=1 -DU_ENABLE_TRACING=1 -DU_ENABLE_RESOURCE_TRACING=0 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DGOOGLE_PROTOBUF_INTERNAL_DONATE_STEAL_INLINE=0 -DHAVE_PTHREAD -DLEVELDB_PLATFORM_CHROMIUM=1 -DVK_NO_PROTOTYPES -DUSE_VULKAN_XCB -DLIBGAV1_MAX_BITDEPTH=10 -DLIBGAV1_THREADPOOL_USE_STD_MUTEX -DLIBGAV1_ENABLE_LOGGING=0 -DLIBGAV1_PUBLIC= -I../.. -Igen -I../../third_party/perfetto/include -Igen/third_party/perfetto/build_config -Igen/third_party/perfetto -I../../third_party/skia -Igen/third_party/skia -I../../third_party/wuffs/src/release/c -I../../third_party/vulkan/include -I../../third_party/vulkan-deps/vulkan-headers/src/include -I../../third_party/libwebp/src/src -I../../net/third_party/quiche/overrides -I../../net/third_party/quiche/src/quiche/common/platform/default -I../../net/third_party/quiche/src -Igen/third_party/dawn/include -I../../third_party/dawn/include -I../../third_party/khronos -I../../gpu -I../../third_party/libyuv/include -Igen/shim_headers/flac_shim -I../../base/allocator/partition_allocator/src -Igen/base/allocator/partition_allocator/src -I../../third_party/abseil-cpp -I../../third_party/boringssl/src/include -I../../third_party/protobuf/src -Igen/protoc_out -I../../third_party/ipcz/include -I../../third_party/crashpad/crashpad -I../../third_party/crashpad/crashpad/compat/linux -I../../third_party/crashpad/crashpad/compat/non_win -I../../third_party/zlib -I../../third_party/icu/source/common -I../../third_party/icu/source/i18n -I../../third_party/ced/src -I../../third_party/mesa_headers -I../../third_party/libwebm/source -I../../third_party/leveldatabase -I../../third_party/leveldatabase/src -I../../third_party/leveldatabase/src/include -I../../third_party/libaom/source/libaom -Igen/net/third_party/quiche/src -I../../third_party/libgav1/src -I../../third_party/libgav1/src/src -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wno-invalid-offsetof -Wno-vla-extension -Wno-thread-safety-reference-return -Wno-delayed-template-parsing-in-cxx20 -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables -m64 -msse3 -Xclang -fdebug-compilation-dir -Xclang . -ftrivial-auto-var-init=pattern -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-math-errno -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -I/mnt/build/builds/oe/kirkstone/oe-quirky/build-amd64/tmp/work/nocona-64-poky-linux/chromium-x11/120.0.6099.109-r6/recipe-sysroot/usr/include/glib-2.0 -I/mnt/build/b [... cut ...]
../../media/gpu/chromeos/image_processor_backend.cc:37:36: error: defaulting this default constructor would delete it after its first declaration
ImageProcessorBackend::PortConfig::PortConfig() = default;
                                   ^
../../media/gpu/chromeos/image_processor_backend.h:77:18: note: default constructor of 'PortConfig' is implicitly deleted because field 'fourcc' of const-qualified type 'const media::Fourcc' would not be initialized
    const Fourcc fourcc;
                 ^
1 error generated.
```

bkauler, 2023-12-21:
I'm stumbling around in the dark somewhat, as hardly know anything about C++; however, this looks like a clue:

https://stackoverflow.com/questions/77283333/why-is-the-explicitly-defaulted-default-constructor-implicitly-deleted-when-ther

...it looks like the problem goes away with clang 15. Kirkstone has clang 14.0.6.

So, how would we go about backporting clang 15?
...and hopefully doing so won't break anything else.

bkauler, 2024-01-01:
I reported that backported meta-clang from langdale, but then mesa failed. However, the reason my be because I have a "forward ported" mesa in kirkstone. Kirkstone has mesa 22.0.3 but I rolled back to 21.3.7, for reason given here: https://www.phoronix.com/news/2021-Retire-Mesa-Classic-Main
So, mesa 22.0.3 might have worked with langdale meta-clang.

Anyway, I have decided to leave kirkstone, have built 120.0.6099.109 successfully with master branches from the OE repos.

MaxIhlenfeldt, 2024-01-08:
> So, how would we go about backporting clang 15?

We're already planning to move to clang 16, as clang 15 still has some problems, see #716. Closing this as WontFix in the hope that the clang update will happen soon, please let me know if you disagree with this.

closed 2024-01-08 by MaxIhlenfeldt