OSSystems/meta-browser #775: Update to Chromium 120

MaxIhlenfeldt, 2023-12-08:
Release notes:
* https://chromereleases.googleblog.com/2023/12/stable-channel-update-for-desktop.html
* https://chromereleases.googleblog.com/2023/12/stable-channel-update-for-desktop_6.html

MaxIhlenfeldt, 2023-12-08:
Already started working on this, but forgot to create the issue.

bkauler, 2024-01-01:
Hi, success, 120.0.6099.109 builds for me.

I have decided to move to scarthgap when it is released. As there are no scarthgap branches yet, downloaded from master branches, and success compiling chromium. So 120.0.6099.109 can be merged I reckon.

Note, I created chromium-x11_%.bbappend to turn on some things, especially vaapi:

`PACKAGECONFIG:append:class-target = " cups proprietary-codecs use-vaapi"`

closed 2024-01-08 by MaxIhlenfeldt