OSSystems/meta-browser #784: Update to Chromium 121

MaxIhlenfeldt, 2024-01-31:
Release notes:
* https://chromereleases.googleblog.com/2024/01/stable-channel-update-for-desktop_23.html
* https://chromereleases.googleblog.com/2024/01/stable-channel-update-for-desktop_30.html

MaxIhlenfeldt, 2024-01-31:
Will start working on this as soon as I've cleaned up our patches to not include anything dunfell-specific any more. I'm doing test builds right now to verify the patches that are supposedly not needed for kirkstone and later releases actually aren't needed. Adding Rust to the build of course slowed it down even further, but I should be able to work on this update next week at the latest.

MaxIhlenfeldt, 2024-02-08:
@kraj it turns out Chromium 121 has code that relies on `std::pointer_traits` being SFINAE, which was only recently implemented in libc++ in https://github.com/llvm/llvm-project/commit/078651b6de4b767b91e3e6a51e5df11a06d7bc4f. I think it might be easier to backport that patch to libc++ than to remove that requirement in Chromium - [the diff][1] is quite small. Would that be something that could live in `meta-clang` or should we add it in our layer using a `.bbappend` file?

[1]: https://source.chromium.org/chromium/chromium/src/+/main:third_party/libc++/src/include/__memory/pointer_traits.h;l=1;bpv=1;bpt=0;drc=7cee6b00d34adc4da2ff2a4d373e44fc27a9a223;dlc=6f868b200390639e4b70fd5b176c7d45a0a76a3f

kraj, 2024-02-08:
@MaxIhlenfeldt please propose a backport directly to meta-clang

nrpt-m, 2024-02-14:
@MaxIhlenfeldt, @rakuco 
cc: @rwmacleod 
Hi all, I am leaving WindRiver and from now on words @rjanani-p will support you for all the new Chromium PR testing.
It was great working with all of you and thanks for all the support !!

Best Regards,
Narpat Mali (@nrpt-m)

MaxIhlenfeldt, 2024-02-15:
@nrpt-m thank you for your contributions and testing, and all the best for your future endeavours!

closed 2024-03-20 by MaxIhlenfeldt