OSSystems/meta-browser #785: Adding Rust Breaks Chromium on Kirkstone

bobolopolis, 2024-01-31:
Commit 0438fba4e773 added rust support for Chromium, but this breaks on kirkstone:

```
ERROR: Nothing PROVIDES 'rust' (but /mnt/ssd/poky-kirkstone/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_120.0.6099.224.bb DEPENDS on or otherwise requires it)
rust was skipped: Rust recipe doesn't work for target builds at this time. Fixes welcome.
ERROR: Required build target 'chromium-x11' has no buildable providers.
Missing or unbuildable dependency chain was: ['chromium-x11', 'rust']
```

Is target rust really needed?

Also added in this commit is a bbappend to rust that adds an install:
```
rust_do_install:append() {                                              
     install -m 0644 ${WORKDIR}/rust-targets/${RUST_TARGET_SYS}.json ${D}${libdir}/rust   lib/${RUST_TARGET_SYS}/target.json
 }
```
This breaks building rust-native on kirkstone as well:
```
install: cannot stat '/mnt/ssd/poky-kirkstone/build/tmp/work/x86_64-linux/rust-native/1.59.0-r0/rust-targets/x86_64-unknown-linux-gnu.json': No such file or directory
```

MaxIhlenfeldt, 2024-02-01:
>  this breaks on kirkstone:

Sorry, we should have mentioned more clearly that for kirkstone you need to use a newer version of Rust via https://git.yoctoproject.org/meta-lts-mixins/?h=kirkstone/rust-1.70 (rust-1.68 seems to work as well).

> Is target rust really needed?

Yes, Chromium has Rust code that can't be disabled since version 121.

> This breaks building rust-native on kirkstone as well:

I suppose that should be fixed by using the mixin mentioned above. Please let us know if that's not the case.

bobolopolis, 2024-02-02:
I understand needing rust-native, but seems strange it needs the target rust too. I'm not very familiar with rust though, so maybe that's expected.

I added meta-lts-mixins and get farther in the build, but unfortunately the build fails in chromium. On the kirkstone/rust branch, it fails with:
```
| FAILED: prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libprofiler_builtins.rlib
| ln -f obj/build/rust/std/lib/libprofiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libprofiler_builtins.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libprofiler_builtins.rlib && cp -af obj/build/rust/std/lib/libprofiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libprofiler_builtins.rlib)
| cp: cannot stat 'obj/build/rust/std/lib/libprofiler_builtins.rlib': No such file or directory
```

On the kirkstone/rust-1.70 branch, it fails with:

```
| FAILED: obj/third_party/rust/qr_code/v2/lib/libqr_code-qr_code-2.rlib
| "python3" "../../build/rust/rustc_wrapper.py" --rustc=../../../recipe-sysroot-native/usr/bin/rustc --depfile=obj/third_party/rust/qr_code/v2/lib/libqr_code-qr_code-2.rlib.d --rsp=obj/third_party/rust/qr_code/v2/lib/libqr_code-qr_code-2.rlib.rsp -- -Clinker="x86_64-poky-linux-clang++ -target x86_64-poky-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/mnt/ssd/poky-kirkstone/build/tmp/work/core2-64-poky-linux/chromium-x11/120.0.6099.224-r0/recipe-sysroot" --crate-name qr_code ../../third_party/rust/qr_code/v2/crate/src/lib.rs --crate-type rlib -Cmetadata=qr_code-2 --edition=2018 -Cforce-unwind-tables=no -Crelocation-model=pic -Cembed-bitcode=no -Coverflow-checks=on -Cdefault-linker-libraries -Zdep-info-omit-d-target -Zmacro-backtrace -Zremap-cwd-prefix=. --color=always --target=x86_64-poky-linux-gnu -Cembed-bitcode=no -Ccodegen-units=1 --cfg cr_rustc_revision=\"custom\" -Copt-level=3 -Cembed-bitcode=no --cap-lints=allow -Dunsafe_op_in_unsafe_fn --sysroot=prebuilt_rustc_sysroot --emit=dep-info=obj/third_party/rust/qr_code/v2/lib/libqr_code-qr_code-2.rlib.d,link -o obj/third_party/rust/qr_code/v2/lib/libqr_code-qr_code-2.rlib LDFLAGS RUSTENV OUT_DIR=../../../../../../out/Release/gen/third_party/rust/qr_code/v2/lib CARGO_PKG_AUTHORS=kennytm\ \<kennytm@gmail.com\>,\ Riccardo\ Casatta\ \<riccardo.casatta@gmail.com\> CARGO_PKG_VERSION=2.0.0 CARGO_PKG_NAME=qr_code CARGO_PKG_DESCRIPTION=QR\ code\ encoder\ in\ Rust,\ support\ structured\ append\ \(data\ in\ multiple\ qrcodes\) CARGO_MANIFEST_DIR=/mnt/ssd/poky-kirkstone/build/tmp/work/core2-64-poky-linux/chromium-x11/120.0.6099.224-r0/chromium-120.0.6099.224/third_party/rust/qr_code/v2/lib/crate
| error[E0786]: found invalid metadata files for crate `std`
|   |
|   = note: invalid metadata version found: /mnt/ssd/poky-kirkstone/build/tmp/work/core2-64-poky-linux/chromium-x11/120.0.6099.224-r0/chromium-120.0.6099.224/out/Release/prebuilt_rustc_sysroot/lib/rustlib/x86_64-poky-linux-gnu/lib/libstd.rlib
| 
| error[E0786]: found invalid metadata files for crate `std`
|   --> ../../third_party/rust/qr_code/v2/crate/src/lib.rs:37:5
|    |
| 37 | use std::ops::Index;
|    |     ^^^
```
There are a lot more similar errors about invalid metadata after that too.

I have a build running with the kirkstone/rust-1.68 branch that seems to be getting farther, but since these builds take a while I won't find out until tomorrow.

Were you able to do a build on kirkstone? Maybe this is something weird in my environment? I'm building for qemux86-64 in an Ubuntu 22.04 container.

bobolopolis, 2024-02-02:
My build with the kirkstone/rust-1.68 branch worked. I then made a clean build with kirkstone/rust-1.70 (empty sstate cache) and that worked as well. The kirkstone/rust branch still fails for me. Not sure why the 1.70 branch wasn't working, maybe tmp wasn't in a clean state.

I'm not sensitive to the rust version elsewhere in my build, so using the kirkstone/rust-1.70 branch is good enough for me at this point. Would be nice to add something to the readme about the new requirements for kirkstone. Thanks for the help!

closed 2024-02-02 by bobolopolis

bobolopolis, 2024-02-02:
In looking at the kirkstone/rust and kirkstone/rust-1.70 branches of meta-lts-mixins more, the difference is "profiler" support. The kirkstone/rust-1.70 branch sets this in the rust_1.70.0.bb file
```
150     # Support for the profiler runtime to generate e.g. coverage report,
151     # PGO etc.                                                          
152     config.set("build", "profiler", e(True)) 
```

On the kirkstone/rust branch, this is toggled to False, and it appears Chromium needs the profiler enabled.