OSSystems/meta-browser #786: Chromium 120.0.6099.224 fails to build.

Gerry546, 2024-02-01:
Hi,

I cannot seem to get this recipe to build with the new rust support
I use all the default flags of the recipe and follow master on all dependend layers.

I would like to know what I can do in order to get chromium to work again. :-)

I have taken out the snippet of the `do_compile` output where it seems to break for some reason:
<pre>
[192/64373] rm -f obj/base/third_party/double_conversion/libdouble_conversion.a && "aarch64-oe-linux-llvm-ar" -T -r -c -s -D obj/base/third_party/double_conversion/libdouble_conversion.a @"obj/base/third_party/double_conversion/libdouble_conversion.a.rsp"
[193/64373] touch obj/build/linux/libudev/udev0_loader.stamp
[194/64373] touch obj/build/linux/libudev/udev1_loader.stamp
[195/64373] touch obj/build/linux/libudev/libudev.stamp
[196/64373] ln -f obj/build/rust/std/lib/libmemchr.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libmemchr.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libmemchr.rlib && cp -af obj/build/rust/std/lib/libmemchr.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libmemchr.rlib)
[197/64373] ln -f obj/build/rust/std/lib/libminiz_oxide.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libminiz_oxide.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libminiz_oxide.rlib && cp -af obj/build/rust/std/lib/libminiz_oxide.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libminiz_oxide.rlib)
[198/64373] ln -f obj/build/rust/std/lib/libobject.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libobject.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libobject.rlib && cp -af obj/build/rust/std/lib/libobject.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libobject.rlib)
[199/64373] ln -f obj/build/rust/std/lib/libprofiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libprofiler_builtins.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libprofiler_builtins.rlib && cp -af obj/build/rust/std/lib/libprofiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libprofiler_builtins.rlib)
FAILED: prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libprofiler_builtins.rlib 
ln -f obj/build/rust/std/lib/libprofiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libprofiler_builtins.rlib 2>/dev/null || (rm -rf prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libprofiler_builtins.rlib && cp -af obj/build/rust/std/lib/libprofiler_builtins.rlib prebuilt_rustc_sysroot/lib/rustlib/aarch64-oe-linux-gnu/lib/libprofiler_builtins.rlib)
cp: cannot stat 'obj/build/rust/std/lib/libprofiler_builtins.rlib': No such file or directory
[200/64373] aarch64-oe-linux-clang++ -target aarch64-oe-linux  -mcpu=cortex-a72 -march=armv8-a+crc -mbranch-protection=standard -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/recipe-sysroot -MMD -MF obj/base/allocator/partition_allocator/src/partition_alloc/allocator_core/reservation_offset_table.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DIS_PARTITION_ALLOC_IMPL -I../.. -Igen -I../../base/allocator/partition_allocator/src -Igen/base/allocator/partition_allocator/src -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wno-invalid-offsetof -Wno-vla-extension -Wno-thread-safety-reference-return -Wno-delayed-template-parsing-in-cxx20 -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables -mbranch-protection=standard -mno-outline -Xclang -fdebug-compilation-dir -Xclang . -ftrivial-auto-var-init=pattern -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Xclang -target-feature -Xclang +mte -Wexit-time-destructors -O3 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-math-errno -std=c++20 -Wno-trigraphs -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe  -feliminate-unused-debug-types   -fmacro-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224=/usr/src/debug/chromium-ozone-wayland/120.0.6099.224  -fdebug-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224=/usr/src/debug/chromium-ozone-wayland/120.0.6099.224  -fmacro-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/out/Release=/usr/src/debug/chromium-ozone-wayland/120.0.6099.224  -fdebug-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/out/Release=/usr/src/debug/chromium-ozone-wayland/120.0.6099.224  -fdebug-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/recipe-sysroot=  -fmacro-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/recipe-sysroot=  -fdebug-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/recipe-sysroot-native= -g1 -fvisibility-inlines-hidden -stdlib=libc++ -c ../../base/allocator/partition_allocator/src/partition_alloc/reservation_offset_table.cc -o obj/base/allocator/partition_allocator/src/partition_alloc/allocator_core/reservation_offset_table.o
[201/64373] aarch64-oe-linux-clang++ -target aarch64-oe-linux  -mcpu=cortex-a72 -march=armv8-a+crc -mbranch-protection=standard -mlittle-endian --dyld-prefix=/usr -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/recipe-sysroot -MMD -MF obj/base/allocator/partition_allocator/src/partition_alloc/allocator_core/partition_alloc.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DIS_PARTITION_ALLOC_IMPL -I../.. -Igen -I../../base/allocator/partition_allocator/src -Igen/base/allocator/partition_allocator/src -Wall -Wextra -Wimplicit-fallthrough -Wextra-semi -Wunreachable-code-aggressive -Wthread-safety -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wloop-analysis -Wno-unneeded-internal-declaration -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-ignored-pragma-optimize -Wno-deprecated-builtins -Wno-bitfield-constant-conversion -Wno-deprecated-this-capture -Wno-invalid-offsetof -Wno-vla-extension -Wno-thread-safety-reference-return -Wno-delayed-template-parsing-in-cxx20 -Wshadow -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -ffp-contract=off -flto=thin -fsplit-lto-unit -fwhole-program-vtables -mbranch-protection=standard -mno-ou [... cut ...] .0.6099.224/recipe-sysroot=  -fdebug-prefix-map=/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/recipe-sysroot-native= -g1 -fvisibility-inlines-hidden -stdlib=libc++ -c ../../base/allocator/partition_allocator/src/partition_alloc/starscan/pcscan_internal.cc -o obj/base/allocator/partition_allocator/src/partition_alloc/allocator_core/pcscan_internal.o
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
</pre>

bobolopolis, 2024-02-02:
I had the same issue: https://github.com/OSSystems/meta-browser/issues/785

I'm on kirkstone, and using the kirkstone/rust-1.70 branch of meta-lts-mixins gave me a working build.

Seems you need rust with profiler support enabled as well. The recent rust recipes in oe-core and meta-lts-mixins disable the profiler: https://git.yoctoproject.org/poky/commit/meta/recipes-devtools/rust?id=321aebfa281bd28e368c684ece57867f6bd0cbe7

The kirkstone/rust-1.70 (and kirkstone/rust-1.68) branches still have the profiler enabled.

Gerry546, 2024-02-03:
You are correct that this dependency is apparently needed which is indeed a problem. However, having enabled it again I bump into the next problem:

<pre>
[4560/63971] python3 ../../third_party/catapult/tracing/bin/generate_about_tracing_contents --outdir gen/content/browser/tracing
FAILED: gen/content/browser/tracing/about_tracing.js gen/content/browser/tracing/about_tracing.html 
python3 ../../third_party/catapult/tracing/bin/generate_about_tracing_contents --outdir gen/content/browser/tracing
/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/html_generation_controller.py:21: SyntaxWarning: invalid escape sequence '\('
  if re.search('url\(.+\)', contents):
/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/style_sheet.py:63: SyntaxWarning: invalid escape sequence '\('
  return re.sub('url\((?P<quote>"|\'|)(?P<url>[^"\'()]*)(?P=quote)\)',
/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/style_sheet.py:75: SyntaxWarning: invalid escape sequence '\('
  'url\((?:["|\']?)([^"\'()]*)(?:["|\']?)\)',
/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/js_utils.py:7: SyntaxWarning: invalid escape sequence '\/'
  return js.replace('</script>', '<\/script>')
/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/parse_html_deps.py:296: SyntaxWarning: invalid escape sequence '\/'
  raise Exception('Escape script tags with <\/script>')
Traceback (most recent call last):
  File "/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/html_module.py", line 28, in Parse
    parser_results = parse_html_deps.HTMLModuleParser().Parse(self.contents)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/parse_html_deps.py", line 298, in Parse
    return HTMLModuleParserResults(html)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/common/py_vulcanize/py_vulcanize/parse_html_deps.py", line 164, in __init__
    self._soup = bs4.BeautifulSoup(html, 'html5lib')
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tom/projects/Estalor-Yocto/build/tmp/work/cortexa72-oe-linux/chromium-ozone-wayland/120.0.6099.224/chromium-120.0.6099.224/third_party/catapult/third_party/beautifulsoup4-4.9.3/py3k/bs4/__init__.py", line 243, in __init__
    raise FeatureNotFound(
bs4.FeatureNotFound: Couldn't find a tree builder with the features you requested: html5lib. Do you need to install a parser library?
</pre>

MaxIhlenfeldt, 2024-02-05:
I already stumbled across this myself at the end of last week but didn't have the time yet to take a closer look (and forgot to file an issue). Thanks for reporting, and also thanks to @bobolopolis for figuring out the cause! I'll think about how to best fix this, and will check the `html5lib` issue afterwards.

MaxIhlenfeldt, 2024-02-05:
Removing `profiler_builtins` from [this list](https://source.chromium.org/chromium/chromium/src/+/main:build/rust/std/BUILD.gn;l=93;drc=11b66a50ac6b05ddcc1bb739200684c947ef6a62) fixes the error, although I'm not sure if we'll run into other errors later on in the build.

I'm now also getting the same warnings errors as @Gerry546. I suspect that this is due to master's update from Python 3.11 to 3.12. Chromium tracks this at https://issues.chromium.org/issues/40283283, which e.g. also mentions the `SyntaxWarning`s. They should be fixed in m122, and I hope that the `bs4`/`html5lib` issue will also be gone in that version.

I'll re-visit this issue once the m122 update is ready, and hopefully we can just include the `profiler_builtins` change mentioned above in the update PR and have the master build working again.

MaxIhlenfeldt, 2024-02-06:
Hm, I just realized that the next update I'll do is m121, not m122. I'll see if it's feasible to backport the patches needed to fix the Python 3.12 issues.

rwmacleod, 2024-02-14:
@MaxIhlenfeldt have you considered sending the rust bbappends changes to oe-core / master and to the kirkstone mixing layer?

Yashinde145, 2024-02-14:
The profiler option was disabled in oe-core master to fix rust reproducibility issues but it is enabled in kirkstone branch and build works fine there.

We can check the bbappends  changes and then send it oe-core, if everything works fine.

MaxIhlenfeldt, 2024-02-15:
> We can check the bbappends changes and then send it oe-core, if everything works fine.

I'd be very grateful if you could handle that!

MaxIhlenfeldt, 2024-02-15:
> Hm, I just realized that the next update I'll do is m121, not m122. I'll see if it's feasible to backport the patches needed to fix the Python 3.12 issues.

Quick update: backporting https://crrev.com/c/5092950 should fix the compilation with Python 3.12. I'll include it in the m121 update, which I'll upload soon.

Gerry546, 2024-03-12:
How do you manage to apply this as a backport? I cannot seem to `devtool modify` chromium-ozone-wayland without devtool throwing an error:
` ERROR: ExecutionError('git submodule foreach --recursive  "git tag -f devtool-base"', 128, '', "fatal: No url found for submodule path 'third_party/rust-src' in .gitmodules\n")`

And I can't add this as a patch directly via a `.bbappend` as these files are not in unpacked source.

MaxIhlenfeldt, 2024-03-13:
@Gerry546 please see #791 for the WIP update to m121, which contains the backported patch.

Gerry546, 2024-03-14:
Thanks. That works. :) Good to have something compile again

closed 2024-03-20 by MaxIhlenfeldt