OSSystems/meta-browser #789: master branch became incompatible with kirkstone

lumag, 2024-02-06:
After the commit 0438fba4e773824fd2b41ded2728076deeeedeb1 the layer became incompatible with the kirkstone OE branch, because it depends on `rust-targets` subdir, which was not available before https://github.com/openembedded/openembedded-core/commit/9160e4a37561d8ac882057450a818621bec13bed.

```
| install: cannot stat '/oe/build/tmp-rpb-glibc/work/x86_64-linux/rust-native/1.59.0-r0/rust-targets/x86_64-unknown-linux-gnu.json': No such file or directory
```

MaxIhlenfeldt, 2024-02-06:
Unfortunately we require using Rust from meta-lts-mixins' `kirkstone/rust-1.68` or `kirkstone/rust-1.70` branch. I don't think that Chromium can be compiled with Rust 1.59 anyway.

I'll close this for now, but if you're unable to use a mixin layer, feel free to give compiling Chromium a try with Rust 1.59 on a regular machine. If that works I'll reopen this issue and will see if I can adapt our recipe to work without the `rust-targets` subdir.

closed 2024-02-06 by MaxIhlenfeldt

rakuco, 2024-02-06:
> Unfortunately we require using Rust from meta-lts-mixins' `kirkstone/rust-1.68` or `kirkstone/rust-1.70` branch. I don't think that Chromium can be compiled with Rust 1.59 anyway.

Would it be possible to add some version (or branch?) check like `do_check_llvm_version` to help people? If not, these requirements and different branches should at least be mentioned in the README.

lumag, 2024-02-06:
@MaxIhlenfeldt @rakuco  then could you please remove kirkstone from the list of compatible versions? That's the main source of information and a guarding check in case of the mistakes.

lumag, 2024-02-06:
Another option is to get necessary changes backported to the OE-Core/kirkstone.

MaxIhlenfeldt, 2024-02-06:
The easiest solution would be to remove kirkstone from `LAYERSERIES_COMPAT` but advertise in the README that kirkstone will still work if you use the layer from meta-lts-mixins. Not sure how open OE would be to backport the mixin layer.

lumag, 2024-02-06:
Then backport from langsdale instead of mixins

MaxIhlenfeldt, 2024-02-06:
I don't have the time to backport patches, regardless of whether from langdale or from mixins. @rakuco should we just do the approach outlined in https://github.com/OSSystems/meta-browser/issues/789#issuecomment-1929460221?

rakuco, 2024-03-15:
@MaxIhlenfeldt Sorry for the silence here! I remember seeing some discussions about meta-lts-mixins and something changing for the Rust recipes across different branches; what's the situation at the moment? Does kirkstone still work with meta-lts-mixins, or at all?

MaxIhlenfeldt, 2024-03-20:
@rakuco kirkstone should work with meta-chromium 120, but only when using meta-lts-mixins. Given that it won't be compatible with 121 and higher versions due to the outdated compiler, I'd propose the following:
* create a kirkstone branch off of c76bc3a68356bf11daad4659b5a787cd7714b2a9
* change the kirkstone branch's README to say that you have to use meta-lts-mixins
* remove kirkstone from `LAYERSERIES_COMPAT` on master, and possible on the kirkstone branch as well

Feel free to go ahead with these changes if you agree, or let me know I should go ahead / what you'd like to differently.

rakuco, 2024-03-20:
Sounds good to me; I just didn't understand the part about removing kirkstone from `LAYERSERIES_COMPAT` in the kirkstone branch. Isn't it going to make the branch unbuildable?

If you work on a patch, I can approve the PR quite quickly.

lumag, 2024-03-20:
This is not necessary (and is ridiculous). Just declare dependency on the lts-mixins layer. This way the chromium layer / kirkstone branch won't be buildable without the meta-lts-mixins

MaxIhlenfeldt, 2024-03-20:
Please refrain from labelling my suggestions as ridiculous, I don't think that's very nice.

I'll make the changes outlined in https://github.com/OSSystems/meta-browser/issues/789#issuecomment-2009458009 then, and will send a PR for the kirkstone branch that updates the PR and adds the dependency on meta-lts-mixins.

rakuco, 2024-03-20:
> Please refrain from labelling my suggestions as ridiculous, I don't think that's very nice.

+1

> I'll make the changes outlined in [#789 (comment)](https://github.com/OSSystems/meta-browser/issues/789#issuecomment-2009458009) then, and will send a PR for the kirkstone branch that updates the PR and adds the dependency on meta-lts-mixins.

Awesome, thank you!